Louisvdw/dbus-serialbattery #1031: 2 batteries with the same unique identifier

metapixl, 2024-04-02:
### Describe the problem

I have two FSK 300HP batteries that have the same unique identifier so only one is loading using the latest driver version.  The log files says to change it but not sure where/how to do that and there are no options in the battery internal BMS to change it that I can see. 

### Driver version

1.2.20240227beta

### Venus OS device type

Cerbo GX

### Venus OS version

3.30

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 140.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.500
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10,  15, 20, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2, 0.4, 0.8,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20,   0,   5,  10, 15, 45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.2, 0.3, 0.4,  1,  1,  0

; Charge current SoC limits
CC_SOC_LIMIT1 = 98
CC_SOC_LIMIT2 = 95
CC_SOC_LIMIT3 = 91

; Charge current limits
CC_CURRENT_LIMIT1_FRACTION = 0.1
CC_CURRENT_LIMIT2_FRACTION = 0.3
CC_CURRENT_LIMIT3_FRACTION = 0.5

; Discharge current SoC limits
DC_SOC_LIMIT1 = 10
DC_SOC_LIMIT2 = 20
DC_SOC_LIMIT3 = 30

; Discharge current limits
DC_CURRENT_LIMIT1_FRACTION = 0.1
DC_CURRENT_LIMIT2_FRACTION = 0.3
DC_CURRENT_LIMIT3_FRACTION = 0.5

; --------- Additional settings ---------
; Specify one or more BMS types to load else leave empty to try to load all available
; -- Available BMS:
; Daly, Ecs, HeltecModbus, HLPdataBMS4S, Jkbms, Lifepower, LltJbd, Renogy, Seplos
; -- Available BMS, but disabled by default (just enter one or more below and it will be enabled):
; ANT, MNB, Sinowealth
BMS_TYPE = LltJbd

; Publish the config settings to the dbus path "/Info/Config/"
PUBLISH_CONFIG_VALUES = True

; Battery temperature
; Specify how the battery temperature is assembled
; 0 Get mean of temperature sensor 1 to sensor 4
; 1 Get only temperature from temperature sensor 1
; 2 Get only temperature from temperature sensor 2
; 3 Get only temperature from temperature sensor 3
; 4 Get only temperature from temperature sensor 4
TEMP_BATTERY = 0

; Temperature sensor 1 name
TEMP_1_NAME = BMS Temp

; Temperature sensor 2 name
TEMP_2_NAME = Case Temp 1

; Temperature sensor 2 name
TEMP_3_NAME = Case Temp 2

; Temperature sensor 2 name
TEMP_4_NAME = Temp 4

; Enter custom battery names here or change it over the GUI
; Example:
;     /dev/ttyUSB0:My first battery
;     /dev/ttyUSB0:My first battery,/dev/ttyUSB1:My second battery
CUSTOM_BATTERY_NAMES = /dev/ttyUSB0:SFK-300HP-CS-BAT1,/dev/ttyUSB1:SFK-300HP-RS-BAT2
```

### Relevant log output

```shell
2024-04-02 02:22:54.727852500 INFO:SerialBattery:Serial Number/Unique Identifier: 60_300.0Ah
2024-04-02 02:22:57.544478500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-04-02 02:22:57.878798500 INFO:SerialBattery:DeviceInstance = 1
2024-04-02 02:22:57.880452500 ERROR:SerialBattery:** DRIVER STOPPED! Another battery with the same serial number/unique identifier "60_300.0Ah" found! **
2024-04-02 02:22:57.881078500 ERROR:SerialBattery:Please check that the batteries have unique identifiers.
2024-04-02 02:22:57.881788500 ERROR:SerialBattery:Change the battery capacities to be unique.
2024-04-02 02:22:57.882339500 ERROR:SerialBattery:Example for batteries with 280Ah:
2024-04-02 02:22:57.882868500 ERROR:SerialBattery:- Battery 1: 279 Ah
2024-04-02 02:22:57.883390500 ERROR:SerialBattery:- Battery 2: 280 Ah
2024-04-02 02:22:57.883899500 ERROR:SerialBattery:- Battery 3: 281 Ah
2024-04-02 02:22:57.884405500 ERROR:SerialBattery:This little difference does not matter for the battery.
2024-04-02 02:22:57.885130500 ERROR:SerialBattery:To see which battery already uses this serial number/unique identifier check this file "/var/tmp/dbus-serialbattery_1.pid"
```

### Any other information that may be helpful

Previous driver version did load both batteries.

Thank you for your help!

mr-manuel, 2024-04-02:
Just change the capacity from one battery as the message say.

mr-manuel, 2024-04-02:
You have to change the capacity on the BMS itself. With the app of the BMS or PC software.

metapixl, 2024-04-03:
Appreciate your help. Thanks!

andy-home-vic, 2024-12-01:
Same problem here with 2 jkbms. But i have Not the Same config with "CUSTOM_BATTERY_NAMES" and today the latest Beta. 
Where can i find the installed Version number?
Thanks