Louisvdw/dbus-serialbattery #1034: (JK-BMS) Battery discharges alternatingly when SoC reaches 100 % and no discharge requested

jjdejong, 2024-04-11:
### Describe the bug

After the absorption phase, when the charge current reaches 0 and the SoC reaches 100 %, the battery is discharged from time to time at around 1A even with excess solar production. See curve below:
[image]

When I disable the serial driver and reboot the Victron, things return to normal:
[image]

### How to reproduce

Install the serial driver with Bluetooth connection.
Wait until the battery is fully charged with the excess solar production.
Observe discharge phases of the battery even under excess solar production.

### Expected behavior

No discharge phases when the battery is fully charged and there is excess solar production.

### Driver version

v1.2.20240401

### Venus OS device type

GX Card (integrated in Victron device)

### Venus OS version

v3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:EE:68:33
SOC_RESET_VOLTAGE = 3.550
LINEAR_LIMITATION_ENABLE = False
```

### Relevant log output

```shell
No errors in the logs.
```

### Any other information that may be helpful

The Victron is configured with an absorption voltage of 55.2V.

mr-manuel, 2024-04-11:
Please measure with an ampere meter. My JKBMS is showing discharge when the MultiPlus say there is no discharge. In the low value segment the values are very different between the two and it can also be that one is showing positive values and the other negative values.

jjdejong, 2024-04-11:
We're speaking of values around 1A / 50W, which are not negligible. Same for you then?

kommando828, 2024-04-11:
JK BMS's all need Voltage calibration under no load and also Current calibration under load.

jjdejong, 2024-04-12:
So the BMS is the culprit, not the driver. Not a bug in the driver then.

closed 2024-04-12 by jjdejong