Louisvdw/dbus-serialbattery #1035: No ability to change the unique ID

ytechie, 2024-04-12:
### Describe the bug

I was using an older version (v1.0.20230531) with my 2 Lifepower batteries connected via USB->485. I got occasional errors, so I really want to upgrade to a newer version. However, the new version (I tried all of the latest 1.2 variants) requires unique battery ID's. I don't see a way to change the ID in the Lifepower battery configuration software - I'm not sure it's possible (I'm double checking with them).

`** DRIVER STOPPED! Another battery with the same serial number/unique identifier "199.8Ah" found! **`

### How to reproduce

1. Have 2 or more lifepower batteries connected
2. Use a version newer than v1.0.20230531

### Expected behavior

I would like both batteries to work without an ID conflict.

### Driver version

v1.2.20240404dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.30

### BMS type

Life/Tian Power

### Cell count

8

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

LOGGING = DEBUG

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; LifePower battery specs - from the manual
; 1. Bulk/Absorption: 28.1V (+/- .2V)
; 2. Float: 27V (+/- .2V)
; 3. Low DC cutoff: 23.5-22V* (depending on load - start high, lower if needed)
; 4. Battery Charge Temperature Range: 32°F - 122°F 5. Battery Discharge Temperature Range: -4°F - 131°F
; Discharging current: 100a
; Charging current: 100a max, 30-50 recommended
; Charging range: 32°F to 122°F (0°C to 50°C)
; Discharging range: -4°F to 131°F (-20°C to 55°C)

MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.488
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

MAX_BATTERY_CHARGE_CURRENT = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

BMS_TYPE = Lifepower

SOC_CALCULATION = FALSE

LINEAR_LIMITATION_ENABLE = False

; --------- Temperature limitation (affecting CCL/DCL) ---------
; Description:
;     Maximal charge / discharge current will be in-/decreased depending on temperature
; Example:
;     The temperature limit will be monitored to control the currents. If there are two temperature senors,
;     then the worst case will be calculated and the more secure lower current will be set.
; Charge current control management referring to temperature enable (True/False).
CCCM_T_ENABLE = True
; Charge current control management referring to temperature enable (True/False).
DCCM_T_ENABLE = True

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
TEMPERATURES_WHILE_CHARGING	 =    0,    2,    5,   10,   15,   20,   35,   40,   50
MAX_CHARGE_CURRENT_T_FRACTION    = 0.00, 1.00, 1.00, 1.00, 1.00, 1.00, 1.00, 1.00, 0.00

TEMPERATURES_WHILE_DISCHARGING   =  -20,   -10,    5,   10,   15,   45,   55
MAX_DISCHARGE_CURRENT_T_FRACTION = 0.00, 1.00, 1.00, 1.00, 1.00, 1.00, 0.00
```

### Relevant log output

```shell
2024-04-12 15:06:19.599618500 ERROR:SerialBattery:** DRIVER STOPPED! Another battery with the same serial number/unique identifier "199.8Ah" found! **
2024-04-12 15:06:19.600252500 ERROR:SerialBattery:Please check that the batteries have unique identifiers.
2024-04-12 15:06:19.600923500 ERROR:SerialBattery:Change the battery capacities to be unique.
2024-04-12 15:06:19.601649500 ERROR:SerialBattery:Example for batteries with 280 Ah:
2024-04-12 15:06:19.602291500 ERROR:SerialBattery:- Battery 1: 279 Ah
2024-04-12 15:06:19.602859500 ERROR:SerialBattery:- Battery 2: 280 Ah
2024-04-12 15:06:19.603398500 ERROR:SerialBattery:- Battery 3: 281 Ah
2024-04-12 15:06:19.603923500 ERROR:SerialBattery:This little difference does not matter for the battery.
2024-04-12 15:06:19.604466500 ERROR:SerialBattery:To see which battery already uses this serial number/unique identifier check this file "/var/tmp/dbus-serialbattery_1.pid"
```

mr-manuel, 2024-04-12:
This is not a bug. just read the complete error log. I copy the relevant part over for you:

```
2024-04-12 15:06:19.600923500 ERROR:SerialBattery:Change the battery capacities to be unique.
2024-04-12 15:06:19.601649500 ERROR:SerialBattery:Example for batteries with 280 Ah:
2024-04-12 15:06:19.602291500 ERROR:SerialBattery:- Battery 1: 279 Ah
2024-04-12 15:06:19.602859500 ERROR:SerialBattery:- Battery 2: 280 Ah
2024-04-12 15:06:19.603398500 ERROR:SerialBattery:- Battery 3: 281 Ah
2024-04-12 15:06:19.603923500 ERROR:SerialBattery:This little difference does not matter for the battery.
```

ytechie, 2024-04-12:
Fair enough - it's asking me to do something that I believe may be impossible.

mr-manuel, 2024-04-12:
On every BMS you need to be able to change the capacity. Is it custom build or did you buy it as it is?
a) Custom build: You already had to set the capacity. Just set it again
b) You bought it: Ask your dealer on how you can change the capacity or if you get the protocol documentation because you need to read the serial number via serial connection.

ytechie, 2024-04-12:
I bought them - they are EG4 Lifepower batteries. I am checking with them to confirm, but even their own software doesn't read the serial number nor give a way to change the capacity. 
[image]

mr-manuel, 2024-04-12:
Let's wait what they answer. Maybe there is another unique parameter we can read.

mr-manuel, 2024-04-12:
Temporarely you could try to add 

```python
    def unique_identifier(self) -> str:
        return self.port
```

before https://github.com/Louisvdw/dbus-serialbattery/blob/59a6d00c096c3e708aaf9cd827275c6500450374/etc/dbus-serialbattery/bms/lifepower.py#L47

Keep attention at the indentation when copying the code.

Anyway this is only a possible temporary solution and not recommended.

ytechie, 2024-04-12:
Oh, that's genius. I'll give it a try. That would at least get me on the latest code while I figure out a better long-term solution.

ytechie, 2024-04-16:
No response from them - it's kind of expected since they don't have a published spec for reading from their batteries in this way.

Utilizing the port as a unique ID has been working perfectly. It allowed me to use the latest version and now I have clean logs.

mr-manuel, 2024-04-17:
Ok, only when changing ports it will also change custom names and VRM ID's.

mr-manuel, 2024-04-17:
With the latest nightly you can set this in the config file.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/1cc775d29cd8a7431acdd4e45584187693edf35b/etc/dbus-serialbattery/config.default.ini#L386-L390

closed 2024-04-17 by mr-manuel

andy-home-vic, 2024-12-01:
I Set USE_PORT_AS_UNIQUE_ID =True but my 2jkbms don't Work. IT Shows only one in Venus