Louisvdw/dbus-serialbattery #1037: Crash with LINEAR_LIMITATION_ENABLE = False

mdkeil, 2024-04-15:
### Describe the bug

I often switched between LINEAR_LIMITATION_ENABLE = True / False for testing. I don't know the reason, but since today afternoon, I cant run the driver with LINEAR_LIMITATION_ENABLE = False

### How to reproduce

Run the driver with LINEAR_LIMITATION_ENABLE = False

### Expected behavior

NTR

### Driver version

v1.2.20240408

### Venus OS device type

Raspberry Pi 2

### Venus OS version

3.30

### BMS type

Seplos

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 100
MAX_BATTERY_DISCHARGE_CURRENT = 100                                                
MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

VOLTAGE_DROP = 0.020

SOC_RESET_VOLTAGE = 3.500
#SOC_RESET_AFTER_DAYS = 15                                                         
CVL_ICONTROLLER_MODE = True
LINEAR_LIMITATION_ENABLE = True                                                    CVCM_ENABLE = True

CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.015
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.010
MAX_VOLTAGE_TIME_SEC = 2100                                                        
BMS_TYPE = Seplos
; Exclute this serial devices from the driver startup
; Example: /dev/ttyUSB2, /dev/ttyUSB4                                              EXCLUDED_DEVICES = /dev/ttyUSB0

CCCM_SOC_ENABLE = True                                                             DCCM_SOC_ENABLE = True
SOC_WHILE_CHARGING                 =  100,   98,   95,   90,   85
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.05, 0.15, 0.30, 0.50,   1
CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.45, 3.40, 3.30                      MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.05, 0.10, 0.30, 1                         CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.85, 3.05,  3.30
```

### Relevant log output

```shell
root@raspberrypi2:~# tail -n 500 /data/log/dbus-serialbattery.ttyUSB1/current | tai64nlocal
2024-04-15 18:10:19.412765500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-04-15 18:10:19.412777500 INFO:SerialBattery:> CVCM:     True
2024-04-15 18:10:19.412781500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True2024-04-15 18:10:19.412785500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True2024-04-15 18:10:19.412789500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True2024-04-15 18:10:19.412793500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-04-15 18:10:26.620998500 Traceback (most recent call last):
2024-04-15 18:10:26.621009500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 811, in publish_battery
2024-04-15 18:10:26.621015500     self.battery.manage_charge_voltage()
2024-04-15 18:10:26.621018500   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 241, in manage_charge_voltage
2024-04-15 18:10:26.621025500     self.manage_charge_voltage_step()
2024-04-15 18:10:26.621028500   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 734, in manage_charge_voltage_step
2024-04-15 18:10:26.621203500     if not self.charge_mode.startswith("Float"):
2024-04-15 18:10:26.621208500 AttributeError: 'NoneType' object has no attribute 'startswith'
```

mr-manuel, 2024-04-16:
Nice catch! Please install v1.3.20240416dev, there I fixed it.

mdkeil, 2024-04-16:
Thxs for the quick fix.. initial reinstall (v1.3.20240416dev) with `LINEAR_LIMITATION_ENABLE = False` and the driver doesnt crashed.. I think there only some minor error messages.

```
2024-04-16 06:36:52.075938500 INFO:SerialBattery:
2024-04-16 06:36:52.076731500 INFO:SerialBattery:Starting dbus-serialbattery
2024-04-16 06:36:52.079360500 INFO:SerialBattery:Venus OS v3.30
2024-04-16 06:36:52.080209500 INFO:SerialBattery:dbus-serialbattery v1.3.20240416dev
2024-04-16 06:37:08.086694500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-04-16 06:37:08.096163500 INFO:SerialBattery:Testing Seplos
2024-04-16 06:37:09.216236500 INFO:SerialBattery:Connection established to Seplos
2024-04-16 06:37:09.848264500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-04-16 06:37:10.307074500 INFO:SerialBattery:DeviceInstance = 1
2024-04-16 06:37:10.309381500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_[1.pid](http://1.pid/)
2024-04-16 06:37:10.310078500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-04-16 06:37:10.310902500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-04-16 06:37:10.351903500 INFO:SerialBattery:publish config values = True
2024-04-16 06:37:10.386090500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-04-16 06:37:10.387180500 INFO:SerialBattery:========== Settings ==========
2024-04-16 06:37:10.387801500 INFO:SerialBattery:> Connection voltage: 53.25 V | Current: 14.11 A | SoC: None%
2024-04-16 06:37:10.388485500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-04-16 06:37:10.389164500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2024-04-16 06:37:10.389887500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9 V | MAX CELL VOLTAGE: 3.45 V
2024-04-16 06:37:10.390596500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A
2024-04-16 06:37:10.391555500 INFO:SerialBattery:> CVCM:     True
2024-04-16 06:37:10.392178500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True2024-04-16 06:37:10.393005500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True2024-04-16 06:37:10.393697500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True2024-04-16 06:37:10.394402500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-04-16 06:37:10.395282500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-04-16 06:37:10.480210500 INFO:SerialBattery:Monitoring external current using: None/None
2024-04-16 06:37:10.481245500 ERROR:SerialBattery:Exception occurred: AttributeError("'NoneType' object has no attribute 'split'") of type <class 'AttributeError'> in /opt/victronenergy/dbus-serialbattery/[dbus-serialbattery.py](http://dbus-serialbattery.py/) line #253
2024-04-16 06:37:15.460476500 Traceback (most recent call last):
2024-04-16 06:37:15.460487500   File "/usr/lib/python3.8/site-packages/serial/[serialposix.py](http://serialposix.py/)", line 500, in read
2024-04-16 06:37:15.460493500     raise SerialException(
2024-04-16 06:37:15.460495500 serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-04-16 06:37:15.460503500
2024-04-16 06:37:15.460505500 During handling of the above exception, another exception occurred:
2024-04-16 06:37:15.460510500
2024-04-16 06:37:15.460731500 Traceback (most recent call last):
2024-04-16 06:37:15.460735500   File "/opt/victronenergy/dbus-serialbattery/[dbushelper.py](http://dbushelper.py/)", line 760, in publish_battery
2024-04-16 06:37:15.460741500     result = self.battery.refresh_data()
2024-04-16 06:37:15.460745500   File "/opt/victronenergy/dbus-serialbattery/bms/[seplos.py](http://seplos.py/)", line 127, in refresh_data
2024-04-16 06:37:15.460751500     result_status = [self.read](http://self.read/)_status_data()
2024-04-16 06:37:15.460754500   File "/opt/victronenergy/dbus-serialbattery/bms/[seplos.py](http://seplos.py/)", line 209, in read_status_data
2024-04-16 06:37:15.460899500     data = [self.read](http://self.read/)_serial_data_seplos(
2024-04-16 06:37:15.460904500   File "/opt/victronenergy/dbus-serialbattery/bms/[seplos.py](http://seplos.py/)", line 310, in read_serial_data_seplos
2024-04-16 06:37:15.460911500     data = ser.readline()
2024-04-16 06:37:15.460914500   File "/usr/lib/python3.8/site-packages/serial/[serialposix.py](http://serialposix.py/)", line 509, in read
2024-04-16 06:37:15.460919500     raise SerialException('read failed: {}'.format(e))
2024-04-16 06:37:15.461001500 serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-04-16 06:37:17.787361500 INFO:SerialBattery:
2024-04-16 06:37:17.788267500 INFO:SerialBattery:Starting dbus-serialbattery
2024-04-16 06:37:17.790790500 INFO:SerialBattery:Venus OS v3.30
2024-04-16 06:37:17.791507500 INFO:SerialBattery:dbus-serialbattery v1.3.20240416dev
2024-04-16 06:37:33.807530500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-04-16 06:37:33.808374500 INFO:SerialBattery:Testing Seplos
2024-04-16 06:37:34.941306500 INFO:SerialBattery:Connection established to Seplos
2024-04-16 06:37:35.499266500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-04-16 06:37:35.773050500 INFO:SerialBattery:DeviceInstance = 1
[... cut ...]  None | BALANCE FET: None
2024-04-16 06:37:35.868585500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-04-16 06:37:35.961107500 INFO:SerialBattery:Monitoring external current using: None/None
2024-04-16 06:37:35.962483500 ERROR:SerialBattery:Exception occurred: AttributeError("'NoneType' object has no attribute 'split'") of type <class 'AttributeError'> in /opt/victronenergy/dbus-serialbattery/[dbus-serialbattery.py](http://dbus-serialbattery.py/) line #253
```

mr-manuel, 2024-04-16:
I fixed that also. Please reinstall the same version.

mdkeil, 2024-04-16:
check!

closed 2024-04-16 by mr-manuel