Louisvdw/dbus-serialbattery #1038: Calculation of CVL_ICONTROLLER has got corrupted.

cflenker, 2024-04-18:
### Describe the bug

Hello @mr-manuel,
Yesterday I installed the latest release of the driver on my friends system and the CVL_ICONTROLLER did not work correctly.
I am afraid the commit https://github.com/Louisvdw/dbus-serialbattery/commit/963534469352802c8b608b54d66e93c1d7e7c5f1 corrupted the CVL_ICONTROLLER.
You renamed `controlvoltage` to `control_voltage`.
But `controlvoltage` kept the result of the ICONTROLLER whereas `control_voltage` kept the result of the penalty-sum-algorithm.
Using the same variable for both leads to overwriting the ICONTROLLER-result with the penaltysum-Result.
It would be nice if you could fix that.
Thank's a lot! Christian

### How to reproduce

Reach the MAX_CELL_VOLTAGE  to activate the CVL_CONTROLLER

### Expected behavior

CVL_ICONTROLLER working in a smooth way

### Driver version

1.2.20240408

### Venus OS device type

Cerbo GX

### Venus OS version

3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
; CF: MAX_CELL_VOLTAGE auf 3.55 gesetzt, damit penalty nicht zu frueh greift
;     und die CVL runterholt was zu einem starken toggeln führt
[... 112 lines cut ...]
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300

; CUSTOM_BATTERY_NAMES = /dev/ttyUSB0:JKBMS1_Flenker,/dev/ttyUSB1:JKBMS2_Flenker

; --------- Battery monitor specific settings ---------
; If you are using a SmartShunt or something else as a battery monitor, the battery voltage reported
; from the BMS and SmartShunt could differ. This causes, that the driver never goapplies the float voltage,
; since max voltage is never reached.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 and the battery is now charged until the SmartShunt measures 55.20 V. The BMS
;     now measures 55.05 V since there is a voltage drop of 0.15 V. Since the dbus-serialbattery measures
;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
;     Set VOLTAGE_DROP to 0.15
VOLTAGE_DROP = 0.05
```

### Relevant log output

```shell
not relevant
```

cflenker, 2024-04-18:
Ohhh, it seems to be fixed on master of https://github.com/mr-manuel/venus-os_dbus-serialbattery.
I just updated the system to `1.3.20240417dev`. Let's see if this works properly.

mr-manuel, 2024-04-19:
Just let me know, there was a bit a mess which I cleaned up.

mr-manuel, 2024-04-30:
@cflenker any updates to this?

mr-manuel, 2024-05-10:
Fixed with `v1.3.20240417dev`.

closed 2024-05-10 by mr-manuel