Louisvdw/dbus-serialbattery #1043: driver no longer works

ray90408, 2024-04-22:
### Describe the problem

I changed the config.ini while the system was running and rebooted. The battery (JKBMS has disappeared from the Venus GUI. A reinstallation was unsuccessful. How can you really delete everything about the serial battery driver without having to reinstall the Raspberry? The uninstall routine is not enough. I have tried this several times.

Who can help a beginner?

### Driver version

v1.2.20240401 

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.31

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
;https://github.com/Louisvdw/dbus-serialbattery/blob/master/etc/dbus-serialbattery/config.default.ini

MAX_BATTERY_CHARGE_CURRENT = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 90.0

MIN_CELL_VOLTAGE   = 3.000
;MAX_CELL_VOLTAGE   = 3.450
MAX_CELL_VOLTAGE   = 3.460

FLOAT_CELL_VOLTAGE = 3.375

SOC_RESET_VOLTAGE = 3.5

CELL_VOLTAGES_WHILE_CHARGING   = 3.50, 3.45, 3.40, 3.35
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.05,  0.5,    1

;CELL_VOLTAGES_WHILE_DISCHARGING   = 3.00, 3.30, 3.15, 3.20
;MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1

CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False

BMS_TYPE = Jkbms

;false = Float aus für Kalibration_SOC BMS-Einstellungen: OVPR 3,44V  OVP 3,451V    MAX_CELL_VOLTAGE   = 3.459
CVCM_ENABLE = false
```

### Relevant log output

```shell
root@raspberrypi4:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
[... 125 lines cut ...]
2024-04-22 16:36:19.655734500 Traceback (most recent call last):
2024-04-22 16:36:19.655739500   File "/opt/victronenergy/dbus-serialbattery/dbus                                 -serialbattery.py", line 20, in <module>
2024-04-22 16:36:19.655743500     from dbushelper import DbusHelper
2024-04-22 16:36:19.655746500   File "/opt/victronenergy/dbus-serialbattery/dbus                                 helper.py", line 8, in <module>
2024-04-22 16:36:19.655749500     from utils import logger, publish_config_varia                                 bles
2024-04-22 16:36:19.655752500   File "/opt/victronenergy/dbus-serialbattery/util                                 s.py", line 24, in <module>
2024-04-22 16:36:19.655837500     config.read([default_config_file_path, custom_                                 config_file_path])
2024-04-22 16:36:19.655841500   File "/usr/lib/python3.8/configparser.py", line                                  697, in read
2024-04-22 16:36:19.655844500     self._read(fp, filename)
2024-04-22 16:36:19.655847500   File "/usr/lib/python3.8/configparser.py", line                                  1017, in _read
2024-04-22 16:36:19.655850500     for lineno, line in enumerate(fp, start=1):
2024-04-22 16:36:19.655853500   File "/usr/lib/python3.8/codecs.py", line 322, i                                 n decode
2024-04-22 16:36:19.655881500     (result, consumed) = self._buffer_decode(data,                                  self.errors, final)
2024-04-22 16:36:19.655884500 UnicodeDecodeError: 'utf-8' codec can't decode byt                                 e 0xfc in position 717: invalid start byte
2024-04-22 16:39:03.353481500 *** CCGX booted (0) ***
2024-04-22 16:40:51.760808500 *** CCGX booted (0) ***
2024-04-22 17:04:53.378607500 *** CCGX booted (0) ***
2024-04-22 17:14:19.383121500 *** CCGX booted (0) ***
2024-04-22 17:15:19.118780500 INFO:SerialBattery:
2024-04-22 17:15:19.119005500 INFO:SerialBattery:Starting dbus-serialbattery
2024-04-22 17:15:19.119938500 INFO:SerialBattery:Venus OS v3.31
2024-04-22 17:15:19.120210500 INFO:SerialBattery:dbus-serialbattery v1.2.2024040                                 

root@raspberrypi4:~# tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
```

mr-manuel, 2024-04-22:
You corrupted your config.ini file by editing it. Probably you did that by downloading and uploading it with sftp. Delete the config.ini and reinstall the driver.

ray90408, 2024-04-23:
Thank you very much for the quick help. The driver is running again.

closed 2024-04-23 by ray90408