Louisvdw/dbus-serialbattery #1045: New JK-BMS JK-B2A8S30P not yet supported?

Lucifer06, 2024-04-26:
### Describe the problem

New JK-BMS JK-B2A8S30P seems not recognised with serial battery. Tried several versions 1.2 and beta 1.3 but no joy.
On the same Cerbo GW I have 2x B2A8S20P with no issue.
the new B2A8S30P is the 300A version for 4S and 8S. It is running the same FW version, according to the iOS app.
From the iOS App, I see that the Device ID is by default = 1, and can't be change to 0 on the legacy B2A8S20P, this is set by default to 0
Also I see that I have few choices for the protocol on this new B2A8S30P. I tried few options, but no joy.
Any hint? I'm not sure providing the USB log will be relevant as I don't see anything interesting appart that trying to recognise the BMS 3x times and then declare there is no BMS.
Also I noticed that it seems to hang the BMS part after some period because I don't see anymore the red led of the BMS blinking, and it becomes impossible to switch off the BMS with the power switch, unless I disconnect the USB interface.

### Driver version

v1.2 & 1.3

### Venus OS device type

Cerbo GX

### Venus OS version

v3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

4

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

; --------- Battery Current limits ---------  
MAX_BATTERY_CHARGE_CURRENT = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0
```

### Relevant log output

```shell
@400000006629508d30cb9d84 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@400000006629508d30d6ad8c INFO:SerialBattery:Testing Jkbms
@400000006629508e06d74a14 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006629508e25f39574 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
@40000000662950a03916d47c INFO:SerialBattery:
@40000000662950a03923385c INFO:SerialBattery:Starting dbus-serialbattery
@40000000662950a0393bfc34 INFO:SerialBattery:dbus-serialbattery v1.0.20231125beta
@40000000662950d10e515244 *** CCGX booted (0) ***
```

Lucifer06, 2024-04-26:
Tomorrow, I will try to connect this BMS via bluetooth. I believe it should work...

mr-manuel, 2024-04-27:
Ask your reseller for the protocol documentation, this should help me to get it fixed.

kommando828, 2024-04-27:
This is the RS485 protocol for the new JK JK-PB1A16S10P  JK-PB1A16S15P  JK-PB2A16S20P etc

[Jikong.BMS.RS485.Modbus.universal.protocol.V1.0.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/15137111/Jikong.BMS.RS485.Modbus.universal.protocol.V1.0.pdf)

Lucifer06, 2024-04-27:
I made the request with the vendor to receive updated protocol for this model.
I tried a second BMS unit, and this is the same. I also tried to change the UART1 protocol (by default it was "000 - 4G-SPS Remote module communication") with "001 - JKBMS RS485 Modbus v1.0".
There is also choice of:
"002 - NIU U SERIES"
"003 - China tower shared battery cabin..."
"006 - Growatt_BMS_RS485_Protocol"
but no luck still the same results.

See the logs with version 1.2.xxx and 1.3.xx:
`@40000000662d83c53567e1f4 INFO:SerialBattery:Starting dbus-serialbattery
@40000000662d83c53593c454 INFO:SerialBattery:Venus OS v3.40~8
@40000000662d83c535a62b44 INFO:SerialBattery:dbus-serialbattery v1.2.20240408
@40000000662d83d536afc374 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@40000000662d83d536baa884 INFO:SerialBattery:Testing Jkbms
@40000000662d83d60c7fab3c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d83d62a8ca1ac INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@40000000662d83d62a971574 INFO:SerialBattery:Testing Jkbms
@40000000662d83d70037debc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d83d71e4b8fd4 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@40000000662d83d71e588824 INFO:SerialBattery:Testing Jkbms
@40000000662d83d72f685c34 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d83d811ec95bc ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1
@40000000662d83e709aebca4 INFO:SerialBattery:
@40000000662d83e709bb999c INFO:SerialBattery:Starting dbus-serialbattery
@40000000662d83e709ddd354 INFO:SerialBattery:Venus OS v3.40~8
@40000000662d83e709e78b9c INFO:SerialBattery:dbus-serialbattery v1.3.20240427dev
@40000000662d83f70aef761c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@40000000662d83f70afec414 INFO:SerialBattery:Testing Jkbms
@40000000662d83f71cc906b4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d83f73af2dbc4 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@40000000662d83f73af2feec INFO:SerialBattery:Testing Jkbms
@40000000662d83f813fa47dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d83f832173054 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@40000000662d83f832231f04 INFO:SerialBattery:Testing Jkbms
@40000000662d83f909104a24 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d83f926fac85c ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1
@40000000662d83fb14afff64 INFO:SerialBattery:
@40000000662d83fb14ba638c INFO:SerialBattery:Starting dbus-serialbattery
@40000000662d83fb14d8c8cc INFO:SerialBattery:Venus OS v3.40~8
@40000000662d83fb14e301fc INFO:SerialBattery:dbus-serialbattery v1.3.20240427dev
@40000000662d840b15e2f56c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@40000000662d840b15f11e6c INFO:SerialBattery:Testing Jkbms
@40000000662d840b27bd3d9c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d840c0a6ea58c INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@40000000662d840c0a7cc2d4 INFO:SerialBattery:Testing Jkbms
@40000000662d840c1c34ae24 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d840c3a4e46f4 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@40000000662d840c3a5904f4 INFO:SerialBattery:Testing Jkbms
@40000000662d840d100d0434 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662d840d2e2a5624 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1
`

Lucifer06, 2024-04-27:
@kommando828 I looked at your doc, and effectively, I have this protocol available to be set on the UART1, with the IOS app.
@mr-manuel I suppose this driver has not yet been implemented in dbus-serialbattery ?

kommando828, 2024-04-28:
JK Pinouts are reversed to normal, maybe try reversing A and B, reversing Can_L and Can_H worked for me for  JK-PB2A16S20P direct BMS Canbus to inverter.
[image]

Lucifer06, 2024-04-28:
The BMS is not this form factor and the RD485 adapter (provided by JK) is connecting on the GPS port.
according to Google translate A is yellow and B is white. I also used the same adapter from the working B2A8S20P, and still can't get the BMS recognised.
I'm afraid the GPS port is not anymore supporting the same protocol as the B2A8S20P. 
I have my test lab setup, I will try to attach the 485 interface to a computer and see if we can't get something out of this new 300A BMS...

Lucifer06, 2024-04-28:
Here is what it looks like:
[image]

kommando828, 2024-04-28:
Assuming the two left most connectors are to the temperature sensors, have you tried connecting to the CAN/RS485 port to the right of the GPS. Just making odd suggestions as the documentation of JK is so bad you have to experiment to get anywhere. If this CAN/RS485 port is the same as on the JK-PB2A then you only need to RS485 USB on the PC and not the JK RS485 box as it is already RS485 and not TTL.

Lucifer06, 2024-04-28:
I tried with Windows app JK-BMS-MONITOR 2.7.0, BMS configured as Dev Address 1 and UART protocol 001 - JKBMS RS485 Modbus v1.0
No communication :-(
I'm talking with JK BMS vendor, but I'm afraid it will go nowhere...

kommando828, 2024-04-28:
You would assume that the very latest released BMS would follow the other most recent releases, but in JK world it seems every release they start from the beginning and change everything. Feature wise they are the best BMS but support wise they leave a lot to be desired. Seems the only customer they listen to is Andy at Off Grid Garage.

Lucifer06, 2024-04-29:
@kommando828 good advice, I will try to connect to the RS485 connector, next to the GPS one.
Does some one know what is the name of this connector, so I can source it?

kommando828, 2024-04-29:
Your photo is private so I can't blow the photo up to see the connector. Could you post a closeup of the connector, the various sellers listings are silent on what type it is or any closeup's.

Lucifer06, 2024-04-29:
here you are:
[image]

kommando828, 2024-04-30:
Yes now that I am awake ;)

What a weird connector.

kommando828, 2024-04-30:
Looks like a JST GH type

https://www.mattmillman.com/info/crimpconnectors/common-jst-connector-types/

Lucifer06, 2024-05-01:
[image]

[image]

Lucifer06, 2024-05-01:
Looking at the picture, I think the SMD drivers are not soldered. and I think the option of Can or 485 is a choice of with one they solder.
Game over!

Lucifer06, 2024-05-01:
On the top left of the picture, I think this is a MAX 485 that is supposed to fit there.
If I find the time I will try to fit one here and check if I get some life over 485 port....

kommando828, 2024-05-01:
If a MAX485 should go there then you should find TTL signals on the correct pins that feed the signal to the MAX485.

Lucifer06, 2024-05-07:
JK gave me access to the following Google drive: [https://drive.google.com/drive/folders/1RZ7IP5g2uS3PzJo2RcMnLJw6343AhlqI?spm=a2g0s.imconversation.0.0.7ddc3e5fEeTB9L](url)

I'm attaching the spec sheet for the JK-B2A8S-10P-20P-30P series. Both original, and a tentative of English translation. If you guys have better tool for chinese->english, feel free to share.

[JK-B2A8S-10P-20P-30P-EN-V1.0.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/15233266/JK-B2A8S-10P-20P-30P-EN-V1.0.pdf)
[Uploading JK-B2A8S-10P-20P-30P主动均衡保护板使用说明书-V1.0.pdf…]()

Lucifer06, 2024-05-07:
My discussion with the store is going in a dead loop.
I'm afraid the conclusion is that the GPS port is deactivated, despite they are selling their GPS->RS485 adapter as a bundle with the BMS...

mr-manuel, 2024-06-21:
Can you write me on Discord? Maybe I have an idea now. Did you already measured, if you get some signals from the GPS port? It already happend that there where some broken GPS ports out there.

Lucifer06, 2024-06-21:
Hi Louis, not sure how I can reach you on Discord, but I can tell you that I don't believe this is because my GPS port is broken, because I have 2x units behaving exactly the same.
I'm still discussing with JK BMS store, and according to them, they have identified that the GPS to RS485 adapter thet was sent with the units were not the required low voltage models.
I have serious doubt about this as the 200A version of the same BMS works perfectly with the same adapters. Anyways they are sending 2x new Low voltage adapters and will give a try, who knows, may be something changed on their GPS port....

pos-ei-don, 2024-06-21:
> Hi Louis, not sure how I can reach you on Discord, but I can tell you that I don't believe this is because my GPS port is broken, because I have 2x units behaving exactly the same. I'm still [...]

JKBMS has two different serial Devices. you need to be shure to have the correct one, belive me.

Lucifer06, 2024-06-29:
OK after 2x months of fight with JKBMS, they have spotted that the RS485 adapters they sent bundled with the BMS were the wrong ones, and I'm happy to report here that latest v1.3 dbus-serialbattery works like a charm with the new 300A JK-B2A8S30P. IT was not so easy to spot the root cause because I tried the RS485 adapters that came with 200A version (JK-B2A8S30P) and it didn't worked. So most probably, the electronic must be more sensible to this low voltage RS485 requirement....

closed 2024-06-29 by Lucifer06

pater79, 2025-08-19:
> OK after 2x months of fight with JKBMS, they have spotted that the RS485 adapters they sent bundled with the BMS were the wrong ones, and I'm happy to report here that latest v1.3 dbus-serialbattery [...]

Hi, I have a JK-B2A8S30P and the RS485 cable (I'll have to find out if it's the valid one or not). I want to communicate with the Cerbo GX and at the same time with Home Assistant. I'm new to this and it's not clear to me what software I have to put in, where can I find this version? "1.3 of dbus-serialbattery" I've been searching on GitHub and I can't find it. I'm new to this, could you help me and tell me the link where the file is to download, and how to introduce it to the Cerbo. I have formatted the pen drive in fat32. Thank you very much, Best regards and thanks for the great contribution and time you dedicate. Best regards