Louisvdw/dbus-serialbattery #1046: Random min/max cell voltages go very high or very low in a single read.

cupertinomiranda, 2024-04-28:
### Describe the bug

I noticed that sometimes dbus-serialbattery reports very low or high single cell voltage, affecting SOC calculation and general state of the system.
To diagnose the problem I made node-red in venus os to report as a sensor in home assistant, every single read from cell voltages.
I could then identify that the reported low/high value occurred random and in a single reading. It occurs randomly, usually once a day.

### How to reproduce

Wait!

### Expected behavior

Not having random reads from cell voltages.

### Driver version

v1.2.20240408

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = DEBUG
#LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
; Description: May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
[... 340 lines cut ...]
; factory default address will be 1
HELTEC_MODBUS_ADDR = 1

; --------- Voltage drop ---------
; If you have a voltage drop between the BMS and the charger because of wire size or length
; then you can specify the voltage drop here. The driver will then add the voltage drop
; to the calculated CVL to compensate.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 V and the battery is now charged until the charger reaches 55.20 V.
;     The BMS now measures 55.05 V since there is a voltage drop of 0.15 V on the cable.
;     Since the dbus-serialbattery reads the voltage of 55.05 V from the BMS the max voltage
;     of 55.20 V is never reached and max voltage is kept forever.
;     By setting the VOLTAGE_DROP to 0.15 V the voltage on the charger is increased and the
;     target voltage on the BMS is reached.
VOLTAGE_DROP = 0.00
```

### Relevant log output

```shell
Nothing particular
```

cupertinomiranda, 2024-04-28:
For completeness, I was able to solve the issue with the following patch.
Please noticed I did not generalized the problem for other BMS.

I wonder if it occurs in other platforms as well and the solution could be generalized.

```
diff --git a/etc/dbus-serialbattery/bms/jkbms.py b/etc/dbus-serialbattery/bms/jkbms.py
index da27924..68b4a33 100644
--- a/etc/dbus-serialbattery/bms/jkbms.py
+++ b/etc/dbus-serialbattery/bms/jkbms.py
@@ -12,6 +12,7 @@ class Jkbms(Battery):
         super(Jkbms, self).__init__(port, baud, address)
         self.type = self.BATTERYTYPE
         self.unique_identifier_tmp = ""
+        self.read_error_count = 0

     BATTERYTYPE = "Jkbms"
     LENGTH_CHECK = 1
@@ -94,10 +95,21 @@ class Jkbms(Battery):
         if cellbyte_count == 3 * self.cell_count and self.cell_count == len(self.cells):
             offset = 1
             celldata = self.get_data(status_data, b"\x79", offset, 1 + cellbyte_count)
+            error = False
             for c in range(self.cell_count):
-                self.cells[c].voltage = (
-                    unpack_from(">xH", celldata, c * 3 + 1)[0] / 1000
+                min_value = self.cells[c].voltage * 0.9
+                max_value = self.cells[c].voltage * 1.1
+                value = (
+                        unpack_from(">xH", celldata, c * 3 + 1)[0] / 1000
                 )
+                if((value < min_value or value > max_value) and self.read_error_count < 5):
+                    error = True
+                else:
+                    self.cells[c].voltage = value
+            if error:
+                self.read_error_count += 1
+            else:
+                self.read_error_count = 0

         # MOSFET temperature
         offset = cellbyte_count + 3
@@ -223,7 +235,13 @@ class Jkbms(Battery):
                     self.cells[c].balance = False

         # logger.info(self.hardware_version)
-        return True
+        if (self.read_error_count >= 5):
+            self.read_error_count = 0
+
+        if (self.read_error_count == 0):
+            return True
+        else:
+            return False

     def unique_identifier(self) -> str:
         """

```

mr-manuel, 2024-04-28:
You state that your JKBMS is connected via Bluetooth, but you edited the serial JKBMS driver? That makes no sense to me.

cupertinomiranda, 2024-04-28:
Your right! The funny is that the issue does not show in days, while it was happening every day. :-)
Strange !

cupertinomiranda, 2024-04-29:
Removed my patch, thinking that there might be some fix in the master branch, that made the issue not being reproduced for 3 days.
Well today it happened:
[image]
[image]

Will prepare a new patch to fix this in the proper driver.

mr-manuel, 2024-04-29:
Please use my branch, to keep things simple.

mr-manuel, 2024-04-29:
I hope you tested, if the error exists also in the latest nightly available.

cupertinomiranda, 2024-05-12:
That particular patch is no good. I am now very sure there is a race between the collection of the data and the submission to dbus.
I know this, both from analyzing the code and also the fact that it usually affects 2 values. Also, I had it happen in current and temperature readings simultaneously, which is suspicious.

I have a new patch, but has reported a suspicious min cell voltage. I cannot really know for sure if it was an actual reading from the BMS or a corruption. I will run it for a week to make sure it does not persist. I will create a pull request, once done.

I include the patch for reference. The patch seems big, but it is not.
I create a new thread for decoding and a list to which decoding data can be collected from the thread.
Also I create a mutex to disable the decoder from running in parallel with dbus data submission.
The decoding thread is needed since I think the bluetooth conection is very sensible to timeout. I could not lock the decoding in the same thread as bluetooth data collection, without constantly having bluetooth disconnections.

```
diff --git a/etc/dbus-serialbattery/bms/jkbms_ble.py b/etc/dbus-serialbattery/bms/jkbms_ble.py
index d9f049a..9a4b570 100644
--- a/etc/dbus-serialbattery/bms/jkbms_ble.py
+++ b/etc/dbus-serialbattery/bms/jkbms_ble.py
@@ -7,6 +7,7 @@ from time import sleep, time
 from bms.jkbms_brn import Jkbms_Brn
 import os
 import sys
+import threading
 
 # from bleak import BleakScanner, BleakError
 # import asyncio
@@ -148,114 +149,116 @@ class Jkbms_Ble(Battery):
 
         # result = self.read_soc_data()
         # TODO: check for errors
-        st = self.jk.get_status()
-        if st is None:
-            return False
-
-        last_update = int(time() - st["last_update"])
-        if last_update >= 15 and last_update % 15 == 0:
-            logger.info(
-                f"Jkbms_Ble: Bluetooth connection interrupted. Got no fresh data since {last_update}s."
-            )
-            # show Bluetooth signal strength (RSSI)
-            bluetoothctl_info = os.popen(
-                "bluetoothctl info "
-                + self.address
-                + ' | grep -i -E "device|name|alias|pair|trusted|blocked|connected|rssi|power"'
-            )
-            logger.info(bluetoothctl_info.read())
-            bluetoothctl_info.close()
-
-            # if the thread is still alive but data too old there is something
-            # wrong with the bt-connection; restart whole stack
-            if not self.resetting and last_update >= 60:
-                logger.error(
-                    "Jkbms_Ble: Bluetooth died. Restarting Bluetooth system driver."
+        with self.jk.mutex:
+            st = self.jk.get_status()
+
+            if st is None:
+                return False
+
+            last_update = int(time() - st["last_update"])
+            if last_update >= 15 and last_update % 15 == 0:
+                logger.info(
+                    f"Jkbms_Ble: Bluetooth connection interrupted. Got no fresh data since {last_update}s."
                 )
-                self.reset_bluetooth()
-                sleep(2)
-                self.jk.start_scraping()
-                sleep(2)
+                # show Bluetooth signal strength (RSSI)
+                bluetoothctl_info = os.popen(
+                    "bluetoothctl info "
+                    + self.address
+                    + ' | grep -i -E "device|name|alias|pair|trusted|blocked|connected|rssi|power"'
[... 263 lines cut ...]
                 self.run = False
                 if client.is_connected:
                     try:
+                        logger.debug("client.disconnect()")
                         await client.disconnect()
                     except Exception:
                         (
@@ -557,6 +572,15 @@ class Jkbms_Brn:
 
         logger.info("--> asy_connect_and_scrape(): Exit")
 
+    def decoding(self):
+        while(True):
+            with self.mutex:
+                if len(self.frames) > 0:
+                    logger.debug ("decoding frame")
+                    self.decoding_frame = self.frames.pop()
+                    self.decode();
+            asyncio.sleep(0.1)
+
     def monitor_scraping(self):
         while self.should_be_scraping is True:
             self.bt_thread = threading.Thread(

```

mr-manuel, 2024-05-20:
Please open a PR for it.

cupertinomiranda, 2024-05-23:
I will, however I still see the random values showing up.
I am presuming the corruption actually happens in the device communication, however I we cannot control that I presume.
Did you ever seen this type of behavior before in other devices ?
Is it just mine?
The patch above might still fix a concurrency issue, but did not fixed my problem. :-(

Will change the issue name to better describe the problem.

mr-manuel, 2024-05-23:
I'm not using the BLE connection and most of the users don't track the values in Grafana. So I do not saw this behavior.

howetech, 2024-05-23:
I'm also seeing this behaviour monitoring 2x JKBMS via bluetooth.  Every day or 2 a single cell will report an extremely high value for a single reading and then return to normal.  This is easy to see in Grafana, but it also triggers a cell inbalance alarm in the VRM portal.  

I'm running the latest stable release, but can try the nightly if you think that will help?

I have 2 other systems with different configurations which don't have this issue.  1 is monitoring a JKBMS via serial / RS485 and the other a JBD BMS via bluetooth and neither of these have ever exhibited the same behaviour as the JKBMS monitoring via bluetooth.

[image]

mr-manuel, 2024-05-24:
Can you try with `v1.3.20240524dev`?

howetech, 2024-05-25:
> Can you try with `v1.3.20240524dev`?

Have just installed and will monitor over the next few days

cupertinomiranda, 2024-05-26:
Just to give an update, even with my concurrency fixes in the update process, I still see this random values. I am starting to believe it is related to some firmware bug in jkbms bluetooth.
I will soon try a similar approach to my initial patch, since I think there is no way around then detecting and avoiding those random values. Its not the best but it is the possible compromise.

mr-manuel, 2024-05-26:
Which values do you see? I limited them to be => 1 V and <= 5 V. Therefore they are filtered if different from the BMS.

cupertinomiranda, 2024-05-26:
Ok, but that is not sufficient. You should check from differences in consecutive readings and only allow then if they persist.
Be aware that the corruption might happen in any value coming from the BMS. Even on boolean values coming from BMS.

mr-manuel, 2024-05-27:
Theoretically there is checksum to verify integrity.

https://github.com/Louisvdw/dbus-serialbattery/blob/59a6d00c096c3e708aaf9cd827275c6500450374/etc/dbus-serialbattery/bms/jkbms_brn.py#L375-L379

cupertinomiranda, 2024-05-27:
I know that, however I meant that it was some data corruption within hardware before the checksum is computed. It could also be our rpi that might have some less stable memory that gets corrupted as well. At this point I am out of ideas for the cause. :smile
Will likely replace it by some more recent rpi4 for validation.

mr-manuel, 2024-05-27:
It could also be a faulty BMS or the connection to the VMR portal. Do you also log the values on a local InfluxDB? You could try that

cupertinomiranda, 2024-05-28:
I did also get the values being logged by homeassistant collected from the node-red. So I am sure it is not VRM. Faulty BMS, I doubt unless it is a problem of the firmware itself. I think @howetech confirmed the problem, actually.
@howetech: Can you check which firmware version you have on your jkbms?

[image]

mr-manuel, 2024-05-28:
> I limited them to be => 1 V and <= 5 V. Therefore they are filtered if different from the BMS.

If it is a firmware problem it should been have filtered with this

howetech, 2024-05-28:
[image]
[image]

Since running the latest nightly release I haven't seen any spikes or errors in the data received by MQTT into influxdb.  I have received 2 VRM notifications about cell inbalance during this time, however the VRM graphs do not show it so I'm guessing the device is sending the notification to VRM but its potentially so brief it doesn't show in any of the graphs.

Either way it is better than previously and I'll keeep monitoring it.

cupertinomiranda, 2024-05-28:
@mr-manuel It is filtered with your condition, however the problem is more general then just the cell voltages. I believe that it is why @howetech is still seeing those errors. Indeed @howetech VRM does seem to log some sort of average and not all readings.
@mr-manuel could you easily come up with a solution similar to my initial patch, that did not target BLE by mistake ?
Please notice it would need to filter any of the values coming from the BMS. It is tough for me to do it since I only have a production system to which I can test it on. :disappointed:

mr-manuel, 2024-05-29:
Just open a PR and I willlook through it. Please open the PR in my repository, this will make things easier.

cupertinomiranda, 2024-05-29:
I still did not have the time to work on fixing by checking the values. I will try to do it tomorrow which is an holiday around here. I will open a PR with a rough implementation, if you so allow it. Thanks

closed 2024-06-21 by mr-manuel