Louisvdw/dbus-serialbattery #1047: Float Mode => Multiplus II

Miro-2000, 2024-04-29:
### Describe the problem

My Multiplus II doesn't go into float mode. (bulk to absorption only).
Serial-Battery shows "Float (linear mode)" in Venus OS under JKBMS => Parameters
As you can see in the VRM diagram, CVL is also reduced to the limit.
There are a number of warnings and errors in the LOG-File, but I don't know why. What is goin wrong ?
[image]
[image]

### Driver version

v1.2.20240408

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.31

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Additional settings ---------
; Specify one or more BMS types to load else leave empty to try to load all available
; Available BMS:
;     Daly, Ecs, HeltecModbus, HLPdataBMS4S, Jkbms, Lifepower, LltJbd, Renogy, Seplos
; Available BMS, but disabled by default (just enter one or more below and it will be enabled):
;     ANT, MNB, Sinowealth
BMS_TYPE = Jkbms

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 110.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.950
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
[... 460 lines cut ...]
@40000000662e76a834038f34 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000662e76a8340a3a3c INFO:SerialBattery:> MIN CELL VOLTAGE: 2.95V | MAX CELL VOLTAGE: 3.45V
@40000000662e76a83410d1bc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 110.0A | MAX BATTERY DISCHARGE CURRENT: 80.0A
@40000000662e76a8341751cc INFO:SerialBattery:> CVCM:     True
@40000000662e76a8341dba6c INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000662e76a8342432ac INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000662e76a8342a8bac INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@40000000662e76a83430e0c4 INFO:SerialBattery:Serial Number/Unique Identifier: PV-Werner
@40000000662e76ab089770a4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000662e76f2122b9e74 INFO:SerialBattery:
@40000000662e76f212321e84 INFO:SerialBattery:Starting dbus-serialbattery
@40000000662e76f2124c2694 INFO:SerialBattery:Venus OS v3.31
@40000000662e76f21253d754 INFO:SerialBattery:dbus-serialbattery v1.2.20240408
@40000000662e7702133e912c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@40000000662e7702134553a4 INFO:SerialBattery:Testing Jkbms
@40000000662e770216a4787c INFO:SerialBattery:Connection established to Jkbms
@40000000662e770226a77c24 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
@40000000662e770232aa7774 INFO:SerialBattery:DeviceInstance = 1
@40000000662e770232c5853c INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
@40000000662e770232d490cc INFO:SerialBattery:Used DeviceInstances = ['1']
@40000000662e770232e12774 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
@40000000662e770234c03814 INFO:SerialBattery:publish config values = True
@40000000662e770236420d84 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
@40000000662e7702364eabfc INFO:SerialBattery:========== Settings ==========
```

mr-manuel, 2024-04-29:
Please provide logs in the correct format. It is well described how you have to get the logs. See documentation.

mr-manuel, 2024-04-30:
Anyway, the charging state of the MultiPlus/Quattro is not synched with the BMS charging state. I don't know, if it changes something, if you turn off the MultiPlus battery monitor via the VE.Config.

Please open a question in the Victron Community for that.

Regarding the errors in the logs it looks like you have interferences on your connection from the BMS to the Venus OS. Make sure you follow the recommendations in the documentation like a shielded cable.

closed 2024-04-30 by mr-manuel

Davii1995, 2024-06-11:
> ### Describe the problem
> 
> 
> 
> My Multiplus II doesn't go into float mode. (bulk to absorption only).
> 
> Serial-Battery shows "Float (linear mode)" in Venus OS under JKBMS => Parameters
> 
> As you can see in the VRM diagram, CVL is also reduced to the limit.
> 
> Float Voltage = 3,375 V/cell
> 
> There are a lot  of warnings and errors in the LOG-File, but I don't know why. What is going wrong ?
> 
> [image]
> 
> [image]
> 
> 
> 
> 
> 
> ### Driver version
> 
> 
> 
> v1.2.20240408
> 
> 
> 
> ### Venus OS device type
> 
> 
> 
> Raspberry Pi
> 
> 
> 
> ### Venus OS version
> 
> 
> 
> v3.31
> 
> 
> 
> ### BMS type
> 
> 
> 
> JKBMS / Heltec BMS
> 
> 
> 
> ### Cell count
> 
> 
> 
> 16
> 
> 
> 
> ### Battery count
> 
> 
> 
> 1
> 
> 
> 
[... 1067 lines cut ...]
> 
> @40000000662e770234c03814 INFO:SerialBattery:publish config values = True
> 
> @40000000662e770236420d84 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
> 
> @40000000662e7702364eabfc INFO:SerialBattery:========== Settings ==========
> 
> ```
> 
> 
> 
> 
> 
> ### Any other information that may be helpful
> 
> 
> 
> _No response_

Hey
Ich stehe vor derselben Thematik. 
Hast du es letztendlich irgendwie gelöst bekommen?

Danke und LG
David

mr-manuel, 2024-06-11:
Why are you quoting the whole post? Anyway the MutliPlus will never sync the battery state with the BMS. You can see the battery state of the MultiPlus as a own battery. Therefore they will not sync up. Just ignore it.

Davii1995, 2024-06-11:
Was not my plan to do so...
Thank you. As the CVL is set to float I will ignore it.