Louisvdw/dbus-serialbattery #1055: Battery capacity reduction for C rate computation when a cell goes in to protection.

AlexdeKairos, 2024-05-11:
### Is your feature request related to a problem? Please describe.

Following the CCL and DCL going to 0 in the event a unit goes to protection (Multi battery setup) one additional thing should be taken in to consideration. We use C rate units for many computations. In the event of a battery going in to protection,  its capacity should not be taken fir C rate computations. May be going to 0 if discharge and charge goes to 0 or may be on charging, just for computations, its value removed and same for discharging.  If not, although its and optimization of the charging and discharging,  the total C rate will be related to all units connected and not the "operative" ones. It's not critical but will optimize charging one step forward.

### Describe the solution you'd like

Adjust total capacity to the available battery units. The operational ones without protections active.

mr-manuel, 2024-05-12:
1. If I understood correctly, you want that the total shown capacity is changed, when a battery does not allow discharging anymore?
2. For what would this be useful?
3. Why are you not using the available capacity?

For a single battery I would not do that anyway. For a multi battery setup, you have to open an issue in the repository of the aggregator you are using.

AlexdeKairos, 2024-05-13:
Hi Manuel,
That's not actually the pretended functionality but close...

But you are right that may be is the battery agregator functionality.

Please close this feature request...

mr-manuel, 2024-05-13:
Ok. Only for clarification. Then you meant that, if the available capacity is not usable it should be 0?

closed 2024-05-13 by mr-manuel