Louisvdw/dbus-serialbattery #1056: Using SEPLOS V3 CLV in config.ini seems to be ignored

HF-SPSler, 2024-05-11:
### Describe the problem

Hello everybody,
since two days I´m testing the dbus-serialbattery driver in use with a SEPLOS V3 BMS. Before that, more than one of a year there was a JK BMS connected and used successfully. With SEPLOS V3 charging, discharging and so on wokrs fine but there is one issue with CVL value in remote console. In config.ini I set the following values (which worked fine with the last used JK BMS):

MAX_CELL_VOLTAGE   = 3.460
FLOAT_CELL_VOLTAGE = 3.375

Although the max_cell_voltage is set to 16 x 3.46 = 55,36V, the venus remote console shows a value of 57.6V which means that die parameter in config.ini might be ignored? I tried to find out why this happened, but I can´t find the reason. I would be very happy about any help, tanks.

[image]

### Driver version

1.3.20240428dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.20

### BMS type

Seplos

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = WARNING

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 70.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
; 22.08.2023 Leier Testweise die max. Spannung auf 3,46V (55,36V) erhöht
MAX_CELL_VOLTAGE   = 3.460
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
; Description: May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;              Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;                - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;              As you have to adopt this value to your system, I reccomend to start with
;              OVP voltage - 0.030 (see Example).
[... 131 lines cut ...]
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
```

### Relevant log output

```shell
I don´t know i this is very helpful. I guess this errors occurs even if I was trying to connect the battery to the USB to RS485 adapter. First I was not sure which RJ45 plug has to be used for a correct connection.

@40000000663c11dd37cb785c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11de03bf4c14 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11de0756d284 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e130e1f444 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e22590c87c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e235b3006c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e30a85d70c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e31ae49714 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e32b85598c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e40026140c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e500c61114 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000663c11e531c71c54 ERROR:SerialBattery:** DRIVER STOPPED! Another battery with the same serial number/unique identifier "SP18B2312010006" found! **
@40000000663c11e531d2efac ERROR:SerialBattery:Please check that the batteries have unique identifiers.
@40000000663c11e531d558dc ERROR:SerialBattery:Change the customizable field in your BMS settings to be unique.
@40000000663c11e531dbfffc ERROR:SerialBattery:To see which battery already uses this serial number/unique identifier check this file "/var/tmp/dbus-serialba>
@40000000663c12120e471914 *** CCGX booted (0) ***
```

mr-manuel, 2024-05-12:
1. Are you sure this is whole config file?
2. Please provide the logs in the correct format. Also it seems that your battery does not provide logs after reboot. Make sure that you have selected the correct log folder.
3. Reinstall the driver with the `install.sh`. There is no reason why the drive should behave this way. Have you made some manual changes?

HF-SPSler, 2024-05-13:
Hello Manuel,
 
first, thank you for your fast response. From the way you answered me, it´s obvisously not the whole config file :) I changed the posted config.ini and the actual log-file. I´m sorry for not reading your instructions to post a log file completely.

For the first logfile I used  `tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal` command now. 

Second command shows this: 
`2024-05-13 16:07:17.234830500 INFO: Start service vedirect-interface.ttyUSB1 once
2024-05-13 16:07:20.566341500 INFO: Create daemontools service dbus-cgwacs.ttyUSB1
2024-05-13 16:07:26.617642500 INFO: Start service dbus-cgwacs.ttyUSB1 once
2024-05-13 16:07:31.962327500 INFO: Create daemontools service dbus-fzsonick-48tl.ttyUSB1
2024-05-13 16:07:38.013090500 INFO: Start service dbus-fzsonick-48tl.ttyUSB1 once
2024-05-13 16:07:41.342642500 INFO: Create daemontools service dbus-imt-si-rs485tc.ttyUSB1
2024-05-13 16:07:47.391326500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB1 once
2024-05-13 16:07:50.692310500 INFO: Create daemontools service dbus-modbus-client.serial.ttyUSB1
2024-05-13 16:07:56.737398500 INFO: Start service dbus-modbus-client.serial.ttyUSB1 once
2024-05-13 16:08:04.118074500 INFO: Start service dbus-serialbattery.ttyUSB1 once
`
Yes I will try to reinstall the driver, maybe it´s a problem with the nightly build dev version I installed? Can you recommend a version of the dbus.serialbattery driver for use with the SEPLOS V3? Can you give me any hint which communication protocoll has to be chosen in SEPLOS BMS studio config?

Which kind of changes do you mean, changes in code or config.ini only? Because, in code I did not change anything.

Before I connected the new SEPLOS V3 Battery I used the whole system with a 5kWh DIY battery with JK BMS. So basicly the system was able to run very satisfying.

Thank you very much in advance.

mr-manuel, 2024-05-13:
@marcelrv maybe you know why this is the case that CVL is not the calculated one?

marcelrv, 2024-05-13:
In the seplos v3 integration I update the max battery voltage
` self.max_battery_voltage = spa[0x05] / 100` 
Outside of that I don't think the seplosV3 integration is updating the CVL..

 based on your log
`2024-05-13 16:09:27.373769500 DEBUG:SerialBattery:spa: [4, 16, 5400, 5600, 5680, 5760, 4800, 4640, 4800, 4320, 3400, 3500, 3500, 3600, 3100, 2900, 3000, 2700, 2500, 800, 500, 140, 150, 180, 100, 280, 300, 65396, 65356, 65336, 50, 65236, 300, 65236, 106, 600, 5, 3000, 205, 10, 3500, 3400, 1500, 1000, 800, 200, 30, 3111, 3131, 3151, 3181, 2781, 2751, 2731, 2631, 3111, 3131, 3151, 3181, 2761, 2631, 2731, 2581, 3201, 3231, 3281, 3331, 2761, 2731, 2731, 2631, 3531, 3581, 3581, 3681, 2831, 2731, 3231, 2731, 10, 3500, 50, 30, 960, 230, 200, 150, 100, 28000, 28000, 26208, 48, 60, 240, 10, 7, 0, 13, 0, 500, 300, 5552, 140, 65396, 9, 8]
`
I would expect this to be set to 57,60V. However, I'm not 100% clear how this self.max_battery_voltage is used by the rest of the integration.

for reference, I have a very ' simple' config.ini, where the voltage prob comes from the config.default.ini 
```
; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 150.0
MAX_BATTERY_DISCHARGE_CURRENT = 160.0

; --------- Additional settings ---------
; Specify one or more BMS types to load else leave empty to try to load all available
; Available BMS:
;     Daly, Ecs, HeltecModbus, HLPdataBMS4S, Jkbms, Lifepower, LltJbd, Renogy, Seplos
; Available BMS, but disabled by default (just enter one or more below and it will be enabled):
;     ANT, MNB, Sinowealth
BMS_TYPE = Seplosv3
```

as default config has 55.20 this is what I see in the screen
```
; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
```

[image]

Maybe go first with a simple config and then 'enhancing it' may help narrowing down the issue.

marcelrv, 2024-05-13:
> Yes I will try to reinstall the driver, maybe it´s a problem with the nightly build dev version I installed? Can you recommend a version of the dbus.serialbattery driver for use with the SEPLOS V3? Can you give me any hint which communication protocoll has to be chosen in SEPLOS BMS studio config?

If I recall right, the protocol you set in BMS Studio refers to the canbus protocol the BMS is using.
The driver is using the same 'serial protocol' as BMS studio (modbus tcp), so the 2 are not releated.

mr-manuel, 2024-05-14:
@marcelrv thank you for helping out.

@HF-SPSler comment out these lines and it should work:

https://github.com/Louisvdw/dbus-serialbattery/blob/59a6d00c096c3e708aaf9cd827275c6500450374/etc/dbus-serialbattery/bms/seplosv3.py#L210-L211

HF-SPSler, 2024-05-14:
Hello together,

@marcelrv The batterypack is connected with one of the RS485 ports to a isolated RS485 to USB converter. The CAN port (on the very left side) is unused. This is because I want to gain some experiences regarding the behavior of cell voltages while charging/discharging and some other things. As far as I know, this is unfortunately not possible via CAN.

Today early in the monring there was some time to connect directly to the battery via BMS Studio. I found out, that for CVL in VenusOS the following parameter from the battery is used:

[image]

After changing this value from 57,6V to 56,8V the value of CVL in VenusOS changes too:
[image]

In my opionion, this should be changed in your code, shouldn´t it?

mr-manuel, 2024-05-14:
Aleays the latest version is recommended. 

> In my opionion, this should be changed in your code, shouldn´t it?

Yes, marcelrv wrote this driver and I misset it. Please try what I wrote above.

HF-SPSler, 2024-05-14:
Okay, I will try to comment out the lines you posted above today in the evening.

marcelrv, 2024-05-14:
if the  self.max_battery_voltage  is representing the charge voltage request, then indeed the code should be changed to use parameter 102

[image]

mr-manuel, 2024-05-14:
But is the BMS adapting the CVL or is it fixed to this value?

In case we could add a BMS specific setting where you can select to use the parameters of the BMS or of the driver. Default would lead to the driver values to reduce confusion.

marcelrv, 2024-05-14:
> @marcelrv maybe you know why this is the case that CVL is not the calculated one?

I don't really know. I have not looked that detailed in the behaviour of the BMS.
My educated guess is that it is a fixed value, as I don' recall I ever saw this value change.
My battery is charging right now, is reporting a CVL of 57.6... should be full in 2 hours or so... will check if the value is changing when full/close to full

(screenshot from the CAN bus interface, which is reporting the BMS CVL)
[image]

marcelrv, 2024-05-14:
[image]
yes, andy is indicating paramter 102 as well in his spreadsheet.

I think the real question is @mr-manuel : should a BMS implentation set the self.max_battery_voltage.  (most implementations do set this value)
If yes, should it be set to #batteries * max cell voltage or to the CVL the BMS is reporting.

HF-SPSler, 2024-05-14:
In my opinion, sorry if it´s sound hard, the BMS has not to tell anyone a maximum voltage request. I suggest that only **one** part in a system gives setpoints and this part is in my eyes the driver, or for example the parameters in a MPPT charger if there is no VenusOS/Cerbo. Or am I wrong? I can only display my point of view but I am truely no professional :) 

On the other hand, you can make it optional in config.ini as @mr-manuel suggested it:

> In case we could add a BMS specific setting where you can select to use the parameters of the BMS or of the driver. Default would lead to the driver values to reduce confusion.

So, should I comment out the two lines which are marked above or better not at this moment?

mr-manuel, 2024-05-14:
Yes, comment out the two lines. I have already fixed it in the code, but not uploaded it yet. Have to wait to finish some other tests before.

HF-SPSler, 2024-05-14:
Seems that I´m no linux specialist or too stupid to comment out correctly... every time I try to comment out, the BMS connection to VenusOS failed and I get the Error 67 in Venus Remote Console. It does not matter if I type in 

#self.max_battery_voltage = spa[0x05] / 100
#self.min_battery_voltage = spa[0x05] / 100 

or

# self.max_battery_voltage = spa[0x05] / 100
# self.min_battery_voltage = spa[0x05] / 100  

Sorry for that :(

mdkeil, 2024-05-14:
@mr-manuel 

I think you have to replace the two lines with something like this?

```
self.max_battery_voltage = utils.MAX_CELL_VOLTAGE * self.cell_count
self.min_battery_voltage = utils.MIN_CELL_VOLTAGE * self.cell_count
```

mr-manuel, 2024-05-14:
Putting a `#` in front of the line is correct. Unfortunately I cannot test it. What does the log say? Else write me on Discord.

mr-manuel, 2024-05-20:
It is included in the latest nightly.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/d946e3d1d8bf755872bd03f9080f94641ebba4a6/etc/dbus-serialbattery/config.default.ini#L436-L438

closed 2024-05-20 by mr-manuel