Louisvdw/dbus-serialbattery #1059: Using external current sensor (smart shunt)

squowse, 2024-05-12:
### Describe the problem

I have successfully enabled SOC_CALCULATION in the driver config.ini file. This seems to be working and resetting to 100% correctly.

I changed the EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE to the path found by dbus-spy. First time after reboot, VRM came up with the warning "realtime connection disabled due to gx device overload". The VRM interface did not update.

I disabled the external current sensor in the config.ini, rebooted, tested and then re-enabled it exactly the same.
Now it appears to be working but the current displayed in VRM does not match the current shown using Victron Connect app simultaneously. The rounding is also wrong and is reminiscent of the BMS current reading.

Please see attachments

### Driver version

v1.3.20240510passthrough

### Venus OS device type

Cerbo GX

### Venus OS version

v3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E4:57:F5

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.525
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.325

; --------- SOC calculation ---------
; +++ This has nothing to do with "SOC reset voltage" in a section above +++
[... 359 lines cut ...]
2024-05-12 13:00:05.764171500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20220904)
2024-05-12 13:00:05.923900500 INFO:SerialBattery:publish config values = True
2024-05-12 13:00:06.013940500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c8478ce457f5
2024-05-12 13:00:06.014829500 INFO:SerialBattery:========== Settings ==========
2024-05-12 13:00:06.015862500 INFO:SerialBattery:> Connection voltage: 55.36 V | Current: -1.0 A | SoC: 100.0%
2024-05-12 13:00:06.016728500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
2024-05-12 13:00:06.017563500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-05-12 13:00:06.018465500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9 V | MAX CELL VOLTAGE: 3.525 V
2024-05-12 13:00:06.030775500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A
2024-05-12 13:00:06.031727500 INFO:SerialBattery:> CVCM:     True
2024-05-12 13:00:06.032797500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-05-12 13:00:06.033750500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-05-12 13:00:06.038111500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-05-12 13:00:06.039040500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: False
2024-05-12 13:00:06.042592500 INFO:SerialBattery:Serial Number/Unique Identifier: 2042102165
2024-05-12 13:00:06.215420500 INFO:SerialBattery:Monitoring external current using: "com.victronenergy.battery.ttyS6"/"/Dc/0/Current"
2024-05-12 13:00:06.238500500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2024-05-12 13:00:06.263038500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2024-05-12 13:00:06.526289500 INFO:SerialBattery:SOC initialized from dbus and set to 100.0%
```

### Any other information that may be helpful

[image]
[image]

squowse, 2024-05-12:
[image]

mr-manuel, 2024-05-12:
Please remove the quotes from `EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE` and `EXTERNAL_CURRENT_SENSOR_DBUS_PATH`. I wrongly inserted them in the example.

squowse, 2024-05-12:
```
; --------- Externcal current sensor ---------
; Specify the dbus device and path where the external current sensor is connected to
; You can find it by executing the dbus-spy command
; Example for a external current sensor connected to the VE.Bus port:
;     EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE = "com.victronenergy.vebus.ttyS3"
;     EXTERNAL_CURRENT_SENSOR_DBUS_PATH = "/Dc/0/Current"
EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE = com.victronenergy.battery.ttyS6 
;smart shunt 1
EXTERNAL_CURRENT_SENSOR_DBUS_PATH = `/Dc/0/Current``
```
```

mr-manuel, 2024-05-12:
Please check with the `dbus-spy` or in the Remote Console/GUI.

squowse, 2024-05-12:
The smartshunt current in Remote Console is rounded to 0.1A. The app shows to 0.01A 
[image]

squowse, 2024-05-12:
dbus-spy also shows current rounded to 0.1A

mr-manuel, 2024-05-12:
The rounding is normal, but how much is it delayed?

squowse, 2024-05-12:
VRM? Maybe a second or two.
Yes I think you are right - although the VE.Direct interface seems to get higher precision current measurements, the Cerbo only processes them at 0.1A.
Thanks for your help with this

mr-manuel, 2024-05-12:
But is there also a delay in the dbus-spy or Remote Console/GUI? If yes, how much?

squowse, 2024-05-12:
No real delay between the app and dbus-spy or remote console. Half a second or less.