Louisvdw/dbus-serialbattery #1061: External current sensor (smartshunt) in use. JKbms no longer shown as battery monitor

squowse, 2024-05-12:
### Describe the problem

Cannot select bms as battery monitor.

[image]
[image]
[image]
[image]

### Driver version

	1.3.20240510passthrough

### Venus OS device type

Cerbo GX

### Venus OS version

3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E4:57:F5

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.525
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.325

; --------- SOC calculation ---------
; +++ This has nothing to do with "SOC reset voltage" in a section above +++
; Description:
;     Calculate the SOC in the driver. Do not use the SOC reported by the BMS
; SOC_CALCULATION:
;     True: Calc SOC in the driver, do not use SOC reported from BMS
;         - The SOC is calculated by integration of the current reported
;         - The current reported can be corrected by the map
;           (SOC_CALC_CURRENT_REPORTED_BY_BMS, SOC_CALC_CURRENT_MEASURED_BY_USER)
;         - The SOC is set to 100% if the following conditions apply for at least SOC_RESET_TIME seconds:
;             * Highest cell voltage is higher or equal to MAX_CELL_VOLTAGE
;             * Current is lower than SOC_RESET_CURRENT
;         - The SOC is set to 0% if the following conditions apply for at least SOC_RESET_TIME seconds:
;             * Lowest cell voltage is lower or equal to MIN_CELL_VOLTAGE
;             * Battery is discharging
;         - The calculated SOC is stored in dbus to persist a driver restart
;     False: Use SOC reported from BMS (none of the other parameters apply)
; More info: https://github.com/Louisvdw/dbus-serialbattery/pull/868
SOC_CALCULATION   = True
SOC_RESET_CURRENT = 2	
SOC_RESET_TIME    = 30
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300
; Example to set small currents to zero
;     SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, -0.5, 0.5, 300
;     SOC_CALC_CURRENT_MEASURED_BY_USER = -300,    0,   0, 300

; --------- Externcal current sensor ---------
; Specify the dbus device and path where the external current sensor is connected to
; You can find it by executing the dbus-spy command
; Example for a external current sensor connected to the VE.Bus port:
;     EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE = "com.victronenergy.vebus.ttyS3”
;     EXTERNAL_CURRENT_SENSOR_DBUS_PATH = “/Dc/0/Current”

EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE = com.victronenergy.battery.ttyS6
;smart shunt 1
EXTERNAL_CURRENT_SENSOR_DBUS_PATH = /Dc/0/Current

; --------- SOC reset voltage (needs to match BMS settings) ---------
; +++ This has nothing to do with "SOC calculation" in a section below +++
; Description:
;     May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;     Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;       - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;     As you have to adopt this value to your system, I reccomend to start with
;     OVP voltage - 0.030 (see Example).
;       - Try to increase (add) by 0.005 in steps, if the system does not switch to float mode, even if
;         the target voltage SOC_RESET_VOLTAGE * CELL_COUNT is reached.
;       - Try to decrease (lower) by 0.005 in steps, if the system hits the OVP too fast, before all
;         cells could be balanced and the system goes into protection mode multiple times.
; Example:
;     If OVP is 3.650, then start with 3.620 and increase/decrease by 0.005
; Note:
;     The value has to be higher as the MAX_CELL_VOLTAGE
;     You also have to set CELL_VOLTAGES_WHILE_CHARGING accordingly, if you set CCCM_CV_ENABLE to true
;         else the charging current will be reduced to 0 before the target voltage is reached and the
;         battery will never switch to float
SOC_RESET_VOLTAGE = 3.650
; Specify after how many days the soc reset voltage should be reached again
; The timer is reset when the soc reset voltage is reached
; Leave empty if you don't want to use this
; Example:
;     Value is set to 15
;     day 1: soc reset reached once
;     day 16: soc reset reached twice
;     day 31: soc reset not reached since it's very cloudy
;     day 34: soc reset reached since the sun came out
;     day 49: soc reset reached again, since last time it took 3 days to reach soc reset voltage
SOC_RESET_AFTER_DAYS =
```

### Relevant log output

```shell
X
```

mr-manuel, 2024-05-13:
Without log output it‘s not possible to help you.

squowse, 2024-05-22:
Apologies. I have been away from the cerbo and only just worked out how to get remote access

2024-05-12 14:04:52.905682500 INFO:SerialBattery:
2024-05-12 14:04:52.907689500 INFO:SerialBattery:Starting dbus-serialbattery
2024-05-12 14:04:52.914780500 INFO:SerialBattery:Venus OS v3.30
2024-05-12 14:04:52.928019500 INFO:SerialBattery:dbus-serialbattery v1.3.20240510passthrough
2024-05-12 14:05:10.682824500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E4:57:F5
2024-05-12 14:05:10.684087500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E4:57:F5
2024-05-12 14:05:16.332330500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20220904)
2024-05-12 14:05:16.332340500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-05-12 14:05:21.277609500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-05-12 14:05:22.076330500 INFO:SerialBattery:DeviceInstance = 2
2024-05-12 14:05:22.080750500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_2.pid
2024-05-12 14:05:22.081477500 INFO:SerialBattery:Used DeviceInstances = ['2', '1']
2024-05-12 14:05:22.092073500 INFO:SerialBattery:com.victronenergy.battery.ble_c8478ce457f5
2024-05-12 14:05:22.094338500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20220904)
2024-05-12 14:05:22.247541500 INFO:SerialBattery:publish config values = True
2024-05-12 14:05:22.346572500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c8478ce457f5
2024-05-12 14:05:22.348848500 INFO:SerialBattery:========== Settings ==========
2024-05-12 14:05:22.350373500 INFO:SerialBattery:> Connection voltage: 54.1 V | Current: -1.2 A | SoC: 100.0%
2024-05-12 14:05:22.351573500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
2024-05-12 14:05:22.354216500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-05-12 14:05:22.357128500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9 V | MAX CELL VOLTAGE: 3.525 V
2024-05-12 14:05:22.357965500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A
2024-05-12 14:05:22.360734500 INFO:SerialBattery:> CVCM:     True
2024-05-12 14:05:22.363575500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-05-12 14:05:22.364821500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-05-12 14:05:22.367477500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-05-12 14:05:22.373337500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: False
2024-05-12 14:05:22.377580500 INFO:SerialBattery:Serial Number/Unique Identifier: 2042102165
2024-05-12 14:05:22.556606500 INFO:SerialBattery:Monitoring external current using: com.victronenergy.battery.ttyS6//Dc/0/Current
2024-05-12 14:05:22.582049500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2024-05-12 14:05:22.585848500 INFO:dbusmonitor:Found: com.victronenergy.battery.ble_c8478ce457f5, scanning and storing items
2024-05-12 14:05:22.686669500 INFO:SerialBattery:SOC initialized from dbus and set to 100.0%
2024-05-12 14:05:47.595519500 INFO:dbusmonitor:       com.victronenergy.battery.ble_c8478ce457f5 was skipped because it has no device instance
2024-05-12 14:05:47.605678500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS6, scanning and storing items
2024-05-12 14:05:47.622172500 INFO:dbusmonitor:       com.victronenergy.battery.ttyS6 has device instance 279
2024-05-12 14:05:47.646951500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2024-05-21 18:21:09.849820500 *** CCGX booted (0) ***
2024-05-21 18:21:26.281494500 
2024-05-21 18:21:26.281503500 INFO:Bluetooth details
2024-05-21 18:21:26.428389500 Attempting to disconnect from C8:47:8C:E4:57:F5
2024-05-21 18:21:26.428399500 Successful disconnected
2024-05-21 18:21:32.057119500 Device C8:47:8C:E4:57:F5 (public)
2024-05-21 18:21:32.057128500 	Alias: JK-B2A24S15P
2024-05-21 18:21:32.057130500 	Paired: no
2024-05-21 18:21:32.057131500 	Trusted: no
2024-05-21 18:21:32.057132500 	Blocked: no
2024-05-21 18:21:32.057134500 	Connected: no
2024-05-21 18:21:32.057135500 	LegacyPairing: no
2024-05-21 18:21:32.057137500 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-05-21 18:21:32.057140500 	RSSI: -58
2024-05-21 18:21:32.059435500 
2024-05-21 18:21:45.167193500 INFO:SerialBattery:
2024-05-21 18:21:45.168050500 INFO:SerialBattery:Starting dbus-serialbattery
2024-05-21 18:21:45.177182500 INFO:SerialBattery:Venus OS v3.30
2024-05-21 18:21:45.178016500 INFO:SerialBattery:dbus-serialbattery v1.3.20240510passthrough
2024-05-21 18:22:02.029898500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E4:57:F5
2024-05-21 18:22:02.030893500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E4:57:F5
2024-05-21 18:22:04.453007500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-05-21 18:22:04.453813500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-05-21 18:22:04.468737500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-05-21 18:22:04.784330500 [[0;93mCHG[0m] Device C8:47:8C:E4:57:F5 RSSI: -58
2024-05-21 18:22:04.784339500 [[0;93mCHG[0m] Device C8:47:8C:E4:57:F5 RSSI: -66
2024-05-21 18:22:04.784342500 [[0;93mCHG[0m] Device C8:47:8C:E4:57:F5 RSSI: -58
[... cut ...]
2024-05-21 18:31:21.375833500 INFO:SerialBattery:SOC initialized from dbus and set to 100.0%
2024-05-21 18:31:46.279400500 INFO:dbusmonitor:       com.victronenergy.battery.ble_c8478ce457f5 was skipped because it has no device instance
2024-05-21 18:31:46.282259500 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS6, scanning and storing items
2024-05-21 18:31:46.339868500 INFO:dbusmonitor:       com.victronenergy.battery.ttyS6 has device instance 279
2024-05-21 18:31:46.365658500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
`

mr-manuel, 2024-05-23:
According to the log output everything seems to work. Do you see the battery with the `dbus-spy`?

ramack, 2024-05-23:
It look like I'd have a similar/the same problem. On my side the serialbattery driver looks allright, I can see all data in dbus-spy and the local vrm, but via MQTT I don't get updates of e.g. cell voltages.
I didn't try the vrm app so far and I somehow would blame VenusOS 3.31 for it.

(Don't know if it is the same but thought of it when reading this ticket, so added the comment hoping it can help to bring up ideas, maybe testing with an older VenusOS could be one such.)

squowse, 2024-05-23:
Yes its there with the following info
com.victronenergy.battery.ble_c8478ce457f5

I notice the "consumedAh" is zero. Could this be the issue?
Actually Soc is 100% so I suppose thats normal.
[image]
[image]

mr-manuel, 2024-05-23:
You still don't see the BMS in DVCC? Everything is normal from your screenshots.

squowse, 2024-05-23:
I'm not using DVCC yet. I am trying to get the JKBMS to be "battery monitor" so it will display cell voltages, alarms etc. But it is not selectable. Settings > System setup >Battery Monitor
[image]

squowse, 2024-05-23:
Have turned on DVCC. JKBMS does not appear in list. Settings > DVCC > BMS Control
[image]

mr-manuel, 2024-05-23:
That is very odd, since you see it in the device list and in `dbus-spy`. Probalby a Venus OS bug. I remember to have read something somewhere. Update Venus OS to the latest beta version and it should work.

squowse, 2024-05-23:
I've updated to latest beta but the only thing changed is that Settings > DVCC no longer has a "controlling BMS" menu entry. 
VRM lists the battery monitor as missing. Thats probably not right? Dont worry about VE.Bus device, Multiplus is switched off.

[image]

[image]

mr-manuel, 2024-05-24:
Can you write me on Discord?

mr-manuel, 2024-05-24:
There seems to be something strange with your Venus OS configuration. Unfortunately I don't know what. Please try a factory reset: https://www.victronenergy.com/media/pg/Cerbo_GX/en/reset-to-factory-defaults-and-venus-os-reinstall.html

mr-manuel, 2024-06-03:
Are you still having the same issue?

squowse, 2024-06-20:
So I lost connectivity. VRM logging was the only thing that worked. Have onlly just got back to the installation. Have done a reboot and now I am able to get into remote console etc. The battery is there now but no smart shunt shown. VRM does not show a battery monitor available though. Should I do a complete reset to factory settings and abandon the beta version of Venus OS?

squowse, 2024-06-20:
[image]
[image]
[image]

squowse, 2024-06-20:
[image]

mr-manuel, 2024-06-20:
As I wrote before I do not know what is wrong with your system and why it behaves that ways. Only way to find out is to try a factory reset. I have many other system where it works without problems.

squowse, 2024-06-20:
Yea, i was just asking which versions of venus and serialbattery driver you would recommend. I’m inclined to avoid the venus beta as I loat connectivity last time. This may have been a coincidence.

mr-manuel, 2024-06-20:
You can use the beta, I'm also using it on three systems.

squowse, 2024-06-20:
Latest beta for both venus and serialbattery is best?

squowse, 2024-06-20:
[image]
[image]

squowse, 2024-06-20:
Cerbo reset to factory settings. Latest Venus beta installed. Mr.Manuel nightly build serialbattery driver installed. Log files to follow

squowse, 2024-06-20:
sorry you're going to have to tell me the command line to convert files using the tai64nlocal program. I'm using 
tee current |tai64nlocal > current.txt
and it seems to erase the original file as well as producing a blank file. I managed it before but cannot do it now. Have been trying various things and googling for 30 mins

squowse, 2024-06-20:
Also - the jkbms beeps when it connects by bluetooth and it is now doing this every minute.

mr-manuel, 2024-06-20:
Just check the [documentation](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#datalogdbus-serialbatteryttyusbcurrent-or-datalogdbus-serialbatteryttyama0current). Probably it beeps every minute because it's not connecting properly.

squowse, 2024-06-20:
I have read the documentation thats how i know you need it filtered by the tai64nlocal program. The documentation gives command to display it on screen. I did work out a command to pipe the file via tai64nlocal into another file but I just cant seem to get it right now.

mr-manuel, 2024-06-20:
```bash
cat /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal > /data/log-export.txt
```

closed 2024-07-02 by mr-manuel