Louisvdw/dbus-serialbattery #1062: Driver Fails to Start

WawaRoo, 2024-05-15:
### Describe the problem

I can't get the driver to load and connect to a USB Port.
It looks like the serial service is not starting.

### Driver version

nightly build "v1.3.20240510passthrough" 

### Venus OS device type

Cerbo GX

### Venus OS version

v3.31

### BMS type

Renogy BMS

### Cell count

4

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
GNU nano 4.9.3       /data/etc/dbus-serialbattery/config.ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you>
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting>
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

=====================================================================================================
  GNU nano 4.9.3   /data/etc/dbus-serialbattery/config.default.ini    Modified
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery vol>
; Example:
<= 46.4>[DEFAULT]
```

### Relevant log output

[... 156 lines cut ...]
2024-05-15 01:25:06.516689500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-15 01:25:06.520787500 INFO:SerialBattery:Testing Seplos
2024-05-15 01:25:07.544659500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-15 01:25:07.545366500 INFO:SerialBattery:Testing Seplosv3
2024-05-15 01:25:07.567644500 INFO:SerialBattery:Start testing for Seplos v3 on slave address 0
2024-05-15 01:25:08.415065500 ERROR:SerialBattery:Non blocking exception occurred: UnboundLocalError("local variable 'found' referenced before assignment") of type <class 'UnboundLocalError'> in /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py line #115
2024-05-15 01:25:08.919624500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-05-15 01:25:45.095983500 INFO:SerialBattery:
2024-05-15 01:25:45.096697500 INFO:SerialBattery:Starting dbus-serialbattery
2024-05-15 01:25:45.098711500 INFO:SerialBattery:Venus OS v3.31
2024-05-15 01:25:45.099356500 INFO:SerialBattery:dbus-serialbattery v1.2.20240401
2024-05-15 01:27:06.293524500 *** CCGX booted (0) ***
2024-05-15 02:01:32.378511500 *** CCGX booted (0) ***
2024-05-15 02:24:55.285172500 *** CCGX booted (0) ***
2024-05-15 02:35:18.331851500 *** CCGX booted (0) ***
2024-05-15 02:49:31.348447500 *** CCGX booted (0) ***
```

### Any other information that may be helpful

I tried a these driver versions as well: 
1) latest release "v1.2.20240408" (louisvdw's repo, stable)
2) latest release "v1.2.20240401" (mr-manuel's repo, stable, most up to date)

I did have other Renogy batteries linked; but have unlinked them (ie removed the link cable connecting the BMS's) to just get one renogy battery to talk to the Cerbo GX.

mr-manuel, 2024-05-15:
It seems that after the last reboot you are not getting any logs anymore. Are you sure, you selected the right port?

You could try with `tail -n 100 -F /var/log/dbus-serialbattery.*/current | tai64nlocal` to fetch all logs of all ttyUSB ports.

WawaRoo, 2024-05-15:
Looks like USB0 and USB1 are assinged to serial battery;  see log at bottom. 

I've tried physically moving the cable to try each of the ports;  I'm beginning to wonder if i have a cable issue, but it is a new generic RS232, although i also tried a another USB-RS485 but with CH343.

root@einstein:~# tail -n 100 -F /var/log/dbus-serialbattery.*/current | tai64nlocal
==> /var/log/dbus-serialbattery.ttyACM0/current <==
2024-05-14 01:07:55.477318500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-05-14 01:07:55.477329500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-05-14 01:07:55.631612500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:07:55.632564500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-05-14 01:07:55.774747500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:07:55.775492500 INFO:SerialBattery:Testing Ecs
2024-05-14 01:07:56.190799500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:07:56.192188500 INFO:SerialBattery:Testing HeltecModbus
2024-05-14 01:08:00.033668500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:00.034508500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-05-14 01:08:00.862641500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:00.863339500 INFO:SerialBattery:Testing Jkbms
2024-05-14 01:08:01.132995500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:01.150752500 INFO:SerialBattery:Testing Lifepower
2024-05-14 01:08:01.433551500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:01.451819500 INFO:SerialBattery:Testing LltJbd
2024-05-14 01:08:01.731814500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:01.749781500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-05-14 01:08:02.018714500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:02.036639500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2024-05-14 01:08:02.316094500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:02.333455500 INFO:SerialBattery:Testing Seplos
2024-05-14 01:08:03.357529500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:03.358243500 INFO:SerialBattery:Testing Seplosv3
2024-05-14 01:08:03.379534500 INFO:SerialBattery:Start testing for Seplos v3 on slave address 0
2024-05-14 01:08:04.231997500 ERROR:SerialBattery:Non blocking exception occurred: UnboundLocalError("local variable 'found' referenced before assignment") of type <class 'UnboundLocalError'> in /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py line #115
2024-05-14 01:08:04.733511500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM0
2024-05-14 01:08:19.552813500 INFO:SerialBattery:
2024-05-14 01:08:19.553509500 INFO:SerialBattery:Starting dbus-serialbattery
2024-05-14 01:08:19.555510500 INFO:SerialBattery:Venus OS v3.31
2024-05-14 01:08:19.556145500 INFO:SerialBattery:dbus-serialbattery v1.2.20240401
2024-05-14 01:08:35.571633500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-05-14 01:08:35.572527500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-05-14 01:08:35.720534500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:35.721593500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-05-14 01:08:35.864556500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:35.865385500 INFO:SerialBattery:Testing Ecs
2024-05-14 01:08:35.926893500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:35.927612500 INFO:SerialBattery:Testing HeltecModbus
2024-05-14 01:08:39.769690500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:39.770405500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-05-14 01:08:40.598480500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:40.598490500 INFO:SerialBattery:Testing Jkbms
2024-05-14 01:08:40.884162500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:40.901708500 INFO:SerialBattery:Testing Lifepower
2024-05-14 01:08:41.171623500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:41.197869500 INFO:SerialBattery:Testing LltJbd
2024-05-14 01:08:41.472042500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:41.489772500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-05-14 01:08:41.764194500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:41.788758500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2024-05-14 01:08:42.058895500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-05-14 01:08:42.076574500 INFO:SerialBattery:Testing Seplos
2024-05-14 01:08:43.100546500 ERROR:SerialBattery:>>> ERROR: No repl@400000006642b9f0100328c4 *** CCGX booted (0) ***
2024-05-14 15:30:23.254912500 *** CCGX booted (0) ***
2024-05-14 16:17:42.245450500 *** CCGX booted (0) ***
2024-05-14 17:41:24.281485500 *** CCGX booted (0) ***
2024-05-14 20:45:53.283295500 *** CCGX booted (0) ***
2024-05-14 23:44:16.295686500 *** CCGX booted (0) ***
2024-05-15 01:27:06.279542500 *** CCGX booted (0) ***
2024-05-15 02:01:32.369721500 *** CCGX booted (0) ***
2024-05-15 02:24:55.276295500 *** CCGX booted (0) ***
2024-05-15 02:35:18.323382500 *** CCGX booted (0) ***
2024-05-15 02:49:31.340049500 *** CCGX booted (0) ***
[... 101 lines cut ...]
2024-05-15 02:49:31.348447500 *** CCGX booted (0) ***

================================================================================

root@einstein:~# head /data/var/lib/serial-starter/*
==> /data/var/lib/serial-starter/ttyACM0 <==
sbattery

==> /data/var/lib/serial-starter/ttyS4 <==
mkx

==> /data/var/lib/serial-starter/ttyS5 <==
vedirect

==> /data/var/lib/serial-starter/ttyS6 <==
vedirect

==> /data/var/lib/serial-starter/ttyS7 <==
vedirect

==> /data/var/lib/serial-starter/ttyUSB0 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB1 <==
sbattery

mr-manuel, 2024-05-15:
Very strange that there are no new logs. Try to reinstall the driver and reboot the GX device after.

WawaRoo, 2024-05-16:
Results look very similar.

*** dbus-serialbattery v1.2.20240401 was installed. ***

rebooted
=======================================================
root@einstein:~# head /data/var/lib/serial-starter/*
==> /data/var/lib/serial-starter/ttyACM0 <==

==> /data/var/lib/serial-starter/ttyS4 <==
mkx

==> /data/var/lib/serial-starter/ttyS5 <==
vedirect

==> /data/var/lib/serial-starter/ttyS6 <==
vedirect

==> /data/var/lib/serial-starter/ttyS7 <==
vedirect

==> /data/var/lib/serial-starter/ttyUSB0 <==

==> /data/var/lib/serial-starter/ttyUSB1 <==
root@einstein:~# tail -n 100 -F /var/log/dbus-serialbattery.*/current | tai64nlo                                             cal
==> /var/log/dbus-serialbattery.ttyACM0/current <==
2024-05-14 01:07:55.477318500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-05-14 01:07:55.477329500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-05-14 01:07:55.631612500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:07:55.632564500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-05-14 01:07:55.774747500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:07:55.775492500 INFO:SerialBattery:Testing Ecs
2024-05-14 01:07:56.190799500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:07:56.192188500 INFO:SerialBattery:Testing HeltecModbus
2024-05-14 01:08:00.033668500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:00.034508500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-05-14 01:08:00.862641500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:00.863339500 INFO:SerialBattery:Testing Jkbms
2024-05-14 01:08:01.132995500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:01.150752500 INFO:SerialBattery:Testing Lifepower
2024-05-14 01:08:01.433551500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:01.451819500 INFO:SerialBattery:Testing LltJbd
2024-05-14 01:08:01.731814500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:01.749781500 INFO:SerialBattery:Testing Renogy at address "\x30                                             "
2024-05-14 01:08:02.018714500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:02.036639500 INFO:SerialBattery:Testing Renogy at address "\xf7                                             "
2024-05-14 01:08:02.316094500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:02.333455500 INFO:SerialBattery:Testing Seplos
2024-05-14 01:08:03.357529500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:03.358243500 INFO:SerialBattery:Testing Seplosv3
2024-05-14 01:08:03.379534500 INFO:SerialBattery:Start testing for Seplos v3 on                                              slave address 0
2024-05-14 01:08:04.231997500 ERROR:SerialBattery:Non blocking exception occurre                                             d: UnboundLocalError("local variable 'found' referenced before assignment") of t                                             ype <class 'UnboundLocalError'> in /opt/victronenergy/dbus-serialbattery/dbus-se                                             rialbattery.py line #115
2024-05-14 01:08:04.733511500 ERROR:SerialBattery:ERROR >>> No battery connectio                                             n at /dev/ttyACM0
2024-05-14 01:08:19.552813500 INFO:SerialBattery:
2024-05-14 01:08:19.553509500 INFO:SerialBattery:Starting dbus-serialbattery
2024-05-14 01:08:19.555510500 INFO:SerialBattery:Venus OS v3.31
2024-05-14 01:08:19.556145500 INFO:SerialBattery:dbus-serialbattery v1.2.2024040                                             1
2024-05-14 01:08:35.571633500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-05-14 01:08:35.572527500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-05-14 01:08:35.720534500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:35.721593500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-05-14 01:08:35.864556500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:35.865385500 INFO:SerialBattery:Testing Ecs
2024-05-14 01:08:35.926893500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:35.927612500 INFO:SerialBattery:Testing HeltecModbus
2024-05-14 01:08:39.769690500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:39.770405500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-05-14 01:08:40.598480500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
2024-05-14 01:08:40.598490500 INFO:SerialBattery:Testing Jkbms
2024-05-14 01:08:40.884162500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                             g
[... 98 lines cut ...]
[... cut ...]
2024-05-15 01:25:45.098711500 INFO:SerialBattery:Venus OS v3.31
2024-05-15 01:25:45.099356500 INFO:SerialBattery:dbus-serialbattery v1.2.2024040                                             1
2024-05-15 01:27:06.293524500 *** CCGX booted (0) ***
2024-05-15 02:01:32.378511500 *** CCGX booted (0) ***
2024-05-15 02:24:55.285172500 *** CCGX booted (0) ***
2024-05-15 02:35:18.331851500 *** CCGX booted (0) ***
2024-05-15 02:49:31.348447500 *** CCGX booted (0) ***
2024-05-15 23:13:19.313666500 *** CCGX booted (0) ***

mdkeil, 2024-05-16:
The log will be shorter If you directly set seplosv3 via config.ini

mr-manuel, 2024-05-16:
The serial starter is not starting the driver. Probably your adapter is not recognized anymore correctly. Check with `lsusb`.

Are you using a USB hub? If yes, connect the USB to serial adapter directly to a USB port.

The log of the serial starter should contain this output: https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#datalogserial-startercurrent

WawaRoo, 2024-05-16:
No USB hub;  I'm going to get a new cable;  will take a few days.

mr-manuel, 2024-05-16:
What is the output of `lsusb`?

WawaRoo, 2024-05-22:
root@einstein:~# lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 0bda:d723 Realtek Semiconductor Corp. 802.11n WLAN Adapter
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

mr-manuel, 2024-05-22:
Your USB to serial adapter is not recognized. That is the problem.

mr-manuel, 2024-05-27:
Do you still need assistance?

WawaRoo, 2024-05-28:
I have ordered a victron cable (after trying 3 different generic ones);   I wont have it until the weekend.   Ill let you know if i get things to work, or if i still have issues.   Thx

WawaRoo, 2024-06-22:
Yes...so got new cable....wired on RJ45....and got connected to my battery.

Given that i have 4 batteries (Renogy) linked together via rj45,  i installed dbus-aggregate-batteries;  It does not see any batteries.
Wondering how to troubleshoot?

mr-manuel, 2024-06-23:
Please post a picture of your cabeling. You know that you currently can connect only one battery per serial adapter? See also https://github.com/Louisvdw/dbus-serialbattery/issues/142.

WawaRoo, 2024-06-23:
[image]
In the pic of cabling, i can attach the renogy Bluetooth adapter in place of the cable to the Cerbo, and see all of the batteries.

I was thinking i may need a serial adapter per BMS.

Question:  I will need to add a USB hub, as the cerbo only has 3 ports, and i need 4,  Will this work?
Q2:  Is aggregate battery a good solution for connecting the 4 BMS's?  Or is it OK to just run 4 instances of serial battery?

mr-manuel, 2024-06-23:
https://louisvdw.github.io/dbus-serialbattery/faq/#what-can-i-do-if-the-bms-communication-is-unstable

https://louisvdw.github.io/dbus-serialbattery/faq/#how-to-aggregate-multiple-batteries

closed 2024-06-23 by mr-manuel

ewinkley, 2024-08-19:
@WawaRoo how did you connect the Renogy BT adapter to the Cerbo? Are you referring to the Renogy BT-2 module? In what port on the Cerbo did you connect?