Louisvdw/dbus-serialbattery #1066: DalyBMS 100% SOC is not achieved and decreases

e-i-k-e, 2024-05-23:
### Describe the problem

Hello,

I have been running the driver for a long time on a Daly BMS with a 100 Ah battery made of 16 Lifepo4 cells.

I updated the driver to the current version v1.2.20240408 and adjusted the config accordingly.

I noticed that the next day the SOC ended at around 95% and over the next few days the max. SOC continued to reduce, currently it is around 86% with a displayed battery voltage of 55.6 V (3.475 V per Cell).

I'm attaching my config for a better overview and would be very happy about some support.

[image]

### Driver version

v1.2.20240408 

### Venus OS device type

Raspberry Pi

### Venus OS version

3.40 ~16

### BMS type

Daly Smart BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 85.0
MAX_BATTERY_DISCHARGE_CURRENT = 85.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
[... 361 lines cut ...]
; factory default address will be 1
HELTEC_MODBUS_ADDR = 1

; --------- Voltage drop ---------
; If you have a voltage drop between the BMS and the charger because of wire size or length
; then you can specify the voltage drop here. The driver will then add the voltage drop
; to the calculated CVL to compensate.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 V and the battery is now charged until the charger reaches 55.20 V.
;     The BMS now measures 55.05 V since there is a voltage drop of 0.15 V on the cable.
;     Since the dbus-serialbattery reads the voltage of 55.05 V from the BMS the max voltage
;     of 55.20 V is never reached and max voltage is kept forever.
;     By setting the VOLTAGE_DROP to 0.15 V the voltage on the charger is increased and the
;     target voltage on the BMS is reached.
VOLTAGE_DROP = 0.26
```

### Relevant log output

```shell
-
```

mr-manuel, 2024-05-24:
The log output is important to be able to help you.

e-i-k-e, 2024-05-24:
@mr-manuel I added above, is this the correct or relevant log?

mr-manuel, 2024-05-27:
I looked up the code. The SOC is reset, when the battery switches to float mode.

Please check the FAQ for why it isn't switching to float: https://louisvdw.github.io/dbus-serialbattery/faq/#why-is-my-battery-not-switching-to-floatbulk

closed 2024-06-25 by mr-manuel