Louisvdw/dbus-serialbattery #1071: Unreliable serial communications

jmaunder, 2024-06-01:
### Describe the problem

My system: EasySolar-II GX, 6 x 50 Ah 17s LiFePO4 with JBD (Jiabaida) SP22S003 BMS. BMS communication is BMS UART <> Isolation module ADUM1201 <> Generic USB to TTL module <> USB cable <> USB port on GX card. Communication works OK with JBDtools. BMS modules have Bluetooth as well, but I found this to also be too unreliable, hence I am trying to do the serial connection.
Problem: seemingly random intermittent loss of the serial connection, eventually it does not re-connect. Not reliable enough to put it into service.
What I've tried: With USB hub and 2 x batteries, without USB hub and 1 x battery (direct connection). With aggregate-batteries and without it. Log file corresponds to 1 x battery and no USB hub, aggregate-batteries inactive.

### Driver version

v1.2.20240401

### Venus OS device type

GX Card (integrated in Victron device)

### Venus OS version

v3.13

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

17

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Specify one or more BMS types to load else leave empty to try to load all available
; Available BMS:
;     Daly, Ecs, HeltecModbus, HLPdataBMS4S, Jkbms, Lifepower, LltJbd, Renogy, Seplos
; Available BMS, but disabled by default (just enter one or more below and it will be enabled):
;     ANT, MNB, Sinowealth
BMS_TYPE = LltJbd

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 50.0

; Battery temperature
; Specify how the battery temperature is assembled
; 0 Get mean of temperature sensor 1 to sensor 4
; 1 Get only temperature from temperature sensor 1
; 2 Get only temperature from temperature sensor 2
; 3 Get only temperature from temperature sensor 3
; 4 Get only temperature from temperature sensor 4
TEMP_BATTERY = 1

; Temperature sensor 1 name
TEMP_1_NAME = MOSFET Temp.

; Temperature sensor 2 name
TEMP_2_NAME = Pack Temp. 1

; Temperature sensor 2 name
TEMP_3_NAME = Pack Temp. 2
[... 132 lines cut ...]
@40000000665ba56f336e967c INFO:SerialBattery:Connection established to LltJbd
@40000000665ba5702cf383d4 INFO:SerialBattery:Found existing battery with DeviceInstance = 7
@40000000665ba57036282c84 INFO:SerialBattery:DeviceInstance = 7
@40000000665ba570363e6404 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_7.pid
@40000000665ba5703645e1fc INFO:SerialBattery:Used DeviceInstances = ['5', '7', '6', '1', '3', '2', '4']
@40000000665ba570364f118c INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
@40000000665ba571224affd4 INFO:SerialBattery:publish config values = True
@40000000665ba571236f9dfc INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB0
@40000000665ba5712378ae4c INFO:SerialBattery:========== Settings ==========
@40000000665ba5712383453c INFO:SerialBattery:> Connection voltage: 55.92V | Current: 0.0A | SoC: None%
@40000000665ba571238c70e4 INFO:SerialBattery:> Cell count: 17 | Cells populated: 17
@40000000665ba571239565dc INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000665ba57123a6bb5c INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@40000000665ba57123bad3e4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 50.0A
@40000000665ba57123cd2b34 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 60.0A | MAX BATTERY DISCHARGE CURRENT: 70.0A (read from BMS)
@40000000665ba57123d58bbc INFO:SerialBattery:> CVCM:     True
@40000000665ba57123dec31c INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@40000000665ba57123e65c6c INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@40000000665ba57123ed45f4 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@40000000665ba57123f592f4 INFO:SerialBattery:Serial Number/Unique Identifier: 76_50.5Ah
```

### Any other information that may be helpful

serial-battery is promising, but unless I can have reliable BMS communication I cannot use it. Motivated to get it working 100%. Very little dbus, Raspberry Pi, Unix and Python knowledge but I can learn. Some electronics knowledge. Rest of the system working well. Until I have serial-battery working I will rely on Multi charging algorithm and BMS only for protection, not control.

mr-manuel, 2024-06-02:
Please read carefully how to provide logs in correct format. Check the documentation or read carefully the instructions when you open an issue.

jmaunder, 2024-06-03:
> Please read carefully how to provide logs in correct format. Check the documentation or read carefully the instructions when you open an issue.

Thanks for your reply, not sure what I've done wrong? I tried to follow the instructions as best I could. Happy to supply more information, whatever is needed.

mr-manuel, 2024-06-03:
> Please copy and paste any relevant log output here. See [here](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#driver-log-files) how to get the driver log files. The content will be automatically formatted into code, so no need for backticks.
> ⚠️ IMPORTANT: Do not forget to pipe the output through `tai64nlocal`, if you don't copy the commands from the documentation.

mr-manuel, 2024-06-03:
Anyway, your problem seems to be a unreliable USB to TTL adapter, since it disappears from the USB port. You could also check with `lbusb`, if your adapter is still recognized after a failure.

jmaunder, 2024-06-03:
I'm not sure it is, I have tried two different USB-TTL adapters and the problem persists. I tried 'lsusb' and the FTDI adapter appears there, so the adapter is present, but the serial-battery is not.
When I run the command <tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal> I get no output. Opening 'current' directly from WinSCP below is a snippet of recent entries. There are no error or warning entries in the log.
'''
@40000000665bd1f6181aa974 INFO: Start service dbus-modbus-client.serial.ttyUSB0 once
@40000000665bd1f924d6afdc INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000665bd2c833d5ef84 INFO: Start service gps-dbus.ttyUSB0 once
@40000000665bd2d2087fdd7c INFO: Start service vedirect-interface.ttyUSB0 once
@40000000665bd2d7167c2444 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000665bd2dc23e5914c INFO: Start service dbus-fzsonick-48tl.ttyUSB0 once
@40000000665bd2df300df2d4 INFO: Start service dbus-imt-si-rs485tc.ttyUSB0 once
@40000000665bd2e300dc256c INFO: Start service dbus-modbus-client.serial.ttyUSB0 once
@40000000665bd2e60cfc266c INFO: Start service dbus-serialbattery.ttyUSB0 once
@40000000665bd2fd253faadc INFO: Start service gps-dbus.ttyUSB0 once
@40000000665bd3003190a6ec INFO: Start service vedirect-interface.ttyUSB0 once
@40000000665bd30402509144 INFO: Start service dbus-cgwacs.ttyUSB0 once
@40000000665bd30910b219ec INFO: Start service dbus-fzsonick-48tl.ttyUSB0 once
@40000000665bd30c1da6eb64 INFO: Start service dbus-imt-si-rs485tc.ttyUSB0 once
@40000000665bd30f2a38578c INFO: Start service dbus-modbus-client.serial.ttyUSB0 once
@40000000665bd31236e02a14 INFO: Start service dbus-serialbattery.ttyUSB0 once
'''

mr-manuel, 2024-06-21:
Did you go through the whole [troubleshooting](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/) section? There is exactly explained what to check.

Make sure to also check [What to check, if it doesn't work](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#what-to-check-if-it-doesnt-work).

closed 2024-06-25 by mr-manuel