Louisvdw/dbus-serialbattery #1073: New install Bluetooth BMS not connecting - BleakDbusError in logs

cpjelectrics, 2024-06-02:
### Describe the problem

New install on Rapsberry pi 4. Bluetooth BMS not connecting asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed. Device not showing in GUI as not connected. Tried official Mr Manual release and the 72 hour stable release with same result

### Driver version

v1.2.20240227beta

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.31

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Bluetooth BMS ---------
; Description:
;     Specify the Bluetooth BMS and it's MAC address that you want to install. Leave emty to disable
; Available Bluetooth BMS:
;     Jkbms_Ble, LltJbd_Ble
; Example for one BMS:
;     BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
; Example for multiple BMS:
;     BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS = Jkbms_Ble A4:C1:37:53:A6:1D
```

### Relevant log output

```shell
2024-06-02 16:01:19.719504500 Device A4:C1:37:53:A6:1D (public)
2024-06-02 16:01:19.719509500   Alias: Campy2BMS
2024-06-02 16:01:19.719512500   Paired: no
2024-06-02 16:01:19.719514500   Trusted: yes
2024-06-02 16:01:19.719516500   Blocked: no
2024-06-02 16:01:19.719518500   Connected: no
2024-06-02 16:01:19.719521500   LegacyPairing: no
2024-06-02 16:01:19.719523500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-06-02 16:01:19.719526500   RSSI: -44
2024-06-02 16:01:19.720465500
2024-06-02 16:01:20.424846500 INFO:SerialBattery:
2024-06-02 16:01:20.425288500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-02 16:01:20.426333500 INFO:SerialBattery:Venus OS v3.31
2024-06-02 16:01:20.426725500 INFO:SerialBattery:dbus-serialbattery v1.2.20240227beta
2024-06-02 16:01:36.685309500 INFO:SerialBattery:Init of Jkbms_Ble at A4:C1:37:53:A6:1D
2024-06-02 16:01:36.685811500 INFO:SerialBattery:Test of Jkbms_Ble at A4:C1:37:53:A6:1D
2024-06-02 16:01:39.632535500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-06-02 16:01:39.632878500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-06-02 16:01:39.638046500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-06-02 16:01:39.798267500 [CHG] Device A4:C1:37:53:A6:1D RSSI: -44
2024-06-02 16:01:39.798274500 [CHG] Device A4:C1:37:53:A6:1D RSSI: -36
2024-06-02 16:01:39.798280500 [CHG] Device A4:C1:37:53:A6:1D RSSI: -36
2024-06-02 16:01:39.798286500 [CHG] Device A4:C1:37:53:A6:1D RSSI is nil
2024-06-02 16:01:41.743950500 Starting bluetooth: bluetoothd.
2024-06-02 16:01:41.762960500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-06-02 16:01:45.482114500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-06-02 16:01:45.483007500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-06-02 16:01:45.486531500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-06-02 16:01:46.198479500 ERROR:SerialBattery:No BMS found at A4:C1:37:53:A6:1D
2024-06-02 16:01:46.198802500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-06-02 16:01:47.581229500 Starting bluetooth: bluetoothd.
2024-06-02 16:01:47.600543500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-06-02 16:01:49.832970500
2024-06-02 16:01:49.833206500 INFO:Bluetooth details
2024-06-02 16:01:49.868197500 Attempting to disconnect from A4:C1:37:53:A6:1D
2024-06-02 16:01:49.868202500 Successful disconnected
2024-06-02 16:01:54.935734500 Device A4:C1:37:53:A6:1D (public)
2024-06-02 16:01:54.935739500   Alias: Campy2BMS
2024-06-02 16:01:54.935741500   Paired: no
2024-06-02 16:01:54.935744500   Trusted: yes
2024-06-02 16:01:54.935746500   Blocked: no
2024-06-02 16:01:54.935748500   Connected: no
2024-06-02 16:01:54.935751500   LegacyPairing: no
2024-06-02 16:01:54.935753500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-06-02 16:01:54.935756500   RSSI: -45
2024-06-02 16:01:54.937959500
2024-06-02 16:01:55.651389500 INFO:SerialBattery:
2024-06-02 16:01:55.651665500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-02 16:01:55.652595500 INFO:SerialBattery:Venus OS v3.31
2024-06-02 16:01:55.652863500 INFO:SerialBattery:dbus-serialbattery v1.2.20240227beta
```

AlexdeKairos, 2024-06-02:
You are using JK Bluetooth driver but mentioned you have JBD bms. Wich one you have? Review your selection

cpjelectrics, 2024-06-05:
Thanks, I incorrectly thought that Fogstar was JKBMS. I changed the config.ini to JBD BMS, rebooted and few times and its working as expected

closed 2024-06-05 by cpjelectrics