Louisvdw/dbus-serialbattery #1075: Hello, I have a problem with the current display with my Daly 4S.

AlfredSch, 2024-06-03:
### Describe the bug

Hello, I have a problem with the current display with my Daly 4S. With the Daly app everything is ok, when I connect it to the Raspberry the charging current/discharging is displayed the wrong way round. (charge is negative discharge is positive) can I adjust this somewhere? Or is this an error?
I am using the latest versions (serialbattery and Venus v3.32)
Greetings and thanks
Fred

### How to reproduce

charge is negative discharge is positive

### Expected behavior

the charging current/discharging is displayed the wrong way round. (charge is negative discharge is positive) 

### Driver version

dbus-serialbattery v1.2.20240408

### Venus OS device type

Raspberry Pi 4

### Venus OS version

Venus OS v3.32

### BMS type

Daly Smart BMS

### Cell count

4

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
standard file
```

### Relevant log output

```shell
INFO:SerialBattery:
2024-06-03 09:43:57.582881500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-03 09:43:57.584224500 INFO:SerialBattery:Venus OS v3.32
2024-06-03 09:43:57.584632500 INFO:SerialBattery:dbus-serialbattery v1.2.20240408
2024-06-03 09:44:33.777472500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-06-03 09:44:33.777879500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-06-03 09:44:34.220252500 INFO:SerialBattery:Connection established to Daly
2024-06-03 09:44:34.320043500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-06-03 09:44:34.423237500 INFO:SerialBattery:DeviceInstance = 1
2024-06-03 09:44:34.424019500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-06-03 09:44:34.424318500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-06-03 09:44:34.424615500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-06-03 09:44:34.562200500 INFO:SerialBattery:publish config values = True
2024-06-03 09:44:34.575361500 INFO:SerialBattery:Battery Daly connected to dbus from /dev/ttyUSB0
2024-06-03 09:44:34.575628500 INFO:SerialBattery:========== Settings ==========
2024-06-03 09:44:34.575944500 INFO:SerialBattery:> Connection voltage: 13.6V | Current: -0.49000000000000005A | SoC: None%
2024-06-03 09:44:34.576210500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 0
2024-06-03 09:44:34.576477500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-06-03 09:44:34.576757500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-06-03 09:44:34.577030500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2024-06-03 09:44:34.577293500 INFO:SerialBattery:> CVCM:     True
2024-06-03 09:44:34.577563500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-06-03 09:44:34.577829500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-06-03 09:44:34.578093500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-06-03 09:44:34.578355500 INFO:SerialBattery:Serial Number/Unique Identifier: 20240523
2024-06-03 09:46:51.243945500 INFO:SerialBattery:write force disable charging: true
2024-06-03 09:47:22.320460500 INFO:SerialBattery:write force disable charging: false
```

mr-manuel, 2024-06-03:
This is not a bug, it's a feature. For this there is a FAQ and a well documented `config.default.ini`. I put a lot of effort in it so that people can read it and find solutions ;-) read it and you will find your solution.

closed 2024-06-03 by mr-manuel

AlfredSch, 2024-06-04:
Hello Manuel, 
thanks for the quick reply.
I found it and was able to solve the problem. 
It's all very well documented if you read everything :-)

Thanks and regards
Fred