Louisvdw/dbus-serialbattery #1077: JKBMS Attempting to disconnect every second

djkakadu, 2024-06-06:
### Describe the bug

Hello i have problem with connecting to jkbms_ble

### How to reproduce

if I text these to putty everything is work:

/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:47:80:10:71:98

but if i close these session. connection gone.

### Expected behavior

working

### Driver version

v1.2.20240408

### Venus OS device type

Cerbo GX

### Venus OS version

v3.32

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]
MAX_BATTERY_CHARGE_CURRENT = 160.0
MAX_BATTERY_DISCHARGE_CURRENT = 160.0
MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375
BLUETOOTH_BMS = JKbms_Ble C8:47:80:10:71:98
LINEAR_LIMITATION_ENABLE = True
CVCM_ENABLE = True
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
CELL_VOLTAGES_WHILE_CHARGING      = 3.55, 3.50, 3.45, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION    =    0, 0.05,  0.5,    1
CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1
TIME_TO_GO_ENABLE = True
BLUETOOTH_USE_USB = False
```

### Relevant log output

```shell
@40000000666212ff03357304 INFO:SerialBattery:
@40000000666212ff034040a4 INFO:SerialBattery:Starting dbus-serialbattery
@40000000666212ff03606eec INFO:SerialBattery:Venus OS v3.32
@40000000666212ff036b5014 INFO:SerialBattery:dbus-serialbattery v1.2.20240408
@400000006662130f047d5c54 Traceback (most recent call last):
@400000006662130f047d8364   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 255, in <module>
@400000006662130f047d9304     main()
@400000006662130f047d96ec   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 189, in main
@400000006662130f047da68c     class_ = eval(port)
@400000006662130f047dae5c   File "<string>", line 1, in <module>
@400000006662130f047db62c NameError: name 'JKbms_Ble' is not defined
@400000006662130f0ea778f4 
@400000006662130f0eae1074 INFO:Bluetooth details
@4000000066621311171078ec Attempting to disconnect from C8:47:80:10:71:98
@400000006662131117109c14 [[0;93mCHG[0m] Device C8:47:80:10:71:98 ServicesResolved: no
@40000000666213111710a7cc Successful disconnected
@4000000066621316205e79e4 Device C8:47:80:10:71:98 (public)
@4000000066621316205e9d0c 	Alias: JK_B2A20S20P
@4000000066621316205ea0f4 	Paired: no
@4000000066621316205ea8c4 	Trusted: yes
@4000000066621316205eb094 	Blocked: no
@4000000066621316205eb47c 	Connected: no
@4000000066621316205eb864 	LegacyPairing: no
@4000000066621316205ec034 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@4000000066621316205ecbec 	RSSI: -69
@4000000066621316206c0a8c 
@4000000066621318034881ec INFO:SerialBattery:
@4000000066621318035372b4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006662131803721674 INFO:SerialBattery:Venus OS v3.32
@4000000066621318037bbb34 INFO:SerialBattery:dbus-serialbattery v1.2.20240408
```

mr-manuel, 2024-06-07:
Can I ask you why you don't provide the log in the format needed? EVERYWHERE you can find a note how to provide the log in the correct format.

Anyway. Python is case sensitive and you have a typo in your `BLUETOOTH_BMS` config setting. You wrote `JKbms_Ble` instead of `Jkbms_Ble`. The second letter has to be lowercase.

djkakadu, 2024-06-17:
Sorry for log format.
Anyway.
I make a mistake with these `JKbms_Ble` like you wrote.

Thank you for your help, now its working.

closed 2024-06-17 by djkakadu