Louisvdw/dbus-serialbattery #1078: JK-BMS Soc Reset general solution

salami738, 2024-06-07:
### Is your feature request related to a problem? Please describe.

Hi, thanks for creating this great software. 

I have a JK-BMS. This has a relatively strong SoC drift. Within one month it loses approx. 15%.
After that, the battery is only displayed as 85% full, for example.

As the JK-BMS has various software errors, you can't really achieve an automatic reset even with the 100% SoC setting in the BMS. 

### Describe the solution you'd like

How about calculating and updating the SoC in Python? Then we could work completely independently of the BMS
and you would have a universal fix for all BMS that have SoC problems.

### Describe alternatives you've considered

I have seen that there is a new feature that forces the BMS to perform an SoC reset via a workaround when a Bluetooth connection is established.
forces an SoC reset. The disadvantage is the unstable Bluetooth connection. Also i don't like having a wireless connection for security related features.

mr-manuel, 2024-06-07:
Else you could also use this to reset the SOC.

https://github.com/Louisvdw/dbus-serialbattery/blob/59a6d00c096c3e708aaf9cd827275c6500450374/etc/dbus-serialbattery/config.default.ini#L28-L57

closed 2024-06-07 by mr-manuel

salami738, 2024-06-07:
Thanks for the fast reply, will checkout the first possible solution. The documentation in the second config is wrong. The JK_BMS does not reset the SoC reliable, when the OVP is reached for one cell. Also this will throw a error in the victron vrm, every time this is reached.

mr-manuel, 2024-06-07:
It should not, since the warning is supressed from the driver in this special case. Could be, that the warning comes from another device. In that case you can reduce OVP or increase the setting on the MultiPlus for example.

salami738, 2024-06-10:
Thank you for your support. I didn't had the latest version. I updated to latest and will now check again.