Louisvdw/dbus-serialbattery #1082: Write configuration to BMS

waltertheodor, 2024-06-15:
### Is your feature request related to a problem? Please describe.

System description:
	4 x Victron smartSolar MPPT 250/100 with 30kWp solar panels
	3 x Victron Multiplus II 48/5000
	4 x battery 48V/(16 x EVE LF280K 280Ah)
		BMS-A V11.26 V11.XW JK-B2A24S20P (in operation)
		BMS-B V11.26 V11.XW JK-B2A24S20P (in operation)
		BMS-C V10.09 V10.XW JK-B2A24S20P (in operation)
		BMS-D V10.09 V10.XW JK-B2A24S20P (in operation)
		BMS-X V11.261 V11.XW JK-B2A24S20P (test system)	
	Venus-OS-Large v3.32 on Raspberrypi 4, 4GB RAM 
		Dbus-Serial-Battery 1.2.20240401 via RS485/USB
		Battery-Aggregator v3.0.66

Problem description:
	The graphic shows four phases of battery operation:
	1. from 20:00 to approx. 06:00 discharging for basic requirements in the house.
	2. from 06:00 to approx. 07:00 Discharging for the higher demand in the morning (heat pump).
	3. from 07:00 to approx. 08:30 charge the batteries to 100%.
	4. from 08:30 to approx. 20:00 float charging with rapidly fluctuating currents between + 5A and - 5A.
	However, the currents "measured" in phase 4 do not actually flow. These values are the result of incorrect measurements by the BMS. The cause is known, but does not contribute to solving the problem here. The result is an unusable charging and discharging balance of the batteries.
[image]

### Describe the solution you'd like

Solution to the problem:
	As soon as the batteries are full, charging is switched off.
	In Victron standard systems, such as the one I operate, the SolarCharger (Victron smartSolar MPPT 250/100) cannot perform this task, as it is supposed to supply the inverter (Victron Multiplus II 48/5000) and the batteries with solar power in parallel. The SolarCharger, inverter and batteries are therefore connected to a common 48V rail. This means that at least the float charge voltage is constantly applied to the batteries.
	In this system, the batteries are only charged via the inverters in an emergency.

Task:
	As soon as battery blocks A to D have reached SOC 100% for a certain time, the charging current is to be switched off simultaneously for all blocks. They are then switched on again at a defined SOC of e.g. 80%, or at night. The logic for this procedure is to be controlled by Node Red via the "CustomControl" node. To do this, it must be possible to control the "Charge ON/OFF" and "Discharge ON/OFF" switches in the BMS via the "CustomControl" node. This is done via the RS485 serial interface on the BMS.
	
	I have successfully tested the control via the RS485 interface with the following command sequences for all the BMS mentioned above:
	Charge On:		4E 57 00 13 00 00 00 00 02 03 00 AB 01 00 00 00 68 00 00 01 D1	Response:	4E 57 00 13 00 00 00 00 02 00 01 AB 00 00 00 00 68 00 00 01 CE
	Charge Off:		4E 57 00 13 00 00 00 00 02 03 00 AB 00 00 00 00 68 00 00 01 D0	Response:	4E 57 00 13 00 00 00 00 02 03 00 AB 00 00 00 00 68 00 00 01 D0
	Discharge On:	4E 57 00 13 00 00 00 00 02 03 00 AC 01 00 00 00 68 00 00 01 D2	Response:	4E 57 00 13 00 00 00 00 02 00 01 AC 00 00 00 00 68 00 00 01 CF
	Discharge Off:	4E 57 00 13 00 00 00 00 02 03 00 AC 00 00 00 00 68 00 00 01 D1	Response:	4E 57 00 13 00 00 00 00 02 00 01 AC 00 00 00 00 68 00 00 01 CF
	
As there has been a discussion about the behavior of the JK-BMS (and other BMS) described above for some time, there seems to be a need for a solution for a large number of operators of such PV systems.
Adding the desired functionality to DBus-Serial-Battery seems to make sense to me.

Walter

mr-manuel, 2024-06-15:
https://github.com/Louisvdw/dbus-serialbattery/discussions/1081#discussioncomment-9777043

closed 2024-06-15 by mr-manuel

waltertheodor, 2024-06-15:
Unfortunately, I am not familiar with Python or Github to make a PR (PullRequest?). Can't someone from the community who has enough knowledge do this?