Louisvdw/dbus-serialbattery #1084: '<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'>

chaerle, 2024-06-15:
### Describe the bug

After some time running, the drivers starts logging the following error and becomes instable, so that also AggregateBatteries does not work any more with the instance.

Log:

`@40000000666dcfd211538df4 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #937
@40000000666dcfd2116c6554 WARNING:SerialBattery:Error while executing calcMaxDischargeCurrentReferringToCellVoltage(). Using default value instead.
@40000000666dcfd21179c71c WARNING:SerialBattery:CELL_VOLTAGES_WHILE_DISCHARGING: [2.7, 2.8, 2.9, 3.1] • MAX_DISCHARGE_CURRENT_CV: [0.0, 6.0, 30.0, 60.0]
`

### How to reproduce

Unknown, just wait

### Expected behavior

no error

### Driver version

v1.2.20240408

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.31

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to TTL

### Config file

```ini
empty (default values)
```

### Relevant log output

```shell
`@40000000666dcfd211538df4 ERROR:SerialBattery:Exception occurred: TypeError("'<=' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #937
@40000000666dcfd2116c6554 WARNING:SerialBattery:Error while executing calcMaxDischargeCurrentReferringToCellVoltage(). Using default value instead.
@40000000666dcfd21179c71c WARNING:SerialBattery:CELL_VOLTAGES_WHILE_DISCHARGING: [2.7, 2.8, 2.9, 3.1] • MAX_DISCHARGE_CURRENT_CV: [0.0, 6.0, 30.0, 60.0]
`
```

mr-manuel, 2024-06-15:
Did you read the message when opening an issue?

[image]

chaerle, 2024-06-15:
Does not happen with v1.3.20240606dev any more.

closed 2024-06-15 by chaerle