Louisvdw/dbus-serialbattery #1088: Lost bluetooth connection to 2 BMS

WP-Rue, 2024-06-28:
### Describe the problem

Broken bluetooth connection to BMS (with internal RP4 Bluetooth device (switched off in Victron Remote Console)

### Driver version

1.3.20240624 stable and  20240627 nighty built also

### Venus OS device type

Raspberry Pi

### Venus OS version

V3.32 and 3.34

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

12

### Battery count

2

### Connection type

Bluetooth

### Config file

```ini
DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
 MAX_BATTERY_CHARGE_CURRENT = 50.0
 MAX_BATTERY_DISCHARGE_CURRENT = 60.0

MIN_CELL_VOLTAGE   = 3.3
MAX_CELL_VOLTAGE   = 4.16
FLOAT_CELL_VOLTAGE = 4.13
SOC_RESET_VOLTAGE = 4.16

SOC_LOW_WARNING = 10
SOC_LOW_ALARM   = 4
; --------- Bluetooth BMS ---------
; Description:
;     Specify the Bluetooth BMS and it's MAC address that you want to install. Leave emty to disable
; Available Bluetooth BMS:
;     Jkbms_Ble, LltJbd_Ble
; Example for one BMS:
;     BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
; Example for multiple BMS:
;     BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
;BLUETOOTH_BMS =LltJbd_Ble A5:C2:37:27:05:C2
BLUETOOTH_BMS =LltJbd_Ble A5:C2:37:27:05:C2, LltJbd_Ble A5:C2:37:27:04:F4
BLUETOOTH_USE_USB = False

CELL_VOLTAGES_WHILE_CHARGING = 4.16
```

### Relevant log output

```shell
[... 145 lines cut ...]
2024-06-28 22:13:22.669689500 	Paired: no
2024-06-28 22:13:22.669692500 	Trusted: no
2024-06-28 22:13:22.669694500 	Blocked: no
2024-06-28 22:13:22.669696500 	Connected: no
2024-06-28 22:13:22.669698500 	LegacyPairing: no
2024-06-28 22:13:22.670686500 
2024-06-28 22:13:23.404845500 INFO:SerialBattery:
2024-06-28 22:13:23.405233500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-28 22:13:23.406355500 INFO:SerialBattery:Venus OS v3.34
2024-06-28 22:13:23.406702500 INFO:SerialBattery:dbus-serialbattery v1.4.20240627dev
2024-06-28 22:13:39.809915500 INFO:SerialBattery:Init of LltJbd_Ble at A5:C2:37:27:05:C2
2024-06-28 22:13:39.810284500 INFO:SerialBattery:Test of LltJbd_Ble at A5:C2:37:27:05:C2
2024-06-28 22:13:44.823378500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2024-06-28 22:13:44.825244500 ERROR:SerialBattery:No BMS found at A5:C2:37:27:05:C2
2024-06-28 22:13:44.825577500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2024-06-28 22:13:46.666743500 
2024-06-28 22:13:46.666747500 INFO:Bluetooth details
2024-06-28 22:13:46.699320500 Attempting to disconnect from A5:C2:37:27:05:C2
2024-06-28 22:13:46.699324500 Successful disconnected
```

### Any other information that may be helpful

with older version of driver (April? ) the 2 BMS are visible in Venus with all information, but after 2 days of operation the connection was broken.
After installation of new driver I had connection to "grey"-BMS again, but also lost this after Updating Venus OS to 3.34

mr-manuel, 2024-06-29:
With `v1.2.20240408` it worked, correct? Could you write me on Discord? Since I would like to connect to your system for troubleshooting.

WP-Rue, 2024-06-29:
I have to install discord first…
And yes- with the older version it worked…

WP-Rue, 2024-06-29:
Sorry - we have to postpone it. I‘ll be abroad the next 3 weeks.Gesendet mit der GMX Mail AppAm 29.06.24 um 06:57 schrieb Manuel
                
            
            
                
                    Von: "Manuel" ***@***.***>Datum: 29. Juni 2024An: "Louisvdw/dbus-serialbattery" ***@***.***>Cc: "Author" ***@***.***>,"WP-Rue" ***@***.***>Betreff: Re: [Louisvdw/dbus-serialbattery] Lost bluetooth connection to 2 BMS (Issue #1088)
                
                
With v1.2.20240408 it worked, correct? Could you write me on Discord? Since I would like to connect to your system for troubleshooting.

mr-manuel, 2024-10-11:
I troubleshooted the Bluetooth problems now way over 100 hours and found no real solution to all this problems. Therefore I decided to not put any other effort into the Bluetooth part. Another reason is, that the users apparently do not appreciate the work I do and they cannot immagine how much time consuming this all is.

For that 1-2 donations, if at all, a month on over 9.000 dbus-serialbattery installations it is not worth it. If everyone would donate 1 €/year then the motivation would be another, but that is not the case.

Feel free to open a PR in my repository to help me with this :-)

mr-manuel, 2024-11-26:
Will continue in https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/86

closed 2024-11-26 by mr-manuel