Louisvdw/dbus-serialbattery #1089: JKBMS throws alarm #31 on latest nightly v1.4.20240629dev (mr-manuel's repo)

rver2024, 2024-06-30:
### Describe the problem

I haven't had time to play with the dbus-serialbattery driver for a bit until recently, I installed the latest nightly v1.4.20240629dev from mr-manuel's repo, and there is a notification about JKBMS alarm #31 which I didn't see before, and I couldn't find anything about this alarm by searching the internet. The JKBMS seems to be working fine otherwise. Anyone else has this alarm? 

Please help advise if you know something about this #31 alarm. Thanks a lot!

### Driver version

v1.4.20240629dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.14

### BMS type

JKBMS / Heltec BMS

### Cell count

8

### Battery count

8

### Connection type

Bluetooth

### Config file

```ini
n/a
```

### Relevant log output

```shell
n/a
```

mr-manuel, 2024-06-30:
This alarm is only to report, that you should check the logs, but you did not attach the logs.

rver2024, 2024-06-30:
I checked the log (/data/log/dbus-blebattery.0/current) and found that the debus-serialbattery driver won't connect to JKBMS, saying no BMS found in the mac address which is correct. I think if I leave it long enough it will eventually connect. Previous version connects quicker but will eventually drops off, some previous version I tried can keep the connection much longer than others but unfortunately I didn't keep a log of which version is the best. Anyway, here is log:

2024-06-30 19:04:57.173805500 INFO:Bluetooth details
2024-06-30 19:04:57.195784500 No default controller available
2024-06-30 19:05:02.231423500
2024-06-30 19:05:03.522090500 INFO:SerialBattery:
2024-06-30 19:05:03.523094500 INFO:SerialBattery:Starting dbus-serialbattery
2024-06-30 19:05:03.525355500 INFO:SerialBattery:Venus OS v3.14
2024-06-30 19:05:03.526028500 INFO:SerialBattery:dbus-serialbattery v1.4.20240629dev
2024-06-30 19:05:19.964603500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:xx:xx:xx
2024-06-30 19:05:19.965345500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:xx:xx:xx
2024-06-30 19:05:20.401426500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-06-30 19:05:20.402242500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-06-30 19:05:20.406744500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-06-30 19:05:22.482389500 Starting bluetooth: bluetoothd.
2024-06-30 19:05:22.498992500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-06-30 19:05:24.536178500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-06-30 19:05:24.536952500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-06-30 19:05:24.540415500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-06-30 19:05:26.622682500 Starting bluetooth: bluetoothd.
2024-06-30 19:05:26.640278500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-06-30 19:05:28.676824500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-06-30 19:05:28.677794500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-06-30 19:05:28.681648500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-06-30 19:05:29.479828500 ERROR:SerialBattery:No BMS found at C8:47:80:xx:xx:xx
2024-06-30 19:05:29.480571500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-06-30 19:05:30.760473500 Starting bluetooth: bluetoothd.
2024-06-30 19:05:30.776594500 INFO:SerialBattery:System Bluetooth daemon should have been restarted

rver2024, 2024-07-01:
I'd to add that I have been using the same config.ini file for a while now but didn't experience this #31 alarm in previous driver versions.

mr-manuel, 2024-07-02:
Some additional checks where implemented with the latest version to show the user, if there are issues in the background which the user don't see in the GUI. Now you write that you are using the same `config.ini` for a while now, but you did not provide it. Are you using a custom `config.ini` or not?

The logs you provided do not match the error. Additionally they do not show a correctly started driver which would be able to show the error #31. Please make sure to fetch the logs when the error is shown.

rver2024, 2024-07-02:
Sorry about the missing information! @mr-manuel. No error in my earlier log was because the JKBMS wasn't connected to CerboGX, I left it long enough and now it is connected with the #31 alarm. I updated my original post to include my config.ini and the log output with errors, I only copied and pasted two sections of the log output which just keep repeating themselves. Thanks!

rver2024, 2024-07-02:
I was mistaken to say that the JKBMS is connected to the CerboGX earlier, it was not even if it showed up on remote console, I think it is frozen or stalled or something since there is no reading update, and the red led on JKBMS is flashing, it should be solid red if it’s Bluetooth is connected.

mr-manuel, 2024-07-02:
Install the latest nightly from today and then make a reboot.

rver2024, 2024-07-03:
Thanks for your quick action! @mr-manuel 

Per your suggestion, I installed the lastest nightly but the JKBMS still wouldn't connect, rebooted a couple of times. Please see the log below:

root@einstein:~# tail -F -n 100 /data/log/dbus-blebattery.0/current | tai64nlocal
2024-07-03 03:05:20.710372500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-03 03:05:22.786620500 Starting bluetooth: bluetoothd.
2024-07-03 03:05:22.802261500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-03 03:05:24.834844500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-07-03 03:05:24.835636500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-03 03:05:24.838979500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-03 03:05:26.916064500 Starting bluetooth: bluetoothd.
2024-07-03 03:05:26.931537500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-03 03:05:28.969373500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-07-03 03:05:28.970137500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-03 03:05:28.973355500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-03 03:05:29.789343500 ERROR:SerialBattery:No BMS found at C8:47:80:xx:xx:xx
2024-07-03 03:05:29.790079500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-07-03 03:05:31.052855500 Starting bluetooth: bluetoothd.
2024-07-03 03:05:31.070312500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-03 03:05:33.351799500
2024-07-03 03:05:33.351940500 INFO:Bluetooth details
2024-07-03 03:05:33.380503500 No default controller available
2024-07-03 03:05:38.420523500
2024-07-03 03:05:39.692644500 INFO:SerialBattery:
2024-07-03 03:05:39.693353500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-03 03:05:39.695326500 INFO:SerialBattery:Venus OS v3.14
2024-07-03 03:05:39.695950500 INFO:SerialBattery:dbus-serialbattery v1.4.20240702dev
2024-07-03 03:05:56.127730500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:xx:xx:xx
2024-07-03 03:05:56.128592500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:xx:xx:xx
2024-07-03 03:05:56.539077500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-07-03 03:05:56.539873500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-03 03:05:56.543587500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-03 03:05:58.620710500 Starting bluetooth: bluetoothd.
2024-07-03 03:05:58.636337500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-03 03:06:00.673358500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-07-03 03:06:00.674112500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-03 03:06:00.677324500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-03 03:06:02.755122500 Starting bluetooth: bluetoothd.
2024-07-03 03:06:02.771833500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-03 03:06:04.807856500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-07-03 03:06:04.808739500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-03 03:06:04.812292500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-03 03:06:05.642423500 ERROR:SerialBattery:No BMS found at C8:47:80:xx:xx:xx
2024-07-03 03:06:05.643205500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-07-03 03:06:06.889298500 Starting bluetooth: bluetoothd.
2024-07-03 03:06:06.905177500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-03 03:06:09.189381500
2024-07-03 03:06:09.189530500 INFO:Bluetooth details
2024-07-03 03:06:09.211075500 No default controller available
2024-07-03 03:06:14.250758500
2024-07-03 03:06:15.529596500 INFO:SerialBattery:
2024-07-03 03:06:15.530295500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-03 03:06:15.532278500 INFO:SerialBattery:Venus OS v3.14
2024-07-03 03:06:15.532897500 INFO:SerialBattery:dbus-serialbattery v1.4.20240702dev
2024-07-03 03:06:31.980265500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:xx:xx:xx
2024-07-03 03:06:31.980982500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:xx:xx:xx
2024-07-03 03:06:32.419960500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('No Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-07-03 03:06:32.420773500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
[... cut ...]
2024-07-03 03:06:40.699286500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-03 03:06:41.494447500 ERROR:SerialBattery:No BMS found at C8:47:80:xx:xx:xx
2024-07-03 03:06:41.495241500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-07-03 03:06:42.775892500 Starting bluetooth: bluetoothd.
2024-07-03 03:06:42.791264500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-03 03:06:45.072245500
2024-07-03 03:06:45.072389500 INFO:Bluetooth details
2024-07-03 03:06:45.093390500 No default controller available

mr-manuel, 2024-07-03:
In this case the issue should be clear: `No Bluetooth adapters found.`

Are you using the internal one or an external?

rver2024, 2024-07-03:
You meant the Bluetooth adapter for CerboGX? It’s the internal one

mr-manuel, 2024-07-03:
Somehow this adapter is not recognized anymore. Could you install the latest Venus OS beta?

rver2024, 2024-07-04:
I upgraded the Venus OS to v3.33 from v3.14, the JKBMS would connect to Cerbo GX after a long time, I got a #119 alarm about settings error on the BMS, looked at the log and it seems to be about the CELL_VOLTAGES_WHILE_DISCHARGING (3.2 V) is higher than MIN_CELL_VOLTAGE (2.9 V), changed the CELL_VOLTAGES_WHILE_DISCHARGING from 3.2v to 2.9v and reboot the Cerbo GX but the JKBMS won't connect even after a long time. I googled around and found people commented that the built-in bluetooth of the Cerbo GX is really meant for assisting the initial setup of Cerbo GX's WiFi, it would shut off when the temperature of the Cerbo GX rises up, it is not for connecting to other accessories like a BMS, Victron's manual indicates this also. I decided to try the wire connection route and ordered a USB-RS485 adapter from Amazon.  I will update when I receive the adapter in a couple of days.

rver2024, 2024-07-06:
I changed my mind of using the wired connection instead of bluetooth one, ordered a TP-Link UB400 adapter, plugged it in to the Cerbo GX, the bluetooth LED lights up (previously it was always off when I used the internal bluetooth adapter but the JKBMS did connect to Cerbo GX, just intermittently) JKBMS connected to Cerbo GX very quickly and everything seems to work fine, I need more time to check the reliability of the connection, initial impression is promising so far. I checked the log and found the bluetoothd restarts every minute, is this a normal behavior of this dbus-serialbattery driver? @mr-manuel Here are the log output:

2024-07-06 19:50:45.980009500 Starting bluetooth: bluetoothd.
2024-07-06 19:50:45.996111500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-06 19:51:52.880185500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-06 19:51:52.894609500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-06 19:51:54.995400500 Starting bluetooth: bluetoothd.
2024-07-06 19:51:55.011611500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-06 19:53:01.896292500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-06 19:53:01.902106500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-06 19:53:04.003490500 Starting bluetooth: bluetoothd.
2024-07-06 19:53:04.020110500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-06 19:54:10.880400500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-06 19:54:10.890754500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-06 19:54:12.984455500 Starting bluetooth: bluetoothd.
2024-07-06 19:54:13.001261500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-07-06 19:55:19.891681500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-07-06 19:55:19.906218500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-07-06 19:55:22.036333500 Starting bluetooth: bluetoothd.
2024-07-06 19:55:22.056219500 INFO:SerialBattery:System Bluetooth daemon should have been restarted

mr-manuel, 2024-07-07:
No this is no normal behaviour.

rver2024, 2024-07-07:
Thanks, @mr-manuel, for your comment!

My system has been running for about 18 hours now since I installed external Bluetooth adapter, the JKBMS connection seems to be very stable, which I hadn’t experienced before when using the Cerbo GX’s internal adapter, so other than the bluetoothd restarting issue, it is promising that the driver using Bluetooth connection is stable enough to be used in a production system.

If this is not normal, can it be addressed by driver software update? Please let me know if you need help to diagnose this issue further.

mr-manuel, 2024-07-07:
I already spend hours and hours on troubleshooting, updating different packages and so on. Every time the same issue and I'm stuck to get some useful results. For solving this, we need someone that has much experience with Bluetooth connections.

Since your initial problem is solved, I will close this issue. There are already other issues opened for Bluetooth connection problems.

closed 2024-07-07 by mr-manuel

rver2024, 2024-07-07:
Sounds good, @mr-manuel. Thanks again!

mr-manuel, 2024-07-07:
You're welcome