Louisvdw/dbus-serialbattery #1091: EG4 LL 12v "TypeError" on nightly build

Justice12t, 2024-07-02:
### Describe the problem

I installed the nightly build on my GX when I saw the EG4-LL bms added.  Figured it could use some testing. It connects and outputs information via the troubleshoot commands but with an error and does not show on my GX device.  EG4 was not an option on the "bms type" drop down.
2024-07-02 13:51:32.740628500 TypeError: 'str' object is not callable

### Driver version

v1.4x-dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.40~33

### BMS type

Life/Tian Power

### Cell count

4

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
GNU nano 4.9.3                    /data/etc/dbus-serialbattery/config.ini                    Modified
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you                         >
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting                         >
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

### Relevant log output

```shell
root@einstein:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB2/current | tai64nlocal
2024-06-30 12:11:07.517837500 INFO:SerialBattery:Cell Total Voltage: 13.554v | Current: 4.55A
2024-06-30 12:11:07.518542500 INFO:SerialBattery:Capacity Left: 400 of 400.0 AH
2024-06-30 12:11:07.519218500 INFO:SerialBattery:SoC: 100% - Charging
2024-06-30 12:11:07.519797500 INFO:SerialBattery:===== DVCC State =====
2024-06-30 12:11:07.520722500 INFO:SerialBattery:DVCC Charger Mode: None
2024-06-30 12:11:07.521152500 INFO:SerialBattery:DVCC Charge Voltage: Nonev
2024-06-30 12:11:07.521795500 INFO:SerialBattery:Charge Current: None | Discharge Current: None
2024-06-30 12:11:07.522680500 INFO:SerialBattery:Charge Limit: None | Discharge Limit: None
2024-06-30 12:11:07.523071500 INFO:SerialBattery:===== Warning/Alarms =====
2024-06-30 12:11:07.523667500 INFO:SerialBattery: No Warnings - 0000
2024-06-30 12:11:07.524281500 INFO:SerialBattery: No Protection Events - 0000
2024-06-30 12:11:07.524880500 INFO:SerialBattery: No Errors - 0000
2024-06-30 12:11:07.525425500 INFO:SerialBattery:===== Temp =====
2024-06-30 12:11:07.525998500 INFO:SerialBattery:Temp 1: 27c | Temp 2: 24c | Temp Mos: 24c
2024-06-30 12:11:07.526760500 INFO:SerialBattery:Avg: 24c | Temp Max: 24c
2024-06-30 12:11:07.527363500 INFO:SerialBattery:Heater Status: False
2024-06-30 12:11:07.527902500 INFO:SerialBattery:===== Battery Stats =====
2024-06-30 12:11:07.528532500 INFO:SerialBattery:SoH: 100% | Cycle Count: 2
2024-06-30 12:11:07.529162500 INFO:SerialBattery:Max Charging Current: 220 A
2024-06-30 12:11:07.529708500 INFO:SerialBattery:===== Cell Stats =====
2024-06-30 12:11:07.530377500 INFO:SerialBattery:Cell 0 Voltage: 3.39
2024-06-30 12:11:07.531017500 INFO:SerialBattery:Cell 1 Voltage: 3.383
2024-06-30 12:11:07.531720500 INFO:SerialBattery:Cell 2 Voltage: 3.391
2024-06-30 12:11:07.532439500 INFO:SerialBattery:Cell 3 Voltage: 3.39
2024-06-30 12:11:07.533168500 INFO:SerialBattery:Cell Max/Min/Diff: (3.391/3.383/0.008)v
2024-06-30 12:11:07.796573500 INFO:SerialBattery:== Modbus packet good ==
2024-06-30 12:11:07.797407500 INFO:SerialBattery:Connection established to EG4_LL
2024-06-30 12:11:07.829009500 Traceback (most recent call last):
2024-06-30 12:11:07.829019500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", li           ne 291, in <module>
2024-06-30 12:11:07.829023500     main()
2024-06-30 12:11:07.829024500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", li           ne 238, in main
2024-06-30 12:11:07.829028500     helper = DbusHelper(battery)
2024-06-30 12:11:07.829030500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 51, i           n __init__
2024-06-30 12:11:07.829033500     for c in self.battery.unique_identifier()
2024-06-30 12:11:07.829147500 TypeError: 'str' object is not callable
2024-06-30 12:12:10.611236500 *** CCGX booted (0) ***
2024-06-30 12:16:51.572259500 *** CCGX booted (0) ***
2024-06-30 12:21:29.614406500 *** CCGX booted (0) ***
2024-07-02 10:58:41.618424500 *** CCGX booted (0) ***
2024-07-02 11:07:01.503530500 *** CCGX booted (0) ***
2024-07-02 11:14:04.620909500 *** CCGX booted (0) ***
2024-07-02 13:39:19.513800500 *** CCGX booted (0) ***
2024-07-02 13:47:04.604420500 *** CCGX booted (0) ***
2024-07-02 13:51:15.357418500 INFO:SerialBattery:
2024-07-02 13:51:15.358068500 INFO:SerialBattery:Starting dbus-serialbattery
2024-07-02 13:51:15.360247500 INFO:SerialBattery:Venus OS v3.40~33
2024-07-02 13:51:15.360928500 INFO:SerialBattery:dbus-serialbattery v1.4.20240629dev
2024-07-02 13:51:31.379993500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-07-02 13:51:31.380004500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-07-02 13:51:31.530896500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-02 13:51:31.530905500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-07-02 13:51:31.672625500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-02 13:51:31.674048500 INFO:SerialBattery:Testing Ecs
2024-07-02 13:51:31.756574500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-02 13:51:31.757324500 INFO:SerialBattery:Testing EG4_Lifepower
2024-07-02 13:51:32.051576500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-07-02 13:51:32.054191500 INFO:SerialBattery:Testing EG4_LL at address "\x7c"
2024-07-02 13:51:32.227997500 INFO:SerialBattery:== Modbus packet good ==
2024-07-02 13:51:32.388582500 INFO:SerialBattery:== Modbus packet good ==
2024-07-02 13:51:32.389806500 INFO:SerialBattery:===== HW Info =====
2024-07-02 13:51:32.390444500 INFO:SerialBattery:Battery Make/Model: FEG4-12.8V400Ah-V1.0
2024-07-02 13:51:32.391040500 INFO:SerialBattery:Hardware Version: Z01T01
2024-07-02 13:51:32.391599500 INFO:SerialBattery:Serial Number: 2022-10-25
2024-07-02 13:51:32.392137500 INFO:SerialBattery:===== BMS Data =====
2024-07-02 13:51:32.392748500 INFO:SerialBattery:Cell Total Voltage: 14.036v | Current: 0.0A
2024-07-02 13:51:32.393382500 INFO:SerialBattery:Capacity Left: 400 of 400.0 AH
2024-07-02 13:51:32.394334500 INFO:SerialBattery:SoC: 100% - Stanby
2024-07-02 13:51:32.395366500 INFO:SerialBattery:===== DVCC State =====
2024-07-02 13:51:32.396122500 INFO:SerialBattery:DVCC Charger Mode: None
2024-07-02 13:51:32.396812500 INFO:SerialBattery:DVCC Charge Voltage: Nonev
2024-07-02 13:51:32.397494500 INFO:SerialBattery:Charge Current: None | Discharge Current: None
2024-07-02 13:51:32.398149500 INFO:SerialBattery:Charge Limit: None | Discharge Limit: None
2024-07-02 13:51:32.398906500 INFO:SerialBattery:===== Warning/Alarms =====
2024-07-02 13:51:32.399477500 INFO:SerialBattery: No Warnings - 0000
2024-07-02 13:51:32.400023500 INFO:SerialBattery: No Protection Events - 0000
2024-07-02 13:51:32.400625500 INFO:SerialBattery: No Errors - 0000
2024-07-02 13:51:32.401973500 INFO:SerialBattery:===== Temp =====
2024-07-02 13:51:32.403132500 INFO:SerialBattery:Temp 1: 26c | Temp 2: 22c | Temp Mos: 23c
2024-07-02 13:51:32.403952500 INFO:SerialBattery:Avg: 22c | Temp Max: 23c
2024-07-02 13:51:32.404791500 INFO:SerialBattery:Heater Status: False
2024-07-02 13:51:32.405334500 INFO:SerialBattery:===== Battery Stats =====
2024-07-02 13:51:32.405947500 INFO:SerialBattery:SoH: 100% | Cycle Count: 2
2024-07-02 13:51:32.406745500 INFO:SerialBattery:Max Charging Current: 220 A
2024-07-02 13:51:32.407357500 INFO:SerialBattery:===== Cell Stats =====
2024-07-02 13:51:32.408066500 INFO:SerialBattery:Cell 0 Voltage: 3.517
2024-07-02 13:51:32.408737500 INFO:SerialBattery:Cell 1 Voltage: 3.487
2024-07-02 13:51:32.409389500 INFO:SerialBattery:Cell 2 Voltage: 3.514
2024-07-02 13:51:32.410036500 INFO:SerialBattery:Cell 3 Voltage: 3.518
2024-07-02 13:51:32.411001500 INFO:SerialBattery:Cell Max/Min/Diff: (3.518/3.487/0.031)v
2024-07-02 13:51:32.680426500 INFO:SerialBattery:== Modbus packet good ==
2024-07-02 13:51:32.681225500 INFO:SerialBattery:Connection established to EG4_LL
2024-07-02 13:51:32.740389500 Traceback (most recent call last):
2024-07-02 13:51:32.740398500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", li           ne 291, in <module>
2024-07-02 13:51:32.740402500     main()
2024-07-02 13:51:32.740403500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", li           ne 238, in main
2024-07-02 13:51:32.740407500     helper = DbusHelper(battery)
2024-07-02 13:51:32.740409500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 51, i           n __init__
2024-07-02 13:51:32.740412500     for c in self.battery.unique_identifier()
2024-07-02 13:51:32.740628500 TypeError: 'str' object is not callable
2024-07-02 13:59:04.613304500 *** CCGX booted (0) ***
```

Justice12t, 2024-07-02:
My bad, sorry!

Yes, that is now working!! everything seems good thus far.  I've got 3 batteries in my system total, so I'll have to look at combining them.  Thanks for the quick fix!

mr-manuel, 2024-07-02:
Is it possible to daisy chain the batteries over RS485 and give them different addresses?

Justice12t, 2024-07-02:
[image]

Yes, they are currently all connected and different addresses.  Would the battery aggregator in your repo work?

mr-manuel, 2024-07-03:
Oh ok. I don't know how the batteries behave in this situation, since they are internally connected and not with RS485.

closed 2024-07-03 by mr-manuel

BlueY21, 2024-07-03:
Those batteries look like they're connected the same way as Renogy batteries. They probably have or could be set to have unique Modbus device addresses.   Then they could all connect to the dbus driver over a single RS485-USB serial cable.

Justice12t, 2024-07-04:
@BlueY21 Yea, I have them set with different addresses.  I just don't think the multi battery beta includes the EG4 code.  Because when I install it, nothing shows.  But when I install the nightly build it seems to work perfect for a single battery.

BlueY21, 2024-07-04:
You're right.  I think the multi-battery on one cable code may only be in
the Renogy module.

mr-manuel, 2024-07-06:
I updated the multiple batteries branch. Maybe it works now.

Justice12t, 2024-07-07:
Thanks.  I can get it to show on the GX now.  I seem to be struggling with the modbus address.  If I leave it blank in the config.default.ini file it will work for one battery.  If I adjust it to 0x31, 0x32, 0x33 as I think it should be with my batteries being numbered 1, 2 and 3 via the dip switches.  Is there a command I can run to see what modbus address it is pulling to see if that points me in the right direction?  Threw a donation your way.  Really appreciate all the work on this.

mr-manuel, 2024-07-07:
Please continue in this discussion: https://github.com/mr-manuel/venus-os_dbus-serialbattery/discussions/46