Louisvdw/dbus-serialbattery #1094: CVL  much too high

MatKra89, 2024-07-08:
### Describe the bug

There is a massive problem with version v1.3.20240705. 
The CVL is wrong and almost shot my cells. 
Both the latest 1.3.20240705 and the nightly build etc. are affected. 
Apparently the software reads CVL from the BMS (max cell voltage). 
In Serialbattery I set 3.45V (bulk) and 3.35V (float) as the maximum voltage, but serialbattery still takes the full 3.65V from the BMS. When the cells reach this voltage, the value jumps back and forth. This means that it charges briefly at full power, then overshoots to 3.6V/cell and then immediately discharges again. I'm now reinstalling an old version, the problem didn't exist there. 
Or am I doing something wrong?

I have a 16s system and that would be a total of 55.2V CVL at 3.45V. With float it would still be 53.6V. 
Here you can see the 58.4V CVL....
[image]

### How to reproduce

cvl that is too high is constantly displayed...

### Expected behavior

What is expected is a cvl similar to the "old" version, i.e. maximum cell voltage * number.... in my case 3.45V * 16 = 55.2V

### Driver version

v1.3.20240705

### Venus OS device type

Raspberry Pi 3

### Venus OS version

3.33

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

3

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

; --------- Cell Voltages ---------
MIN_CELL_VOLTAGE   = 3.000
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.350

; --------- Bluetooth BMS ---------
BLUETOOTH_BMS = Jkbms_Ble C8:47:80:03:8A:2C, Jkbms_Ble C8:47:80:03:82:4A, Jkbms_Ble C8:47:80:03:89:20,

; --------- Bluetooth use USB ---------
[... 208 lines cut ...]
2024-07-08 16:56:07.316176500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-08 16:56:07.316689500 INFO:SerialBattery:Serial Number/Unique Identifier: 3071245268

==> /data/log/dbus-blebattery.2/current <==
2024-07-08 16:56:07.157889500 INFO:SerialBattery:DeviceInstance = 2
2024-07-08 16:56:07.160688500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_2.pid
2024-07-08 16:56:07.162338500 INFO:SerialBattery:Used DeviceInstances = ['2', '1', '3']
2024-07-08 16:56:07.163182500 INFO:SerialBattery:com.victronenergy.battery.ble_c84780038920
2024-07-08 16:56:07.164475500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (20240408)
2024-07-08 16:56:07.188155500 INFO:SerialBattery:publish config values = True
2024-07-08 16:56:07.233599500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c84780038920
2024-07-08 16:56:07.234399500 INFO:SerialBattery:========== Settings ==========
2024-07-08 16:56:07.235163500 INFO:SerialBattery:> Connection voltage: 54.01 V | Current: -1.7 A | SoC: 100%
2024-07-08 16:56:07.235814500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
2024-07-08 16:56:07.236448500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-07-08 16:56:07.237133500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.000 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.350 V
2024-07-08 16:56:07.237859500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A
2024-07-08 16:56:07.238581500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 130.0 A | MAX BATTERY DISCHARGE CURRENT: 130.0 A (read from BMS)
2024-07-08 16:56:07.239219500 INFO:SerialBattery:> CVCM:     True
2024-07-08 16:56:07.239871500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-07-08 16:56:07.240560500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-07-08 16:56:07.241161500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-07-08 16:56:07.241817500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-07-08 16:56:07.248243500 INFO:SerialBattery:Serial Number/Unique Identifier: 3071245114
```

mr-manuel, 2024-07-08:
What I see in your log is:

```
2024-07-08 16:56:07.235814500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 32
```

`Cells populated: 32` is wrong. Can you check this on an old version, if you have that also? I have many systems with JKBMS, but none gives me that issue.

Please try also the latest nightly and enable `GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True`. The reinstall the driver, go to the parameters page and post the screenshots of the items below DCL.

MatKra89, 2024-07-08:
Version v1.2.20240408 is currently running again and here too it says Cell count: 16 | Cells populated: 32

Now that the “old” version is running again, I have reinstalled the current v1.3.20240705. 
very interesting, because now CVL is suddenly correct??
I changed several times before and it was always wrong. Maybe there was something else cached? 
Tomorrow I'll test the behavior of the nightly build when the battery is full. CVL currently fits.....
[image]

mr-manuel, 2024-07-08:
Make sure to install the nightly, since it contains a fix in combination with `GUI_PARAMETERS_SHOW_ADDITIONAL_INFO`, else nothing other changed. See https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/48.

MatKra89, 2024-07-08:
The nightly is now running with the fix, so far so good. No abnormalities so far. 👍
I'll test this for a few days...

First of all, thank you very much for the quick help =)

MatKra89, 2024-07-09:
The nightly build has been running since yesterday evening in conjunction with GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True. 
Unfortunately the bulk voltage is still much too high, 55.2V would be correct.
[image]

[image]

[image]

[image]

[image]

I have a guess... my battery was almost full when I updated serialbattery. Since then it hasn't seen the 80% trigger on bulk. Is it possible that the correct CVL value is written when the trigger threshold falls below 80% and the CVL value is only incorrect immediately after installation?

mr-manuel, 2024-07-09:
Could you try to change
https://github.com/Louisvdw/dbus-serialbattery/blob/fd0fc1a193427df6ca0fadd141aa57826f761819/etc/dbus-serialbattery/config.default.ini#L46
to 3.50, so we see, if the value is comming from there

mr-manuel, 2024-07-10:
Can you also write me on Discord? You find the link by clicking on my profile.

MatKra89, 2024-07-10:
Today one of the three battery blocks went from float to bulk. Unfortunately, again the CVL is too high.
I set SOC_RESET_VOLTAGE = 3,500 in the config and rebooted. Unfortunately also without success.

I don't use Discord, I could offer email or WhatsApp.
Both public because YouTubers ^^

mr-manuel, 2024-07-10:
I would need access to your system, please send me your e-Mail, then I contact you.

MatKra89, 2024-07-10:
kramer-matthias@gmx.de

mr-manuel, 2024-07-10:
I deleted your comment for your privacy. I write you.

mr-manuel, 2024-07-12:
Did you get my e-Mail?

MatKra89, 2024-07-12:
***@***.***

mr-manuel, 2024-07-12:
The bug was fixed with `v1.4.20240712dev`.

closed 2024-07-12 by mr-manuel