Louisvdw/dbus-serialbattery #1099: Renogy - Multi battery setup documentation

mr-manuel, 2024-07-14:
### Is your feature request related to a problem? Please describe.

Create a documentation on how to connect multiple Renogy batteries to one serial adapter.

mr-manuel, 2024-07-14:
@TheVodden could you do that?

1. Describe how you need to prepare the batteries, to get the correct addresses
2. How to connect the batteries correctly
3. How to setup the driver correctly

Please add some pictures/screenshots, so that others could follow more easily. I will add this then to the documentation.

mr-manuel, 2024-07-14:
https://github.com/Louisvdw/dbus-serialbattery/issues/142#issuecomment-2181082259

https://github.com/Louisvdw/dbus-serialbattery/issues/142#issuecomment-2198648326

TheVodden, 2024-07-14:
I’m currently on vacation without a laptop but I’ll do my best to explain how it’s done. When I get back, I can post more comprehensive instructions with pics etc.

My batteries were auto addressed by the Renogy rs485 battery monitor.  I’ve made the assumption that the BT2 addresses them in the same manner but can’t confirm.  0x30 was the battery closest to the monitor and +1 each battery getting further away from the controller/monitor. Batteries are wired as per Renogy instructions here<https://www.renogy.com/content/RBT100LFP12SH-US/LFP100SH-Manual.pdf>. See page 8. Because the Cerbo will act as the master, you will need to unplug the BT2/battery monitor so communication collisions don’t occur.

I confirmed addresses using a modbus program called QModMaster. Im quite familiar with this program but its fairly straightforward. You can connect to each battery individually with a laptop and the same cable you use to connect to the Cerbo.  Connect to each battery individually, (485 not wired together) using function code 0x03, polling slave address 255 (broadcast address), and start address 5223 (1467 HEX) setting number of registers to 1. This will return each battery address you plug into. A full list of battery registers is found here<https://objects.githubusercontent.com/github-production-repository-file-5c1aeb/431221128/8134043?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240714%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240714T212031Z&X-Amz-Expires=300&X-Amz-Signature=b76d6dc1e822788ee4530522fcd8fc961f0bbe05e8ddc2d3811c6877657b9c09&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=431221128&response-content-disposition=attachment%3Bfilename%3DLithium.Iron.Battery.BMS.Modbus.Protocol.V1.7.zh-CN.en.1.pdf&response-content-type=application%2Fpdf> if you want to poll additional data (handy for troubleshooting) Optionally, if different addresses are returned, you can write a new address to the device by changing the function code to write holding registers, but it seems unlikely you should need to do this.

That is basic setup. In general, follow the manufacturers instructions and everything should work out.

Quadron, 2024-07-16:
I had also used teh BT2 first. To check the addresses I now used the CASModbusScanner program (from https://store.chipkin.com/products/tools/cas-modbus-scanner) to scan all the adresses while connected with the usb cable to my windows laptop. Batteries daisychained together as they were connected to the BT2.
I gave me responses "not available, Device exists but does not support this function" on 4 addresses: 48, 49, 50 and 51. Which translate to 0X30, 0X31, 0X32 and 0X33.
I reinstalled dbus-serialbattery to make sure my config file was ok. Then inserted the 4 addesses. 
The logfile now says:
Testing Renogy at address "\x30"
Renogy RBT100LFP12SH-G1

After each address. 
In the end it says 
DRIVER STOPPED! Another battery with the same serialnumber/unique identifier "RBT100LFP12SHG1_100.0Ah"
Change battery capacities to be unique. 

But how do I do that?

Quadron, 2024-07-16:
This is the USB to RJ45 RS485 cable pinout that is working for me:
[image]

mr-manuel, 2024-07-17:
Hi, I will change that for Renogy batteries to use the port as unique identifier by default. They seem to not provide this info.

mr-manuel, 2024-07-19:
The port is now used for the unique identifier for the Renogy battery as default. See https://github.com/mr-manuel/venus-os_dbus-serialbattery/commit/ed1950ba61384e22cd118e91ef510bc86bfde505.

Quadron, 2024-07-20:
"do you see the serial number in their software?"

I have not been alle to find anything in the renogy app that provides a unique number of each battery. Only a numbering 1 through 4 (in my case) based on the rs485 daisychain connection.

ewinkley, 2024-08-22:
> This is the USB to RJ45 RS485 cable pinout that is working for me: ![Victron FTDI to Renogy RJ45 pinout](https://private-user-images.githubusercontent.com/15707629/350008882-cc4f0082-491e-415a-b273-7c25b3150a4c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQzNTE2MjAsIm5iZiI6MTcyNDM1MTMyMCwicGF0aCI6Ii8xNTcwNzYyOS8zNTAwMDg4ODItY2M0ZjAwODItNDkxZS00MTVhLWIyNzMtN2MyNWIzMTUwYTRjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIyVDE4Mjg0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA5YmU0MDUwYzViZTA1N2I0NjljOTZlYzc3ZmQxN2M1ZjQzNDhmZjZhMjMzMjcxMTM4ODExN2NkNTgxNGMzZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Alwmil_2GVtq69zQO-A9ZGbVgjsfO_1XQDe9HHLDqbo)

I can confirm this works for my system. Total AH is identical to what the Renogy DC Home app shows (199.91AH) so I believe it is reporting both batteries (total of two 100AH installed).  And I am using the modded serialbattery driver.

2024-08-22 18:06:35.435169500 INFO:SerialBattery:
2024-08-22 18:06:35.436311500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-22 18:06:35.438857500 INFO:SerialBattery:Venus OS v3.50~13
2024-08-22 18:06:35.439899500 INFO:SerialBattery:dbus-serialbattery v1.4.20240714modbus
2024-08-22 18:06:51.452580500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-22 18:06:51.453649500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-08-22 18:06:51.786599500 INFO:SerialBattery:RENOGY RBT100LFP12SH-G1
2024-08-22 18:06:53.121685500 INFO:SerialBattery:Connection established to Renogy
2024-08-22 18:06:59.404388500 INFO:SerialBattery:DeviceInstance = 1
2024-08-22 18:06:59.407347500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-08-22 18:06:59.408645500 INFO:SerialBattery:Used DeviceInstances = []
2024-08-22 18:06:59.410130500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0__0
2024-08-22 18:06:59.557764500 INFO:SerialBattery:publish config values = False
2024-08-22 18:06:59.604760500 INFO:SerialBattery:Polling data every 1.000 s
2024-08-22 18:06:59.613927500 INFO:SerialBattery:Battery Renogy connected to dbus from /dev/ttyUSB0
2024-08-22 18:06:59.619016500 INFO:SerialBattery:========== Settings ==========
2024-08-22 18:06:59.621630500 INFO:SerialBattery:> Connection voltage: 13.3 V | Current: -0.46 A | SoC: 86.641977780002%
2024-08-22 18:06:59.627761500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2024-08-22 18:06:59.636891500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-22 18:06:59.638045500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-22 18:06:59.639864500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-08-22 18:06:59.641416500 INFO:SerialBattery:> CVCM:     True
2024-08-22 18:06:59.644698500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-22 18:06:59.647415500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-22 18:06:59.650025500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-22 18:06:59.651357500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-08-22 18:06:59.660687500 INFO:SerialBattery:Serial Number/Unique Identifier: RENOGYRBT100LFP12SHG1_99.91Ah
2024-08-22 18:06:59.825010500 INFO:SerialBattery:DeviceInstance = 1
2024-08-22 18:06:59.826390500 INFO:SerialBattery:Changed DeviceInstance = 1
2024-08-22 18:06:59.903238500 INFO:SerialBattery:Changed CustomName = SerialBattery(Renogy)
2024-08-22 18:14:17.996516500 WARNING:SerialBattery:Polling took too long. Set to 3.000 s

mr-manuel, 2024-10-07:
Any of you can write a very simple instruction for other non tecnical users to achieve a multi battery setup with Renogy batteries?

Like:
1. Connect all batteries to the BT2 to give the batteries individual addresses
2. Disconnect the BT2 and connect the batteries to the USB to RS485 adapter
3. Set address `0x30`, `0x31` (two batteries), `0x32` (three batteries), ... (will I add later) in the `config.ini`