Louisvdw/dbus-serialbattery #1106: Venus OS after reboot no BMS

mactoolz, 2024-08-06:
### Describe the problem

After the restart from Venus OS there are no coming all Seplos BMS or sometimes there are no BMS at the device list.

My solution for that is to unplug the original Venus USB RS485 adapter and plug in. They need many many times to get the device into the Venus OS. I think it needs 1 minute if i can see the BMS in my Venus OS device list.

### Driver version

v.1.2.20240227beta

### Venus OS device type

Cerbo GX

### Venus OS version

3.41

### BMS type

Seplos

### Cell count

7

### Battery count

7

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.4
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
; Description:
;     May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;     Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;       - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;     As you have to adopt this value to your system, I reccomend to start with
;     OVP voltage - 0.030 (see Example).
;       - Try to increase (add) by 0.005 in steps, if the system does not switch to float mode, even if
;         the target voltage SOC_RESET_VOLTAGE * CELL_COUNT is reached.
[... 504 lines cut ...]
@4000000066b2337e1d12583c INFO:SerialBattery:Testing Renogy at address "\x30"
@4000000066b2337e2ece9504 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000066b2337e2f48ce14 INFO:SerialBattery:Testing Renogy at address "\xf7"
@4000000066b2337f04af7874 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000066b2337f052cb6f4 INFO:SerialBattery:Testing Seplos
@4000000066b233800d9d7364 INFO:SerialBattery:Connection established to Seplos
@4000000066b233800db24384 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB6
@4000000066b233800ddc4954 INFO:SerialBattery:========== Settings ==========
@4000000066b233800df0794c INFO:SerialBattery:> Connection voltage: 54.05V | Current: 35.03A | SoC: 91.0%
@4000000066b233800dffe684 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
@4000000066b233800e0e84b4 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000066b233800ecac034 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.4V
@4000000066b233800ee05b74 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@4000000066b233800ef239ac INFO:SerialBattery:> CVCM:     True
@4000000066b2338010501814 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000066b233801064d4ac INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000066b2338010c2c38c INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@4000000066b2338011845254 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_287.0Ah
@4000000066b233982c5557a4 INFO:SerialBattery:Found existing battery with DeviceInstance = 8
@4000000066b2339b00cd2594 INFO:SerialBattery:DeviceInstance = 8
@4000000066b2339b00eea7b4 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_8.pid
@4000000066b2339b00fd037c INFO:SerialBattery:Used DeviceInstances = ['2', '4', '3', '1', '5', '7', '8']
@4000000066b2339b01102204 INFO:SerialBattery:com.victronenergy.battery.ttyUSB6
@4000000066b2339b0926c024 INFO:SerialBattery:publish config values = True
```

mr-manuel, 2024-08-06:
Please check the docs, since this is no driver issue, but an adapter, cabeling or USB hub issue.

You could also check the CPU load.

mactoolz, 2024-08-06:
No, there is no wireing problems. After unplug and plug in is everything working. 

If we have a cable problem it doesnt work for many days ...

mactoolz, 2024-08-06:
The cpu load is cery high, for each of them serial adapter the cpu load goes up for 6% ...

mr-manuel, 2024-08-06:
Please describe better your setup. Which adapters are you using, which GX device, USB hub?

mactoolz, 2024-08-06:
7x Victron Original RS485 Adapter
10x USB Hub with seperate power plug
Cerbo GX, 3.41 Version

mactoolz, 2024-08-06:
which parameter is that at the config ?

mr-manuel, 2024-08-06:
https://github.com/Louisvdw/dbus-serialbattery/blob/fd0fc1a193427df6ca0fadd141aa57826f761819/etc/dbus-serialbattery/config.default.ini#L321-L327

mactoolz, 2024-08-06:
why do you think that is the solution ? What happens if the parameter set to seplos BMS Type

mactoolz, 2024-08-06:
i set the parameter, it doesnt work.

mactoolz, 2024-08-06:
[image]

mr-manuel, 2024-08-06:
> why do you think that is the solution ? What happens if the parameter set to seplos BMS Type

It is written in the description!?

> i set the parameter, it doesnt work.

what does not work?

Anyway no one tested it with that much batteries except you. Unfortunately I have no idea to help you further.

mactoolz, 2024-08-07:
ok, but it should be possible to integrate the number of serial adapters in the first place, shouldn't it?

What can I test so that it somehow works in some way?
the error can be found ?

The number of 7 adapters is a bit more than normal, but it should be possible.

mr-manuel, 2024-08-07:
As you can see in the logs, the driver cannot see the USB to serial adapter anymore since it's gone in the OS.

```
@4000000066b232d11ede3e9c Traceback (most recent call last):
@4000000066b232d11ede65ac   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 265, in open
@4000000066b232d11ede754c     self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
@4000000066b232d11ede8104 FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB6'
@4000000066b232d11ede90a4 
@4000000066b232d11ede948c During handling of the above exception, another exception occurred:
@4000000066b232d11edea044 
@4000000066b232d11ee16734 Traceback (most recent call last):
@4000000066b232d11ee172ec   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 754, in publish_battery
@4000000066b232d11ee1828c     result = self.battery.refresh_data()
@4000000066b232d11ee18a5c   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 127, in refresh_data
@4000000066b232d11ee19614     result_status = self.read_status_data()
@4000000066b232d11ee1a1cc   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 209, in read_status_data
@4000000066b232d11ee21ae4     data = self.read_serial_data_seplos(
@4000000066b232d11ee2269c   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 300, in read_serial_data_seplos
@4000000066b232d11ee2363c     with serial.Serial(self.port, baudrate=self.baud_rate, timeout=1) as ser:
@4000000066b232d11ee241f4   File "/usr/lib/python3.8/site-packages/serial/serialutil.py", line 240, in __init__
@4000000066b232d11ee4b6dc     self.open()
@4000000066b232d11ee4beac   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 268, in open
@4000000066b232d11ee4ca64     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
@4000000066b232d11ee4da04 serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB6: [Errno 2] No such file or directory: '/dev/ttyUSB6'
```

Since this is a OS problem and not a driver problem I will close this issue.

You could try to open a question in the Victron Community.

closed 2024-08-07 by mr-manuel

mactoolz, 2024-08-07:
but this is not correct, the serial is available !!!

mactoolz, 2024-08-07:
i think the driver start to early to get the serial adapter, the OS is not ready to get the USB Adapter !

mr-manuel, 2024-08-07:
You could read the docs to understand how the driver works: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/

Serial adapter is recognized -> dbus-serialbattery is tested on it -> if successful then dbus-serialbattery is bind to the serial port -> adapter or something else in the OS crashes and the adapter is not reachable anymre.

dbus-serialbattery is only started, if the serial adapter is recognized, so it is not possible to start too early.

You have to troubleshoot your system messages in this case. This is nothing for this driver.

mactoolz, 2024-08-07:
Ok, again, but if i restart the usb adapter, it works. I think the driver do not wait correctly !!!

mactoolz, 2024-08-08:
Hi,

I really have to say. All the work in honour, but you can't leave it like that and close the post. 

Then take such a constellation as an incentive and try to solve the issue. I really don't think it's right to simply claim that this is an operating system problem. Yes, I know, it's private time, you don't earn any money from it.

But I maintain that the driver is not integrating correctly at this point.

Etc etc.

But simply ignoring the issue is not fair.

mr-manuel, 2024-08-08:
Resuming again your statements for you:

1. > My solution for that is to unplug the original Venus USB RS485 adapter and plug in. 

2. > ```
    > @4000000066b232d11ede3e9c Traceback (most recent call last):
    > @4000000066b232d11ede65ac   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 265, in open
    > @4000000066b232d11ede754c     self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
    > @4000000066b232d11ede8104 FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB6'
    > @4000000066b232d11ede90a4 
    > @4000000066b232d11ede948c During handling of the above exception, another exception occurred:
    > @4000000066b232d11edea044 
    > @4000000066b232d11ee16734 Traceback (most recent call last):
    > @4000000066b232d11ee172ec   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 754, in publish_battery
    > @4000000066b232d11ee1828c     result = self.battery.refresh_data()
    > @4000000066b232d11ee18a5c   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 127, in refresh_data
    > @4000000066b232d11ee19614     result_status = self.read_status_data()
    > @4000000066b232d11ee1a1cc   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 209, in read_status_data
    > @4000000066b232d11ee21ae4     data = self.read_serial_data_seplos(
    > @4000000066b232d11ee2269c   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 300, in read_serial_data_seplos
    > @4000000066b232d11ee2363c     with serial.Serial(self.port, baudrate=self.baud_rate, timeout=1) as ser:
    > @4000000066b232d11ee241f4   File "/usr/lib/python3.8/site-packages/serial/serialutil.py", line 240, in __init__
    > @4000000066b232d11ee4b6dc     self.open()
    > @4000000066b232d11ee4beac   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 268, in open
    > @4000000066b232d11ee4ca64     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
    > @4000000066b232d11ee4da04 serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB6: [Errno 2] No such file or directory: '/dev/ttyUSB6'
    > ```

3. > After unplug and plug in is everything working.

    All this indicated, that the serial to USB adapter is crashing and not available anymore to the OS. Again for the last time: As soon as the OS recognizes a serial adapter it starts the driver. NO serial adapter recognized -> NO driver started. It cannot start to early, it's impossible.

    For sure you checked, that the USB to serial adapter is still reachable under `/dev/ttyUSB*` after it crashed, right? `ls -l /dev/ttyUSB*`

4. > The cpu load is cery high, for each of them serial adapter the cpu load goes up for 6% ...

    Probably you meant "very" high. The Cerbo GX is one of the weaker devices and I'm sure that it cannot handle the load. Even other users have load problems with a much more powerful Raspberry Pi 4 with 6 batteries.

    But it seems to me, that you do not want to read the documentation and troubleshooting the problem. Unfortunately in this case I'm reaching my possibilities to help you.

    [How to troubleshoot high CPU load?](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#how-to-troubleshoot-high-cpu-load)

    [How the driver works](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#how-the-driver-works)

    [How many USB to serial adapters can I connect?](https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/faq/#how-many-usb-to-serial-adapters-can-i-connect)

5. > The number of 7 adapters is a bit more than normal, but it should be possible.

    Please tell me more about this. Did you make a research? Please post some links to the results and sources.

    On over 9.000 installed systems only about 40 systems have more than 5 batteries.

6. > But I maintain that the driver is not integrating correctly at this point.

    This tells me that you understood how the driver and the integration with the OS works, as well as how the OS handles the USB to serial adapters. Feel free to open a PR with the solution.

You also ignored my recommendation:

> You could try to open a question in the Victron Community.

Please post a link to the question you made in the Victron Community yesterday.

> I really have to say. All the work in honour, but you can't leave it like that and close the post.

That could interest you: https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/50#issuecomment-2220120814

By the way, if you honour all the work, did you ever made a donation?