Louisvdw/dbus-serialbattery #1109: Using bluetooth to connect to JkBMS draws over 20W

Elyasb14, 2024-08-14:
### Describe the problem

Hi, I am using a Victron Cerbo GX to connect to a JKBMS via bluetooth. When doing this, I am seeing an over 20W draw extra. We are working with a tight power budget, and I feel like we shouldn't be seeing this much of a power draw. I see the driver polls every 1 second, which seems fast to me. Could we change this to decrease the power draw? 

### Driver version

1.3.20240705

### Venus OS device type

Cerbo GX

### Venus OS version

v3.34

### BMS type

JKBMS / Heltec BMS

### Cell count

4s

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
; --------- BMS specific settings ---------

; -- Unique ID settings
; Some already assembled BMS have no unique ID and no possibility to set one. In this case
; you can use the USB port as the unique ID.
; It may be possible that VRM ID's and custom names are not saved/restored correctly in this case.
USE_PORT_AS_UNIQUE_ID = False

; -- LltJbd settings
; SoC low levels
; Note:
;     SOC_LOW_WARNING is also used to calculate the Time-To-Go even if you are not using a LltJbd BMS
SOC_LOW_WARNING = 20
SOC_LOW_ALARM   = 10

; -- Daly settings
; Battery capacity (amps), if the BMS does not support reading it
BATTERY_CAPACITY = 50
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = 1

; -- JKBMS settings
; Predefines cell count for Jkbms_can
; The cell count should be auto-detected by identifying the highest cell number,
; but this process may be sometimes slow what could cause that cells voltage is not not
; updated in VenusOS. Try this workaround if you experience problems with cell voltage.
JKBMS_CAN_CELL_COUNT = 1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT    = 15

; -- HeltecModbus (Heltec SmartBMS/YYBMS) settings
; Set the Modbus addresses from the adapters
; Separate each address to check by a comma like: 1, 2, 3, ...
; factory default address will be 1
HELTEC_MODBUS_ADDR = 1

; -- Seplos V3 settings
; Use min/max cell voltage, CVL, CCL and DCL from the BMS
SEPLOS_USE_BMS_VALUES = False

; --------- Voltage drop ---------
; If you have a voltage drop between the BMS and the charger because of wire size or length
; then you can specify the voltage drop here. The driver will then add the voltage drop
; to the calculated CVL to compensate.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 V and the battery is now charged until the charger reaches 55.20 V.
;     The BMS now measures 55.05 V since there is a voltage drop of 0.15 V on the cable.
;     Since the dbus-serialbattery reads the voltage of 55.05 V from the BMS the max voltage
;     of 55.20 V is never reached and max voltage is kept forever.
;     By setting the VOLTAGE_DROP to 0.15 V the voltage on the charger is increased and the
;     target voltage on the BMS is reached.
VOLTAGE_DROP = 0.00
```

### Relevant log output

```shell
024-08-14 19:08:28.996257500 INFO:SerialBattery:BAT: JKBMS 11.XW 4 cells (20240510)                                                       [0/1935]
2024-08-14 19:08:29.041518500 INFO:SerialBattery:publish config values = False
2024-08-14 19:08:29.042698500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c84780097290
2024-08-14 19:08:29.043931500 INFO:SerialBattery:========== Settings ==========
2024-08-14 19:08:29.044986500 INFO:SerialBattery:> Connection voltage: 13.5 V | Current: 14.8 A | SoC: 85%
2024-08-14 19:08:29.045943500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 8
2024-08-14 19:08:29.048666500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-14 19:08:29.049840500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-14 19:08:29.050844500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-08-14 19:08:29.051825500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 25.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A (read from BMS)
2024-08-14 19:08:29.052916500 INFO:SerialBattery:> CVCM:     True
2024-08-14 19:08:29.053922500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-14 19:08:29.055058500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-14 19:08:29.055996500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-14 19:08:29.056953500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-14 19:08:29.057899500 INFO:SerialBattery:Serial Number/Unique Identifier: 3101644851
2024-08-14 19:09:32.168350500
2024-08-14 19:09:32.168359500 INFO:Bluetooth details
2024-08-14 19:09:32.201999500 Attempting to disconnect from C8:47:80:09:72:90
2024-08-14 19:09:32.202009500 Successful disconnected
2024-08-14 19:09:37.261777500 Device C8:47:80:09:72:90 (public)
2024-08-14 19:09:37.261786500   Alias: Packout 1
2024-08-14 19:09:37.261789500   Paired: no
2024-08-14 19:09:37.261790500   Trusted: no
2024-08-14 19:09:37.261792500   Blocked: no
2024-08-14 19:09:37.261793500   Connected: no
2024-08-14 19:09:37.261794500   LegacyPairing: no
2024-08-14 19:09:37.261796500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-08-14 19:09:37.261799500   RSSI: -48
2024-08-14 19:09:37.262219500
2024-08-14 19:09:39.542689500 INFO:SerialBattery:
2024-08-14 19:09:39.543425500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-14 19:09:39.547512500 INFO:SerialBattery:Venus OS v3.34
2024-08-14 19:09:39.548193500 INFO:SerialBattery:dbus-serialbattery v1.3.20240705
2024-08-14 19:09:56.041575500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:09:72:90
2024-08-14 19:09:56.042293500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:09:72:90
2024-08-14 19:10:02.554077500 INFO:SerialBattery:BAT: JKBMS 11.XW 4 cells (20240510)
2024-08-14 19:10:02.555392500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-08-14 19:10:05.933741500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-08-14 19:10:06.206297500 INFO:SerialBattery:DeviceInstance = 1
2024-08-14 19:10:06.208285500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-08-14 19:10:06.209068500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-08-14 19:10:06.210017500 INFO:SerialBattery:com.victronenergy.battery.ble_c84780097290
2024-08-14 19:10:06.210965500 INFO:SerialBattery:BAT: JKBMS 11.XW 4 cells (20240510)
2024-08-14 19:10:06.240908500 INFO:SerialBattery:publish config values = False
2024-08-14 19:10:06.241729500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from ble_c84780097290
2024-08-14 19:10:06.242403500 INFO:SerialBattery:========== Settings ==========
2024-08-14 19:10:06.243148500 INFO:SerialBattery:> Connection voltage: 13.5 V | Current: 14.5 A | SoC: 85%
2024-08-14 19:10:06.243745500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 8
2024-08-14 19:10:06.244317500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-14 19:10:06.245124500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-14 19:10:06.246903500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-08-14 19:10:06.247757500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 25.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A (read from BMS)
2024-08-14 19:10:06.248414500 INFO:SerialBattery:> CVCM:     True
2024-08-14 19:10:06.249082500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-14 19:10:06.249722500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-14 19:10:06.250314500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-14 19:10:06.250907500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-14 19:10:06.251495500 INFO:SerialBattery:Serial Number/Unique Identifier: 3101644851
```

Elyasb14, 2024-08-14:
SOLVED: we were previously using a smart shunt as a battery monitor. When we hooked up the BMS to the Cerbo, it was showing the incorrect DC load. To fix this we changed the "smart shunt monitor mode" to "dc energy meter".

closed 2024-08-14 by Elyasb14

mr-manuel, 2024-08-16:
Thanks for sharing the result.