Louisvdw/dbus-serialbattery #1110: DALY BMS #8 internal calculation error on one of 2 Batteries 48V

Approved-OK, 2024-08-17:
### Describe the problem

Hallo,
using 2 DIY Batteries in Parallel , connected via rs485 (tty0 and tty1) to victron cerbo.
Battery Agreggator

[image]

System is working fine for a few days, but sometimes it brakes.
tail command shows error in battery.py line 573 which refers to battery-temperature. Battery temp is about 26degrees
first battery (tty0) works fine at this time.

restarting the driver resolves for some time ( sometime days, sometime only a few ours )

[image]

[image]

### Driver version

v1.3.20240705

### Venus OS device type

Cerbo GX

### Venus OS version

v3.41

### BMS type

Daly Smart BMS

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

;After Change in Putty:
;bash /data/etc/dbus-serialbattery/reinstall-local.sh
;bash /data/etc/dbus-serialbattery/restart-driver.sh

; tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
; tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB1/current | tai64nlocal

;https://github.com/mr-manuel/venus-os_dbus-serialbattery/tree/dev/gui-v2
;https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 40.0
[... 380 lines cut ...]
2024-08-16 17:03:36.070726500 INFO:SerialBattery:========== Settings ==========
2024-08-16 17:03:36.071440500 INFO:SerialBattery:> Connection voltage: 53.1 V | Current: 0.0 A | SoC: 80.7%
2024-08-16 17:03:36.072165500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-08-16 17:03:36.072806500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-16 17:03:36.073467500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.950 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.390 V
2024-08-16 17:03:36.074102500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 40.0 A
2024-08-16 17:03:36.074736500 INFO:SerialBattery:> CVCM:     True
2024-08-16 17:03:36.075299500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-16 17:03:36.076086500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-16 17:03:36.076811500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-08-16 17:03:36.077488500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-08-16 17:03:36.078167500 INFO:SerialBattery:Serial Number/Unique Identifier: 20221229
2024-08-16 19:43:19.757229500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-16 20:45:31.283490500 WARNING:SerialBattery:No data received in read_alarm_data()
2024-08-16 20:45:31.627255500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
```

### Any other information that may be helpful

[image]
[image]
[image]
[image]
[image]
[image]

mr-manuel, 2024-08-17:
Hello,

thanks for providing all relevant informations! Please install the latest nightly, since the error handling is different now.

Approved-OK, 2024-08-17:
i installed nightly build - due to weekend i've time to look at behavior ;-)
thx

Approved-OK, 2024-08-17:
Startup - look good

2024-08-17 07:44:46.410850500 INFO:SerialBattery:
2024-08-17 07:44:46.412694500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-17 07:44:46.437535500 INFO:SerialBattery:Venus OS v3.41
2024-08-17 07:44:46.438297500 INFO:SerialBattery:dbus-serialbattery v1.4.20240806dev
2024-08-17 07:45:02.448620500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-17 07:45:02.449759500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-08-17 07:45:02.904288500 INFO:SerialBattery:Connection established to Daly
2024-08-17 07:45:08.271834500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-08-17 07:45:08.936904500 INFO:SerialBattery:DeviceInstance = 1
2024-08-17 07:45:08.940149500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-08-17 07:45:08.940960500 INFO:SerialBattery:Used DeviceInstances = ['6', '1']
2024-08-17 07:45:08.941997500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1__0
2024-08-17 07:45:08.992286500 INFO:SerialBattery:publish config values = True
2024-08-17 07:45:09.116605500 INFO:SerialBattery:Polling data every 3.000 s
2024-08-17 07:45:09.123546500 INFO:SerialBattery:Battery Daly connected to dbus from /dev/ttyUSB1
2024-08-17 07:45:09.124641500 INFO:SerialBattery:========== Settings ==========
2024-08-17 07:45:09.130156500 INFO:SerialBattery:> Connection voltage: 52.7 V | Current: 1.1900000000000002 A | SoC: 38.8%
2024-08-17 07:45:09.139241500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-08-17 07:45:09.140426500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-17 07:45:09.142214500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.950 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.390 V
2024-08-17 07:45:09.148608500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 40.0 A
2024-08-17 07:45:09.151997500 INFO:SerialBattery:> CVCM:     True
2024-08-17 07:45:09.154656500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-17 07:45:09.155728500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-17 07:45:09.156783500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-08-17 07:45:09.157971500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-08-17 07:45:09.159564500 INFO:SerialBattery:Serial Number/Unique Identifier: 20221229
2024-08-17 07:46:26.160933500 *** CCGX booted (0) ***
2024-08-17 07:46:45.934958500 Traceback (most recent call last):
2024-08-17 07:46:45.934968500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 16, in <module>
2024-08-17 07:46:45.934972500     from dbushelper import DbusHelper
2024-08-17 07:46:45.934975500 ModuleNotFoundError: No module named 'dbushelper'
2024-08-17 07:47:26.729421500 INFO:SerialBattery:
2024-08-17 07:47:26.736971500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-17 07:47:26.748078500 INFO:SerialBattery:Venus OS v3.41
2024-08-17 07:47:26.749046500 INFO:SerialBattery:dbus-serialbattery v1.4.20240806dev
2024-08-17 07:47:42.766423500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-17 07:47:42.767703500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-08-17 07:47:43.235930500 INFO:SerialBattery:Connection established to Daly
2024-08-17 07:47:52.084911500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-08-17 07:47:53.149214500 INFO:SerialBattery:DeviceInstance = 1
2024-08-17 07:47:53.160038500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-08-17 07:47:53.163597500 INFO:SerialBattery:Used DeviceInstances = ['6', '1']
2024-08-17 07:47:53.168755500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1__0
2024-08-17 07:47:53.340198500 INFO:SerialBattery:publish config values = True
2024-08-17 07:47:53.438629500 INFO:SerialBattery:Polling data every 3.000 s
2024-08-17 07:47:53.440833500 INFO:SerialBattery:Battery Daly connected to dbus from /dev/ttyUSB1
2024-08-17 07:47:53.441692500 INFO:SerialBattery:========== Settings ==========
2024-08-17 07:47:53.442719500 INFO:SerialBattery:> Connection voltage: 52.6 V | Current: -1.1199999999999999 A | SoC: 38.8%
2024-08-17 07:47:53.467183500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-08-17 07:47:53.467192500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-17 07:47:53.467342500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.950 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.390 V
2024-08-17 07:47:53.468340500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0 A | MAX BATTERY DISCHARGE CURRENT: 40.0 A
2024-08-17 07:47:53.471735500 INFO:SerialBattery:> CVCM:     True
2024-08-17 07:47:53.471743500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-17 07:47:53.471746500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-17 07:47:53.471749500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-08-17 07:47:53.472574500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-08-17 07:47:53.473446500 INFO:SerialBattery:Serial Number/Unique Identifier: 20221229

Approved-OK, 2024-08-17:
severel crashes 

[image]

2024-08-17 10:16:20.304341500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:20.331153500 DEBUG:SerialBattery:logged to dbus [48.9]
2024-08-17 10:16:20.332873500 DEBUG:SerialBattery:Cells:[1]3.358V [2]3.364V [3]3.358V [4]3.364V [5]3.363V [6]3.357V [7]3.364V [8]3.359V [9]3.37V [10]3.361V [11]3.356V [12]3.36V [13]3.361V [14]3.361V [15]3.359V [16]3.361V
2024-08-17 10:16:20.334216500 DEBUG:SerialBattery:Polling data took 0.797 seconds
2024-08-17 10:16:22.629124500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-17 10:16:22.708639500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-17 10:16:23.254556500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-17 10:16:23.261876500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:23.261885500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:23.274399500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:23.274408500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:23.274411500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:23.386353500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:23.387297500 DEBUG:SerialBattery:current: 10.8219 - current_external: None
2024-08-17 10:16:23.400859500 DEBUG:SerialBattery:logged to dbus [48.9]
2024-08-17 10:16:23.402257500 DEBUG:SerialBattery:Cells:[1]3.358V [2]3.365V [3]3.358V [4]3.365V [5]3.363V [6]3.358V [7]3.364V [8]3.36V [9]3.37V [10]3.361V [11]3.357V [12]3.361V [13]3.361V [14]3.361V [15]3.359V [16]3.362V
2024-08-17 10:16:23.403380500 DEBUG:SerialBattery:Polling data took 0.863 seconds
2024-08-17 10:16:25.632514500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-17 10:16:25.708609500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-17 10:16:26.243966500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
**### 2024-08-17 10:16:26.247177500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("'>' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #710
2024-08-17 10:16:26.249453500 WARNING:root:calcMaxChargeCurrentReferringToTemperature(): get_max_temp() is None or get_min_temp() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-17 10:16:26.251469500 WARNING:root:calcMaxDischargeCurrentReferringToTemperature(): get_max_temp() is None or get_min_temp() is None, using default current instead. If you don't see this warning very often, you can ignore it.**
2024-08-17 10:16:26.275210500 DEBUG:SerialBattery:current: None - current_external: None
2024-08-17 10:16:26.276943500 DEBUG:SerialBattery:current: None - current_external: None
2024-08-17 10:16:26.449980500 DEBUG:SerialBattery:current: None - current_external: None
2024-08-17 10:16:26.455853500 DEBUG:SerialBattery:logged to dbus [48.9]
2024-08-17 10:16:26.457338500 DEBUG:SerialBattery:Cells:
2024-08-17 10:16:26.462760500 DEBUG:SerialBattery:Polling data took 0.919 seconds
2024-08-17 10:16:28.618740500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-17 10:16:28.691011500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-17 10:16:29.282308500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-17 10:16:29.286321500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("'>' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #710
2024-08-17 10:16:29.287551500 DEBUG:SerialBattery:current: None - current_external: None
2024-08-17 10:16:29.290471500 DEBUG:SerialBattery:current: None - current_external: None
2024-08-17 10:16:29.355270500 DEBUG:SerialBattery:current: None - current_external: None
2024-08-17 10:16:29.356333500 DEBUG:SerialBattery:logged to dbus [48.9]
2024-08-17 10:16:29.362206500 DEBUG:SerialBattery:Cells:[1]3.359V [2]3.365V [3]3.359V [4]3.365V [5]3.364V [6]3.358V [7]3.365V [8]3.36V [9]3.37V [10]3.362V [11]3.357V [12]3.361V [13]3.361V [14]3.362V [15]3.359V [16]3.362V
2024-08-17 10:16:29.366064500 DEBUG:SerialBattery:Polling data took 0.820 seconds
2024-08-17 10:16:31.629226500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-17 10:16:31.709951500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-17 10:16:32.241534500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-17 10:16:32.243823500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("'>' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #710

Approved-OK, 2024-08-18:
So it seems, when loading the batteries the failure happens

2024-08-18 09:24:40.674158500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("'>' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #710
2024-08-18 09:24:43.645287500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("'>' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #710
2024-08-18 09:24:46.636793500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("'>' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #710
2024-08-18 09:24:49.648909500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("'>' not supported between instances of 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #710
[image]

[image]

[image]

Enrico126, 2024-08-21:
Hey,

same error for me:
[image]

The error only occurs when the battery is being charged.
I also installed the latest nightly.

**Driver version**
v1.4.2024.08.06dev

**Venus OS device type**
Cerbo GX

**Venus OS version**
v3.41

**BMS type**
Daly Smart BMS

**Cell count**
22

**Battery count**
1

**Connection type**
Serial USB adapter to RS485

seamaster101, 2024-08-24:
Any solution to this problem?

mr-manuel, 2024-08-24:
Daly BMS are not very stable in communication. Until now the tons of errors were supressed and the user never got notified, even if 80% of the BMS readings failed. Seems like there is no other way to continue this way for Daly BMS and ignore the lots of faulty readings.

Approved-OK, 2024-08-24:
Good Morning.
at least no solution. 
I deactivated CVL calculation Mode to False 
CVL_ICONTROLLER_MODE   = False
CVL_ICONTROLLER_FACTOR = 0.2

it seems to be better, but today i saw this
[image]
 
 this is something irritating ?
 But the driver ( nightly build ) seam to work better.
 
 Looking today, weather seams to be very good today in Salzburg..

mr-manuel, 2024-08-24:
`v1.3.20240705` reports every error that occurs with internal error.
`v1.4.20240806dev` reports after 100 occurred errors in 1 hour. I will change this so that it could set by every individual BMS class.

mr-manuel, 2024-08-24:
Please install `v1.4.20240824dev`, I added some error handling for Daly BMS.

seamaster101, 2024-08-24:
For me this happens with JK BMS but I will try to install the version you recommended and will report. Jordan

mr-manuel, 2024-08-24:
@seamaster101 then yoi have a different issue. Please open a new issue for it.