Louisvdw/dbus-serialbattery #1111: #8 calculation error

ikorpershoek, 2024-08-17:
### Describe the problem

After every cerbo reset this alarm is raised. 

### Driver version

Latest official release

### Venus OS device type

Cerbo GX

### Venus OS version

v350-13 / 3.41

### BMS type

JKBMS / Heltec BMS

### Cell count

8

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
standard config file
```

### Relevant log output

```shell
#8 Internal calculation error
```

mr-manuel, 2024-08-17:
Unfortunately you did not read the instructions and did not orovide the needed informations. This way I cannot help you. Please try again to read what you have to provide and where you can find it

ikorpershoek, 2024-08-18:
I'm sorry I tried to provide all the info as far as I understood the questions. It's a plain install of the driver no changes made. So I thought it was enough. I will reread the instructions.

RichGorrod, 2024-08-18:
I have exactly the same issue. CheersRich. Rich Gorrod07768453552

mr-manuel, 2024-08-18:
@RichGorrod since this is a very general error message it's not very likely that you have exactly the same issue. Please read the docs.

Becker884, 2024-08-18:
Just installed Venus 3,41 today with newest serialbatt stable - same problem here.
JK BMS over RS485 with 18x 280Ah cells.

Becker884, 2024-08-18:
i tried v1.4 nighty:
`@4000000066c24bda1549d634 ERROR:SerialBattery:**CONFIG ISSUE**: Minimum value of CELL_VOLTAGES_WHILE_DISCHARGING (3.0 V) is higher than MIN_CELL_VOLTAGE (2.9 V). MIN_CELL_VOLTAGE will never be reached this way. Please check the configuration.`
I changed MIN_CELL_VOLTAGE to 3.000V -> no errors.

maybe this was the same problem in v1.3 ?

Becker884, 2024-08-18:
After a new reboot again error #8 

[image]

Becker884, 2024-08-19:
It´s random:
[image]
sometimes error #8 comes after reboot, sometimes not.
even with fresh installed serialbatt without any modification

mr-manuel, 2024-08-19:
@Becker884 since this is a very general error message it's not very likely that you have exactly the same issue. Open a new issue for your problem.

seamaster101, 2024-08-19:
I also have the same problem with all 4 of my batteries 
[image]

mr-manuel, 2024-08-21:
Since no one of you read all the instructions and provided correct and needed informations I will close this issue. Please do so when opening a new issue.

closed 2024-08-21 by mr-manuel

seamaster101, 2024-08-21:
Don’t be frustrated like that please! Keep the great work you guys have been doing.  What information specifically do you need?Jordan