Louisvdw/dbus-serialbattery #1112: No SoC value from BMS and serialbattery crashes: "SoC: None%" - AttributeError: 'NoneType' object has no attribute 'startswith'

a1ad, 2024-08-19:
### Describe the bug

Battery gets disconnected in the OS. It was working fine for months.

[image]

[image]

[image]

As you can see in the logs: the soc value is None and the software terminates.

### How to reproduce

At every dbus-serialbattery (re)start

### Expected behavior

Battery connect

### Driver version

v1.2.20240408

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

13

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
;met rommel in parallel
MAX_BATTERY_CHARGE_CURRENT    = 60
;standalone berta
;MAX_BATTERY_CHARGE_CURRENT    = 55
MAX_BATTERY_DISCHARGE_CURRENT = 50

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used to calculate the min/max battery voltage
; Example:
;     16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
[... 381 lines cut ...]
@4000000066c39ce9289317b4 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
@4000000066c39ce92899ca8c INFO:SerialBattery:Used DeviceInstances = ['1']
@4000000066c39ce928a0a474 INFO:SerialBattery:com.victronenergy.battery.ttyUSB2
@4000000066c39ce92a28ea54 INFO:SerialBattery:publish config values = True
@4000000066c39ce92b6da33c INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB2
@4000000066c39ce92b740fc4 INFO:SerialBattery:========== Settings ==========
@4000000066c39ce92b7b85ec INFO:SerialBattery:> Connection voltage: 52.63V | Current: -1.14A | SoC: None%
@4000000066c39ce92b81f65c INFO:SerialBattery:> Cell count: 13 | Cells populated: 13
@4000000066c39ce92b884b74 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
@4000000066c39ce92b8f255c INFO:SerialBattery:> MIN CELL VOLTAGE: 3.05V | MAX CELL VOLTAGE: 4.16V
@4000000066c39ce92b95e3ec INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 60.0A | MAX BATTERY DISCHARGE CURRENT: 50.0A
@4000000066c39ce92b9c6fb4 INFO:SerialBattery:> CVCM:     True
@4000000066c39ce92ba31ea4 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  False
@4000000066c39ce92ba9a684 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
@4000000066c39ce92bb0518c INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@4000000066c39ce92bb6ba2c INFO:SerialBattery:Serial Number/Unique Identifier: Dikkere_Bert
@4000000066c39cea33786634 Traceback (most recent call last):
@4000000066c39cea33787da4   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 811, in publish_battery
@4000000066c39cea33789514     self.battery.manage_charge_voltage()
@4000000066c39cea3378a0cc   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 241, in manage_charge_voltage
@4000000066c39cea3378b454     self.manage_charge_voltage_step()
@4000000066c39cea3378c00c   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 734, in manage_charge_voltage_step
@4000000066c39cea337a5264     if not self.charge_mode.startswith("Float"):
@4000000066c39cea337a6204 AttributeError: 'NoneType' object has no attribute 'startswith'
```

mr-manuel, 2024-08-19:
Thanks that you read carefully all the instructions. Here again a part of it:

[image]

a1ad, 2024-08-19:
Thanks for the hint, it's working with the stable v1.3.20240705. It's strange bc I never did an update, and all was working fine for months...

closed 2024-08-19 by a1ad