Louisvdw/dbus-serialbattery #1114: Discharging and Charging limits are ignored

Enrico126, 2024-08-21:
### Describe the problem

Hello,

I have the problem that the set values for max. discharging and charging are not being respected.
The system charges with full power and after a certain time the BMS disconnects (The limit in the BMS is set to 125 amps, which should not actually be reached):
[image]

DVCC: 
[image]

BMS Params:
[image]

VRM View:
[image]

### Driver version

v1.4.20240806dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.41

### BMS type

Daly Smart BMS

### Cell count

22

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
LOGGING = INFO

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

MAX_BATTERY_CHARGE_CURRENT = 55.0
MAX_BATTERY_DISCHARGE_CURRENT = 55.0

MIN_CELL_VOLTAGE   = 2.00
MAX_CELL_VOLTAGE   = 2.50
FLOAT_CELL_VOLTAGE = 2.50
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
CELL_VOLTAGES_WHILE_CHARGING      = 2.52, 2.48, 2.45, 2.44
MAX_CHARGE_CURRENT_CV_FRACTION    =    0, 0.05,  0.5,    1
CELL_VOLTAGES_WHILE_DISCHARGING   = 1.98, 2.00, 2.14, 2.20
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1
BMS_TYPE = Daly
SOC_LOW_WARNING = 10
[... 284 lines cut ...]
2024-08-21 10:45:22.785752500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-21 10:45:23.549293500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-21 10:45:23.552110500 ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #810
2024-08-21 10:45:23.814059500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-21 10:45:23.907044500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-21 10:45:24.638221500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-21 10:45:24.640905500 ERROR:SerialBattery:Non blocking exception occurred: TypeError('unsupported format string passed to NoneType.__format__') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #810
2024-08-21 10:45:29.680318500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-21 10:45:35.258272500 WARNING:SerialBattery:No data received in read_alarm_data()
2024-08-21 10:45:35.702358500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-21 10:45:41.967180500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-21 10:45:42.067027500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-21 10:45:42.811065500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-21 10:45:43.067094500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-21 10:45:43.156873500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-21 10:45:43.912383500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-21 10:45:44.057308500 WARNING:SerialBattery:read_soc_data - triesValid 1
2024-08-21 10:45:44.146863500 WARNING:SerialBattery:read_soc_data - triesValid 0
2024-08-21 10:45:44.931353500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
2024-08-21 10:46:11.993574500 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
```

### Any other information that may be helpful

Active Battery Monitor is the MultiPlus 2 one.

mr-manuel, 2024-08-21:
What do you have selected as the controlling BMS at the end of the DVCC page?

Enrico126, 2024-08-21:
Hey,

i have the following selected:
[image]

I also have this alarm:
[image]

mr-manuel, 2024-08-24:
Please install `v1.4.20240824dev`, I added some error handling for Daly BMS.

Enrico126, 2024-08-24:
Hey,
thanks for the quick response. Unfortunately, I am not able to get the BMS Connection to run with this version:

```
2024-08-24 12:20:24.140326500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-24 12:20:24.142351500 INFO:SerialBattery:Venus OS v3.41
2024-08-24 12:20:24.143085500 INFO:SerialBattery:dbus-serialbattery v1.4.20240824dev
2024-08-24 12:20:40.158470500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-24 12:20:40.159334500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-08-24 12:20:40.366824500 WARNING:SerialBattery:read_soc_data - try #2: voltage: 550, current: -7.0, soc: 35.1
2024-08-24 12:20:40.459707500 WARNING:SerialBattery:read_soc_data - try #3: voltage: 550, current: -7.1, soc: 35.1
2024-08-24 12:20:40.549716500 WARNING:SerialBattery:read_soc_data - try #4: voltage: 550, current: -7.1, soc: 35.1
2024-08-24 12:20:40.553480500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-24 12:20:40.554402500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-08-24 12:20:40.759765500 WARNING:SerialBattery:read_soc_data - try #2: voltage: 550, current: -7.2, soc: 35.1
2024-08-24 12:20:40.859822500 WARNING:SerialBattery:read_soc_data - try #3: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:40.949696500 WARNING:SerialBattery:read_soc_data - try #4: voltage: 550, current: -7.2, soc: 35.1
2024-08-24 12:20:40.953127500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-24 12:20:41.454501500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-08-24 12:20:41.455392500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-08-24 12:20:41.656755500 WARNING:SerialBattery:read_soc_data - try #2: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:41.749701500 WARNING:SerialBattery:read_soc_data - try #3: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:41.836841500 WARNING:SerialBattery:read_soc_data - try #4: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:41.840916500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-24 12:20:41.841784500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-08-24 12:20:42.036719500 WARNING:SerialBattery:read_soc_data - try #2: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:42.130035500 WARNING:SerialBattery:read_soc_data - try #3: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:42.251718500 WARNING:SerialBattery:read_soc_data - try #4: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:42.255831500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-24 12:20:42.757471500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-08-24 12:20:42.758377500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-08-24 12:20:42.946715500 WARNING:SerialBattery:read_soc_data - try #2: voltage: 550, current: -7.2, soc: 35.1
2024-08-24 12:20:43.125823500 WARNING:SerialBattery:read_soc_data - try #4: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:43.130056500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-24 12:20:43.130938500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-08-24 12:20:43.349720500 WARNING:SerialBattery:read_soc_data - try #2: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:43.437114500 WARNING:SerialBattery:read_soc_data - try #3: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:43.530238500 WARNING:SerialBattery:read_soc_data - try #4: voltage: 550, current: -7.3, soc: 35.1
2024-08-24 12:20:43.533925500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-24 12:20:44.035710500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0

```

With the LTS or v1.4.20240806dev installed it is working.

mr-manuel, 2024-08-24:
Indeed I made a mistake. Please install again.

Enrico126, 2024-08-24:
Now the installation worked. I will take a look the next days.

Enrico126, 2024-08-27:
Hey, so far everything looks good with the new driver. So I'll close the ticket for now, there are no more errors in the logs and the limits are also respected.

mr-manuel, 2024-08-27:
You are welcome.

Enrico126, 2024-08-29:
Unfortunately, I have to reopen the ticket.
I think the problem occurred today during discharging, with a little more load
[image]

```
  soc_reset_in_days = round(
      utils.SOC_RESET_AFTER_DAYS - soc_reset_days_ago, 2
  )
```

According to the logs it looks like there is a problem with the value `utils.SOC_RESET_AFTER_DAYS` (undefined)?
I haven't currently set the value in my `config.ini`, but it should have the default value (i guess)?

And subtraction between float and undefined is of course not possible...
I will try to set a value in the `config.ini` and see what happens

mr-manuel, 2024-08-30:
I don‘t think the code you posted matches the version you are using. Which version have you installed?

mr-manuel, 2024-08-30:
Please install the latest nightly version. I added some more checks for the Daly BMS. Seems that it happens very often, that the BMS provides `None` values instead of the correct one.

Enrico126, 2024-08-31:
Yes, you are right. 

I only checked the code in on the current main branch on git..
I installed the latest nightly version, it looks good for now.

Thanks!

closed 2024-08-31 by Enrico126