Louisvdw/dbus-serialbattery #1115: BLE JBD BMS - BleakClient(): Exception occurred

mkerner, 2024-08-25:
### Describe the problem

Driver never connects to battery but does see the appropriate battery. 

### Driver version

v1.3.20240705

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.41

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used for:
;         - Calculating the min/max battery voltage
;         - Trigger the SoC reset when SoC calculation is enabled
; Example:
;     16 cells * 3.45 V/cell = 55.2 V max charge voltage.
;     16 cells * 2.90 V/cell = 46.4 V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage (needs to match BMS settings) ---------
; +++ This has nothing to do with "SOC calculation" in a section below +++
; This is one of the possibilities to reset the SoC to 100%, because of SoC drift.
; Description:
;     May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;     Some BMS may needed to reset the SoC to 100% once in a while, because of SoC drift. Some
;     devices, like JKBMS, will reset their internal SOC value, if they reach the upper voltage level.
;     Using this method, the charging voltage can be raised regularly to achieve that.
[... 477 lines cut ...]
2024-08-25 12:50:29.145118500 [CHG] Device 77:BA:C2:C3:EA:BF TxPower is nil
2024-08-25 12:50:29.145168500 [CHG] Device 77:BA:C2:C3:EA:BF RSSI is nil
2024-08-25 12:50:29.145172500 [CHG] Device 24:00:00:00:60:5F RSSI is nil
2024-08-25 12:50:29.145176500 [CHG] Device 4A:CF:62:2C:B8:EC RSSI is nil
2024-08-25 12:50:29.145179500 [CHG] Device 77:92:47:D8:2F:44 RSSI is nil
2024-08-25 12:50:29.145183500 [CHG] Device F0:B3:EC:0E:37:82 TxPower is nil
2024-08-25 12:50:29.145187500 [CHG] Device F0:B3:EC:0E:37:82 RSSI is nil
2024-08-25 12:50:29.145212500 [CHG] Device A4:C1:38:04:72:84 RSSI is nil
2024-08-25 12:50:29.145216500 [CHG] Device 77:35:CB:D6:F6:CF TxPower is nil
2024-08-25 12:50:29.145220500 [CHG] Device 77:35:CB:D6:F6:CF RSSI is nil
2024-08-25 12:50:29.145223500 [CHG] Device 32:00:00:00:3C:D3 RSSI is nil
2024-08-25 12:50:29.145227500 [CHG] Device D8:60:BD:DF:1A:8E RSSI is nil
2024-08-25 12:50:29.145231500 [CHG] Controller 43:45:C0:00:1F:AC Discovering: no
2024-08-25 12:50:29.145256500 Successful disconnected
2024-08-25 12:50:34.297447500 Device A4:C1:38:04:72:84 (public)
2024-08-25 12:50:34.297454500 	Alias: 280ah
2024-08-25 12:50:34.297456500 	Paired: no
2024-08-25 12:50:34.297458500 	Trusted: no
2024-08-25 12:50:34.297460500 	Blocked: no
2024-08-25 12:50:34.297462500 	Connected: no
2024-08-25 12:50:34.297464500 	LegacyPairing: no
2024-08-25 12:50:34.297466500 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-08-25 12:50:34.297470500 	RSSI: -44
2024-08-25 12:50:34.298363500
```

mkerner, 2024-08-25:
Additional debug log
```
2024-08-25 12:56:24.245315500 
2024-08-25 12:56:25.425326500 INFO:SerialBattery:
2024-08-25 12:56:25.425732500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-25 12:56:25.427100500 INFO:SerialBattery:Venus OS v3.41
2024-08-25 12:56:25.427515500 INFO:SerialBattery:dbus-serialbattery v1.3.20240705
2024-08-25 12:56:41.889901500 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:38:04:72:84
2024-08-25 12:56:41.890382500 INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:38:04:72:84
2024-08-25 12:57:06.927664500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-25 12:57:06.931206500 ERROR:SerialBattery:No BMS found at A4:C1:38:04:72:84
2024-08-25 12:57:06.931662500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2024-08-25 12:57:09.667082500 INFO:SerialBattery:BLE client disconnected
2024-08-25 12:57:09.943377500 
2024-08-25 12:57:09.943507500 INFO:Bluetooth details
2024-08-25 12:57:09.995170500 [CHG] Device A4:C1:38:04:72:84 RSSI: -44
2024-08-25 12:57:09.995178500 [CHG] Device A4:C1:38:04:72:84 RSSI: -54
2024-08-25 12:57:09.995181500 [CHG] Device A4:C1:38:04:72:84 RSSI: -40
2024-08-25 12:57:09.995185500 [CHG] Device A4:C1:38:04:72:84 RSSI: -54
2024-08-25 12:57:09.995189500 [CHG] Device A4:C1:38:04:72:84 RSSI: -44
2024-08-25 12:57:09.995192500 [CHG] Device A4:C1:38:04:72:84 RSSI: -44
2024-08-25 12:57:10.126940500 Attempting to disconnect from A4:C1:38:04:72:84
2024-08-25 12:57:10.126947500 [CHG] Device EC:FE:CD:18:9F:E4 RSSI is nil
2024-08-25 12:57:10.126951500 [DEL] Device EC:FE:CD:18:9F:E4 EC-FE-CD-18-9F-E4
2024-08-25 12:57:10.126955500 [CHG] Device 48:8D:49:9C:D0:13 TxPower is nil
2024-08-25 12:57:10.126959500 [CHG] Device 48:8D:49:9C:D0:13 RSSI is nil
2024-08-25 12:57:10.126962500 [CHG] Device 61:0D:16:6D:9C:CF RSSI is nil
2024-08-25 12:57:10.126966500 [CHG] Device 59:B2:7B:BB:96:B8 TxPower is nil
2024-08-25 12:57:10.127059500 [CHG] Device 59:B2:7B:BB:96:B8 RSSI is nil
2024-08-25 12:57:10.127063500 [CHG] Device D8:60:BD:DF:1A:8E RSSI is nil
2024-08-25 12:57:10.127067500 [CHG] Device 6A:D1:B6:BD:B9:84 RSSI is nil
2024-08-25 12:57:10.127070500 [CHG] Device D9:29:AD:6D:73:64 RSSI is nil
2024-08-25 12:57:10.127074500 [CHG] Device 71:42:B5:10:D8:1B RSSI is nil
2024-08-25 12:57:10.127078500 [CHG] Device A4:C1:38:53:CC:B9 RSSI is nil
2024-08-25 12:57:10.127104500 [CHG] Device 7C:5E:64:5F:79:9B TxPower is nil
2024-08-25 12:57:10.127108500 [CHG] Device 7C:5E:64:5F:79:9B RSSI is nil
2024-08-25 12:57:10.127112500 [CHG] Device 78:20:6C:95:62:40 TxPower is nil
2024-08-25 12:57:10.127116500 [CHG] Device 78:20:6C:95:62:40 RSSI is nil
2024-08-25 12:57:10.127119500 [CHG] Device A4:C1:38:A5:32:4A RSSI is nil
2024-08-25 12:57:10.127168500 [CHG] Device A4:C1:38:EB:10:B5 RSSI is nil
2024-08-25 12:57:10.127172500 [CHG] Device 61:D5:9E:FA:51:A4 RSSI is nil
2024-08-25 12:57:10.127176500 [CHG] Device 42:A0:DB:D1:C7:2E RSSI is nil
2024-08-25 12:57:10.127179500 [CHG] Device 4C:D0:55:6D:34:5F TxPower is nil
2024-08-25 12:57:10.127183500 [CHG] Device 4C:D0:55:6D:34:5F RSSI is nil
2024-08-25 12:57:10.127187500 [CHG] Device 77:BA:C2:C3:EA:BF TxPower is nil
2024-08-25 12:57:10.127275500 [CHG] Device 77:BA:C2:C3:EA:BF RSSI is nil
2024-08-25 12:57:10.127279500 [CHG] Device 24:00:00:00:60:5F RSSI is nil
2024-08-25 12:57:10.127283500 [CHG] Device 70:09:71:8C:C5:8B RSSI is nil
2024-08-25 12:57:10.127287500 [CHG] Device 4E:11:C9:CA:2B:6B RSSI is nil
2024-08-25 12:57:10.127290500 [CHG] Device 77:35:CB:D6:F6:CF RSSI is nil
2024-08-25 12:57:10.127294500 [CHG] Device 4A:CF:62:2C:B8:EC TxPower is nil
2024-08-25 12:57:10.127320500 [CHG] Device 4A:CF:62:2C:B8:EC RSSI is nil
2024-08-25 12:57:10.127324500 [CHG] Device 55:67:18:DB:68:97 TxPower is nil
2024-08-25 12:57:10.127328500 [CHG] Device 55:67:18:DB:68:97 RSSI is nil
2024-08-25 12:57:10.127331500 [CHG] Device CA:F2:AB:47:76:3C RSSI is nil
2024-08-25 12:57:10.127335500 [CHG] Device 32:00:00:00:3C:D3 RSSI is nil
2024-08-25 12:57:10.127386500 [CHG] Device E0:B6:A6:B1:17:88 RSSI is nil
2024-08-25 12:57:10.127390500 [CHG] Device F0:B3:EC:0E:37:82 TxPower is nil
2024-08-25 12:57:10.127393500 [CHG] Device F0:B3:EC:0E:37:82 RSSI is nil
2024-08-25 12:57:10.127397500 [CHG] Device A4:C1:38:04:72:84 RSSI is nil
2024-08-25 12:57:10.127401500 [CHG] Device 42:6A:9E:8A:93:DD RSSI is nil
2024-08-25 12:57:10.127404500 [CHG] Device 38:6A:E5:F3:22:4E RSSI is nil
2024-08-25 12:57:10.127429500 [CHG] Device 77:92:47:D8:2F:44 TxPower is nil
2024-08-25 12:57:10.127433500 [CHG] Device 77:92:47:D8:2F:44 RSSI is nil
2024-08-25 12:57:10.127437500 [CHG] Device 45:42:28:15:79:51 TxPower is nil
2024-08-25 12:57:10.127441500 [CHG] Device 45:42:28:15:79:51 RSSI is nil
2024-08-25 12:57:10.127444500 [CHG] Device 50:E4:FE:CB:89:9F TxPower is nil
2024-08-25 12:57:10.127469500 [CHG] Device 50:E4:FE:CB:89:9F RSSI is nil
2024-08-25 12:57:10.127472500 [CHG] Controller 43:45:C0:00:1F:AC Discovering: no
2024-08-25 12:57:10.127477500 Successful disconnected
2024-08-25 12:57:15.266303500 Device A4:C1:38:04:72:84 (public)
2024-08-25 12:57:15.266310500 	Alias: 280ah
2024-08-25 12:57:15.266312500 	Paired: no
2024-08-25 12:57:15.266314500 	Trusted: no
2024-08-25 12:57:15.266315500 	Blocked: no
2024-08-25 12:57:15.266317500 	Connected: no
2024-08-25 12:57:15.266319500 	LegacyPairing: no
2024-08-25 12:57:15.266321500 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-08-25 12:57:15.266325500 	RSSI: -40
```

mr-manuel, 2024-10-07:
Duplicate of https://github.com/Louisvdw/dbus-serialbattery/issues/1088.

closed 2024-10-07 by mr-manuel