Louisvdw/dbus-serialbattery #1116: EG4 Lifepower - not reading all RS485 modules on the same interface

Zeezichtje, 2024-08-28:
### Describe the bug

I am having a setup with 6 EG4 lifepower batteries. They are dipswitched 1 till 6 (verified via BMS Test program).
Only the batteries 1, 3 and 5 are successfully read by dbus-serialbattery.
Batteries 1 and 2 are 48v batteries. Batteries 3,4,5,6 are 24v batteries. Battery 3+4 and 5+6 are in series to create a 48v power bank.

I saw a user with a similair issue: https://github.com/Louisvdw/dbus-serialbattery/issues/142#issuecomment-2254960773

### How to reproduce

bash /data/etc/dbus-serialbattery/reinstall-local.sh
tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal

### Expected behavior

Batteries 1 till 6 should be visible.

### Driver version

v1.4.20240826dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.41

### BMS type

Life/Tian Power

### Cell count

16 (48v) and 8 (6v)

### Battery count

6

### Connection type

Serial USB adapter to RS485

### Config file

```ini
;MAX_BATTERY_CHARGE_CURRENT = 300.0
;MAX_BATTERY_DISCHARGE_CURRENT = 250.0
BMS_TYPE = EG4_Lifepower
MODBUS_ADDRESSES = 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0X0B, 0X0C, 0X0D, 0X0E, 0x0F, 0x10
;MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2, 0.4, 0.8,  1,  1, 0.6,

also tested with

;MAX_BATTERY_CHARGE_CURRENT = 300.0
;MAX_BATTERY_DISCHARGE_CURRENT = 250.0
BMS_TYPE = EG4_Lifepower
MODBUS_ADDRESSES = 0x01, 0x02, 0x03, 0x04, 0x05, 0x06
;MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2, 0.4, 0.8,  1,  1, 0.6,
```

### Relevant log output

```shell
root@einstein:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-08-28 02:34:58.663552500 ERROR:SerialBattery:Non blocking exception occurred: ValueError('non-hexadecimal number found in fromhex() arg at position 1') of type <class 'ValueError'> in /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py line #144
2024-08-28 02:34:59.153793500 WARNING:SerialBattery:No battery connection at /dev/ttyUSB0 and this Modbus address 0X0E
2024-08-28 02:34:59.154584500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
[... 65 lines cut ...]
2024-08-28 02:35:18.419701500 INFO:SerialBattery:> Connection voltage: 26.0 V | Current: -18.95 A | SoC: 34.6%
2024-08-28 02:35:18.420869500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2024-08-28 02:35:18.422120500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-28 02:35:18.423368500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-28 02:35:18.424583500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-08-28 02:35:18.425884500 INFO:SerialBattery:> CVCM:     True
2024-08-28 02:35:18.427083500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-28 02:35:18.428288500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-28 02:35:18.429471500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-28 02:35:18.431273500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-08-28 02:35:18.433313500 INFO:SerialBattery:Serial Number/Unique Identifier: B_182.2Ah
2024-08-28 02:35:18.434560500 INFO:SerialBattery:Battery EG4 Lifepower connected to dbus from /dev/ttyUSB0
2024-08-28 02:35:18.435850500 INFO:SerialBattery:========== Settings ==========
2024-08-28 02:35:18.437089500 INFO:SerialBattery:> Connection voltage: 25.97 V | Current: -19.35 A | SoC: 35.4%
2024-08-28 02:35:18.438232500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2024-08-28 02:35:18.439452500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-28 02:35:18.440834500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-28 02:35:18.442168500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-08-28 02:35:18.443396500 INFO:SerialBattery:> CVCM:     True
2024-08-28 02:35:18.444742500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-28 02:35:18.447127500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-28 02:35:18.448500500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-28 02:35:18.449740500 INFO:SerialBattery:> CHARGE FET: None | DISCHARGE FET: None | BALANCE FET: None
2024-08-28 02:35:18.450925500 INFO:SerialBattery:Serial Number/Unique Identifier: B_180.6Ah
```

mr-manuel, 2024-08-28:
Thanks for reading carefully all instructions and providing all the needed data.

There is a very similar issue opened at the moment, which I think could also help you: https://github.com/Louisvdw/dbus-serialbattery/issues/1104

It's a different battery, but the same BMS protocol. You can quickly test, if all batteries are recognized, if you give them only odd address numbers: `1,3,5,7,9,11`

Currently I did not figure out, why this is the case.

mr-manuel, 2024-08-28:
Are you maybe able to read the commands that your software sends, so that you can compare any differences between the request and answer commands from the even and odd addresses?

Zeezichtje, 2024-08-29:
Will this help?

Commands communicating with Battery #1.
[2024-08-28 21:42:29:575]-Tx->: 7E 01 33 00 FE 0D
[2024-08-28 21:42:29:707]-Rx<-: 7E 01 33 16 51 54 2D 59 53 30 30 2D 31 36 53 56 31 30 30 41 2D 56 33 2E 34 30 B8 0D
[2024-08-28 21:42:30:122]-Tx->: 7E 32 30 30 31 34 41 34 32 30 30 30 30 46 44 41 32 0D
[2024-08-28 21:42:30:487]-Rx<-: 7E 32 30 30 31 34 41 30 30 41 30 43 41 31 30 30 31 31 30 30 43 38 41 30 43 38 44 30 43 38 41 30 43 38 42 30 43 38 41 30 43 38 46 30 43 38 41 30 43 38 45 30 43 38 45 30 43 38 46 30 43 38 44 30 43 38 42 30 43 38 43 30 43 39 30 30 43 38 43 30 43 38 43 30 34 30 42 45 31 30 42 45 31 30 42 45 31 30 42 45 31 30 42 44 37 30 42 44 37 30 30 44 38 31 34 31 33 30 35 46 39 32 36 38 45 30 31 39 32 30 46 30 30 30 46 30 30 36 32 30 43 39 30 30 43 38 41 30 30 30 36 30 42 45 31 30 42 45 31 30 30 33 31 37 31 41 32 30 30 33 31 32 31 30 43 30 30 31 41 36 36 36 36 30 30 31 39 43 44 39 33 30 30 30 30 30 45 43 32 30 30 30 30 31 33 32 37 37 30 45 31 39 33 35 42 44 31 35 46 0D
[2024-08-28 21:42:30:705]-Tx->: 7E 32 30 30 31 34 41 34 34 30 30 30 30 46 44 41 30 0D
[2024-08-28 21:42:30:948]-Rx<-: 7E 32 30 30 31 34 41 30 30 37 30 35 34 31 30 30 31 31 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 34 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 39 30 30 30 30 30 30 30 30 30 30 30 31 30 33 30 32 30 30 30 30 30 30 30 30 30 30 45 44 43 32 0D
[2024-08-28 21:42:31:157]-Tx->: 7E 01 33 00 FE 0D

Commands communicating with Battery #2.
[2024-08-28 21:48:59:203]-Tx->: 7E 02 33 00 FC 0D
[2024-08-28 21:48:59:401]-Rx<-: 7E 02 33 16 51 54 2D 59 53 30 30 2D 31 36 53 56 31 30 30 41 2D 56 33 2E 34 30 B8 0D
[2024-08-28 21:48:59:829]-Tx->: 7E 32 30 30 32 34 41 34 32 30 30 30 30 46 44 41 31 0D
[2024-08-28 21:49:00:204]-Rx<-: 7E 32 30 30 32 34 41 30 30 41 30 43 41 31 30 30 31 31 30 30 43 39 33 30 43 39 34 30 43 39 34 30 43 39 31 30 43 39 33 30 43 39 32 30 43 39 34 30 43 39 34 30 43 39 33 30 43 39 33 30 43 39 33 30 43 39 33 30 43 39 30 30 43 39 33 30 43 39 34 30 43 39 31 30 34 30 42 45 31 30 42 45 31 30 42 45 31 30 42 45 31 30 42 43 44 30 42 43 44 30 30 46 43 31 34 31 45 30 33 33 45 32 36 41 32 30 31 36 36 30 46 30 30 30 38 30 30 36 32 30 43 39 34 30 43 39 30 30 30 30 34 30 42 45 31 30 42 45 31 30 30 32 42 45 30 43 32 30 30 32 42 43 42 30 42 30 30 31 37 37 34 44 38 30 30 31 36 46 43 45 45 30 30 30 30 30 44 46 35 30 30 30 30 31 32 31 44 34 33 30 35 39 37 45 46 44 31 41 46 0D
[2024-08-28 21:49:00:429]-Tx->: 7E 32 30 30 32 34 41 34 34 30 30 30 30 46 44 39 46 0D
[2024-08-28 21:49:00:681]-Rx<-: 7E 32 30 30 32 34 41 30 30 37 30 35 34 31 30 30 31 31 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 34 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 39 30 30 30 30 30 30 30 30 30 30 30 31 30 33 30 32 30 30 30 30 30 30 30 30 30 30 45 44 43 31 0D
[2024-08-28 21:49:00:885]-Tx->: 7E 02 33 00 FC 0D

Commands communicating with Battery #5.
[2024-08-28 21:52:16:841]-Tx->: 7E 05 33 00 FE 0D
[2024-08-28 21:52:16:973]-Rx<-: 7E 05 33 15 51 54 2D 59 53 30 31 2D 38 53 56 31 30 30 41 2D 56 33 2E 31 30 B6 0D
[2024-08-28 21:52:17:424]-Tx->: 7E 32 30 30 35 34 41 34 32 30 30 30 30 46 44 39 45 0D
[2024-08-28 21:52:17:798]-Rx<-: 7E 32 30 30 35 34 41 30 30 43 30 41 41 31 30 30 31 30 38 30 43 39 45 30 43 39 38 30 43 39 39 30 43 39 41 30 43 39 37 30 43 39 44 30 43 39 42 30 43 39 44 30 34 30 42 44 37 30 42 44 37 30 42 44 37 30 42 44 37 30 42 43 44 30 42 43 44 30 31 46 35 30 41 31 35 30 41 39 41 34 37 32 43 30 32 32 44 30 46 30 30 30 45 30 30 35 42 30 43 39 45 30 43 39 37 30 30 30 37 30 42 44 37 30 42 44 37 30 30 38 38 38 38 46 43 30 30 38 38 30 30 36 46 30 30 32 34 39 32 45 38 30 30 32 33 43 37 43 35 30 30 30 30 31 37 33 38 30 30 30 30 32 35 30 46 33 45 44 39 31 33 42 33 44 38 31 45 0D
[2024-08-28 21:52:18:024]-Tx->: 7E 32 30 30 35 34 41 34 34 30 30 30 30 46 44 39 43 0D
[2024-08-28 21:52:18:275]-Rx<-: 7E 32 30 30 35 34 41 30 30 38 30 34 34 31 30 30 31 30 38 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 34 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 39 30 30 30 30 30 30 30 30 30 30 30 31 30 33 30 32 30 30 30 30 30 30 30 30 30 30 46 30 42 37 0D
[2024-08-28 21:52:18:478]-Tx->: 7E 05 33 00 FE 0D

Zeezichtje, 2024-08-29:
> 1,3,5,7,9

This works. However, what is the correct HEX for number 11? 0x11 and 0x0A do not work.

mr-manuel, 2024-08-29:
Google is your friend: https://www.rapidtables.com/convert/number/decimal-to-hex.html?x=11
`11` -> `0x0B`
`13` -> `0x0D`
`15` -> `0x0F`

mr-manuel, 2024-10-07:
Closed in favor of https://github.com/Louisvdw/dbus-serialbattery/issues/1104.

closed 2024-10-07 by mr-manuel