Louisvdw/dbus-serialbattery #1125: Only one battery showing in Venus and frequently disconnects

NickRoberts174, 2024-09-11:
### Describe the bug

**My Setup**
Victron Cerbo GX
Victron MPPT 250|100 Solar Charge Controller 
Victron Multiplus-II 48/5000/70
Battery 1 - 280AH x18s LiFePo4, Daly 150A BMS, Parralel protection board, 1A active Balancer and WNT board 
Battery 2 - 280AH x18s LiFePo4, Daly 150A BMS, Parralel protection board, 1A active Balancer and WNT board 
Batteries are connected in series. 
dbus serial battery driver version 1.4.20240902dev (nightly release) 

The Daly WNT board offers RS485, RS232 and CAN connection, Batteries are addressed via dip switches. 

The issue I am having is connecting both batteries to the dbus serial driver (via RS485) at once. I have 2x USB to serial converters connected to the cerbo, they both work individually, but if you plug both in at the same time, it causes the first battery to show as disconnected, The driver does not display both modules at the same time, which is what I am hoping to achieve.

Solutions I have already tried
1. Alternative settings of the dip switches (all off, swapped addresses, set the same etc) 
2. CAN-bus connection to the cerbo - again, it recognises the first battery, when connecting the second, the values change between something completely incorrect to the values of battery pack 1. I have the correct patch leads for going between cerbo and daly, have tried using the line terminator, again tried re-addressing the units ect. 
3. have tried each RS485-USB device separately, both work individually and read correctly, but not together. 
4 have NOT tried dbus driver on CAN (cant find the CAN addresses) 

### How to reproduce

Try to add 2x daly smart BMS in series on RS485 to driver 

[image]

### Expected behavior

I was expecting for the driver to displace data for both Battery 1 and Battery 2. 

[image]

### Driver version

1.4.20240902dev (nightly release)

### Venus OS device type

Cerbo GX

### Venus OS version

v3.41

### BMS type

Daly Smart BMS

### Cell count

18

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
[... 579 lines cut ...]
@4000000066e156151398cad4 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156a213545b4c WARNING:SerialBattery:No data received in read_alarm_data()
@4000000066e156a227c0912c INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156a32394a28c INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156a430657914 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156a5301983c4 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156a62670c83c INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156a713d80e24 WARNING:SerialBattery:No data received in read_alarm_data()
@4000000066e156a735e1c12c INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156ba3704b58c INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e156d4018bf924 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e1574f2b7a553c WARNING:SerialBattery:No data received in read_alarm_data()
@4000000066e1575004712b3c INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e157510ea6e09c INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e15765057149a4 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e157653b652d64 WARNING:SerialBattery:No data received in read_alarm_data()
@4000000066e1576618ab25bc INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e157670e1ca1fc INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
@4000000066e1576735b2642c WARNING:SerialBattery:No data received in read_alarm_data()
@4000000066e157680e3c4b74 INFO:SerialBattery:refresh_data: result: False. If you don't see this warning very often, you can ignore it.
```

### Any other information that may be helpful

Nil

mr-manuel, 2024-09-11:
Hello, this is only one log, of one battery/serial adapter. Please post both logs.

NickRoberts174, 2024-09-12:
Hi Louis,There is only one log in that location? Is there another location for the other log? Thanks, Nick

mr-manuel, 2024-09-12:
Hi Nick, I'm not Louis. You can find everything in the documentation: https://mr-manuel.github.io/venus-os_dbus-serialbattery_docs/troubleshoot/#datalogdbus-serialbatteryttyusbcurrent-or-datalogdbus-serialbatteryttyama0current

closed 2024-10-07 by mr-manuel