Louisvdw/dbus-serialbattery #1133: Dbus-serial battery driver not recognized

NorthTown2022, 2024-10-11:
### Describe the problem

dbus-serial battery driver has stopped working and is no longer seen by the GX under Device list.
[image]

### Driver version

v1.4.20240928

### Venus OS device type

Cerbo GX

### Venus OS version

v3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
BMS_TYPE = Jkbms
LOGGING = INFO

MAX_BATTERY_CHARGE_CURRENT = 140
MAX_BATTERY_DISCHARGE_CURRENT = 100

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max ba                                                                             >
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V =                                                                             >
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.350

; --------- SOC reset voltage ---------
SOC_RESET_VOLTAGE = 3.620

; --------- SOC calculation ---------
SOC_CALCULATION = TRUE
SOC_RESET_CURRENT = 7
SOC_RESET_TIME = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300

; Temperature sensor 1 name
TEMP_1_NAME = Temp Cells 4|5

; Temperature sensor 2 name
TEMP_2_NAME = Temp Cells 12|13

LINEAR_LIMITATION_ENABLE = True

CVCM_ENABLE = True
[... 156 lines cut ...]
2024-10-11 23:04:07.814214500     from utils import logger, publish_config_variables
2024-10-11 23:04:07.814217500   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 44, in <module>
2024-10-11 23:04:07.814402500     config.read([default_config_file_path, custom_config_file_path])
2024-10-11 23:04:07.814407500   File "/usr/lib/python3.8/configparser.py", line 697, in read
2024-10-11 23:04:07.814410500     self._read(fp, filename)
2024-10-11 23:04:07.814412500   File "/usr/lib/python3.8/configparser.py", line 1082, in _read
2024-10-11 23:04:07.814415500     raise MissingSectionHeaderError(fpname, lineno, line)
2024-10-11 23:04:07.814533500 configparser.MissingSectionHeaderError: File contains no section headers.
2024-10-11 23:04:07.814538500 file: '/opt/victronenergy/dbus-serialbattery/config.ini', line: 1
2024-10-11 23:04:07.814541500 'BMS_TYPE = Jkbms\n'
^C
root@einstein:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: down 8 seconds, normally up
root@einstein:~#
```

### Any other information that may be helpful

The driver was installed and working perfectly for many months alongside GUI mods on Venus OS 3.30 Large (I am using a flow to control Grafana on raspberry PI).

I was using an older version of the driver (v1.2.202404xx) and started having some issue where CVL was limited to float voltage well before 100% SOC. I tried to update remotely via Tailscale to v1.4.20240928 and was somewhat successful. The new drive showed up under Device list and Remote Console. After several remote reboots I lost connection to the driver.

I am now at the location and tried to reflash the driver, as well follow all troubleshooting. It seems like there is a problem with the software on the Cerbo GX that does not allow data from the USB port. I tried swapping USB ports but no luck.

I have a spare Cerbo GX that has the driver flashed and it recognized SerialBattery in Device list right away so I can definitely rule out any cabling issues.

mr-manuel, 2024-10-12:
> `configparser.MissingSectionHeaderError: File contains no section headers.`

Somehow you deleted the first line of your `config.ini`. Add it again and it should work:

https://github.com/Louisvdw/dbus-serialbattery/blob/fd0fc1a193427df6ca0fadd141aa57826f761819/etc/dbus-serialbattery/config.default.ini#L1

NorthTown2022, 2024-10-12:
Yeah, something happened and my complete config file was empty so I copy and pasted my config from a backup file. Must have missed that first line. Everything works now. Thank you 😀

closed 2024-10-12 by NorthTown2022