Louisvdw/dbus-serialbattery #1134: Could not reach full SOC today

NorthTown2022, 2024-10-16:
### Describe the problem

Charge mode was in Float (Linear mode) all day and could not reach a full SOC. 

### Driver version

v1.4.20240928

### Venus OS device type

Cerbo GX

### Venus OS version

v3.30

### BMS type

JKBMS / Heltec BMS

### Cell count

16s

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
BMS_TYPE = Jkbms
LOGGING = INFO

MAX_BATTERY_CHARGE_CURRENT = 140
MAX_BATTERY_DISCHARGE_CURRENT = 100

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max ba                                                                             >
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V =                                                                             >
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.350

; --------- SOC reset voltage ---------
SOC_RESET_VOLTAGE = 3.620

; --------- SOC calculation ---------
SOC_CALCULATION = TRUE
SOC_RESET_CURRENT = 7
SOC_RESET_TIME = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300

; Temperature sensor 1 name
TEMP_1_NAME = Temp Cells 4|5

; Temperature sensor 2 name
TEMP_2_NAME = Temp Cells 12|13

LINEAR_LIMITATION_ENABLE = True

CVCM_ENABLE = True
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
CCCM_T_ENABLE = True
DCCM_T_ENABLE = True

; --------- Cell Voltage limitation (affecting CVL) ---------
CVL_ICONTROLLER_MODE = True
CVL_ICONTROLLER_FACTOR = 0.2

VOLTAGE_DROP = 0.02
```

### Relevant log output

```shell
2024-10-15 23:47:35.548664500 INFO:SerialBattery:
2024-10-15 23:47:35.559092500 INFO:SerialBattery:Starting dbus-serialbattery
2024-10-15 23:47:35.568756500 INFO:SerialBattery:Venus OS v3.30
2024-10-15 23:47:35.590856500 INFO:SerialBattery:dbus-serialbattery v1.4.20240928
2024-10-15 23:47:51.622911500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-10-15 23:47:51.622920500 INFO:SerialBattery:Testing Jkbms
2024-10-15 23:47:51.775321500 INFO:SerialBattery:Connection established to Jkbms
2024-10-15 23:48:00.038073500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-10-15 23:48:00.695487500 INFO:SerialBattery:DeviceInstance = 1
2024-10-15 23:48:00.698836500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-10-15 23:48:00.705333500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-10-15 23:48:00.710892500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0__0
2024-10-15 23:48:00.912614500 INFO:SerialBattery:publish config values = False
2024-10-15 23:48:00.925593500 INFO:SerialBattery:Polling data every 1.000 s
2024-10-15 23:48:00.936952500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-10-15 23:48:00.938311500 INFO:SerialBattery:========== Settings ==========
2024-10-15 23:48:00.941372500 INFO:SerialBattery:> Connection voltage: 53.03 V | Current: -2.42 A | SoC: 88%
2024-10-15 23:48:00.941380500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-10-15 23:48:00.941383500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-10-15 23:48:00.941859500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.350 V
2024-10-15 23:48:00.960042500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 140.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A
2024-10-15 23:48:00.971308500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 140.0 A | MAX BATTERY DISCHARGE CURRENT: 160.0 A (read from BMS)
2024-10-15 23:48:00.972772500 INFO:SerialBattery:> CVCM:     True
2024-10-15 23:48:00.973896500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-10-15 23:48:00.974547500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-10-15 23:48:00.975276500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-10-15 23:48:00.976568500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-10-15 23:48:00.976871500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_B2A20S20P
2024-10-15 23:47:35.548664500 INFO:SerialBattery:
2024-10-15 23:47:35.559092500 INFO:SerialBattery:Starting dbus-serialbattery
2024-10-15 23:47:35.568756500 INFO:SerialBattery:Venus OS v3.30
2024-10-15 23:47:35.590856500 INFO:SerialBattery:dbus-serialbattery v1.4.20240928
2024-10-15 23:47:51.622911500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-10-15 23:47:51.622920500 INFO:SerialBattery:Testing Jkbms
2024-10-15 23:47:51.775321500 INFO:SerialBattery:Connection established to Jkbms
2024-10-15 23:48:00.038073500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-10-15 23:48:00.695487500 INFO:SerialBattery:DeviceInstance = 1
2024-10-15 23:48:00.698836500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-10-15 23:48:00.705333500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-10-15 23:48:00.710892500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0__0
2024-10-15 23:48:00.912614500 INFO:SerialBattery:publish config values = False
2024-10-15 23:48:00.925593500 INFO:SerialBattery:Polling data every 1.000 s
2024-10-15 23:48:00.936952500 INFO:SerialBattery:Battery JKBMS connected to dbus from /dev/ttyUSB0
2024-10-15 23:48:00.938311500 INFO:SerialBattery:========== Settings ==========
2024-10-15 23:48:00.941372500 INFO:SerialBattery:> Connection voltage: 53.03 V | Current: -2.42 A | SoC: 88%
2024-10-15 23:48:00.941380500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-10-15 23:48:00.941383500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-10-15 23:48:00.941859500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.350 V
2024-10-15 23:48:00.960042500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 140.0 A | MAX BATTERY DISCHARGE CURRENT: 100.0 A
2024-10-15 23:48:00.971308500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 140.0 A | MAX BATTERY DISCHARGE CURRENT: 160.0 A (read from BMS)
2024-10-15 23:48:00.972772500 INFO:SerialBattery:> CVCM:     True
2024-10-15 23:48:00.973896500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-10-15 23:48:00.974547500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-10-15 23:48:00.975276500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-10-15 23:48:00.976568500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-10-15 23:48:00.976871500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_B2A20S20P
```

### Any other information that may be helpful

I am away from the site at the moment but have remote access. 

There is minimal draw on battery, just Starlink and a few other small loads. SOC sat around 89% most of the day (usually full SOC very early in the day especially with only small loads). 

Charge controller was only putting out 100-400W even during sunny periods which is very low.

Hope I am posting just the right information ;)

Screenshots take at different parts of the day:
[image]

[image]

[image]

mr-manuel, 2024-10-16:
Hello,

thank you very much to providing all the needed informations. It#s very rare that users provide all I need.

If you look at this screenshot `-- switch to bulk requirements (Linear Mode) --`, then you see that neither of requirement a) or b) is met. The battery is not charged every day to bulk, if it is still full from the last charge.

[image]

NorthTown2022, 2024-10-16:
Thank you for the explanation. I had read the parameter description but was still puzzled why before I could reach a full SOC (each day) at a lower CVL. 

After looking back at my VRM graph from last month it shows I was charging to 54v on float instead of my current setting of 53.6v. I am pretty confident that if I modify my float setting back to 3.375v the battery will reach full SOC again.

mr-manuel, 2024-10-16:
Or you change `SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT` to a higher value like `95`, if you want always charge to max voltage.

NorthTown2022, 2024-10-17:
Thank you for the suggestion. I added SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT= 90  to config.ini and kept float at 3.35v.  Seems to do the trick nicely 😉

closed 2024-10-17 by NorthTown2022