Louisvdw/dbus-serialbattery #1140: dbus-serialbattery on Venu OS 3.50 GUI_V2

VladKh, 2024-11-01:
### Describe the bug

The "MOSFET temperature" and "Cell voltages" tabs are not displayed in the new Venu OS 3.50 GUI_V2.

### How to reproduce

GUI_V1
[image]

GUI_V2
[image]

### Expected behavior

We all want the "MOSFET temperature" and "Cell voltages" tabs in the new Venu OS 3.50 GUI_V2.

### Driver version

1.4.20240928

### Venus OS device type

Cerbo GX

### Venus OS version

3.50

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]
; --------- Set logging level ---------
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 140.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
MIN_CELL_VOLTAGE   = 3.000
MAX_CELL_VOLTAGE   = 3.400
FLOAT_CELL_VOLTAGE = 3.350

; --------- SOC reset voltage (needs to match BMS settings) ---------
SOC_RESET_VOLTAGE = 3.650
SOC_RESET_AFTER_DAYS =

; --------- SOC calculation ---------
SOC_CALCULATION   = False
SOC_RESET_CURRENT = 7
SOC_RESET_TIME    = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300

; --------- Bluetooth BMS ---------
BLUETOOTH_BMS =

; --------- Bluetooth use USB ---------
BLUETOOTH_USE_USB = False

; --------- CAN BMS ---------
CAN_PORT =
CAN_SPEED = 500

; --------- Modbus (multiple BMS on one serial adapter) ---------
MODBUS_ADDRESSES =

; --------- BMS disconnect behaviour ---------
BLOCK_ON_DISCONNECT = True
BLOCK_ON_DISCONNECT_TIMEOUT_MINUTES = 10
BLOCK_ON_DISCONNECT_VOLTAGE_MIN = 3.25
BLOCK_ON_DISCONNECT_VOLTAGE_MAX = 3.35

; --------- Charge mode ---------
LINEAR_LIMITATION_ENABLE = True
LINEAR_RECALCULATION_EVERY = 60
LINEAR_RECALCULATION_ON_PERC_CHANGE = 33

; --------- External current sensor ---------
EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE =
EXTERNAL_CURRENT_SENSOR_DBUS_PATH =

; --------- Charge Voltage limitation (affecting CVL) ---------
CVCM_ENABLE = True

; -- CVL reset based on cell voltage diff (linear mode)
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.010
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.013
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.080

; -- CVL reset based on SoC option (step mode & linear mode)
MAX_VOLTAGE_TIME_SEC = 900
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 80

; --------- Cell Voltage Current limitation (affecting CCL/DCL) ---------
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True

CELL_VOLTAGES_WHILE_CHARGING = 3.55, 3.50, 3.45, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION = 0, 0.05,  0.5,    1

CELL_VOLTAGES_WHILE_DISCHARGING = 2.70, 2.80, 2.90, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION = 0,  0.1,  0.5,    1

; --------- Cell Voltage limitation (affecting CVL) ---------
CVL_ICONTROLLER_MODE   = False
CVL_ICONTROLLER_FACTOR = 0.2

; --------- Temperature limitation (affecting CCL/DCL) ---------
CCCM_T_ENABLE = True
DCCM_T_ENABLE = True

TEMPERATURES_WHILE_CHARGING   =    0,    2,    5,   10,   15,   20,   35,   40,   55
MAX_CHARGE_CURRENT_T_FRACTION = 0.00, 0.10, 0.20, 0.40, 0.80, 1.00, 1.00, 0.40, 0.00

TEMPERATURES_WHILE_DISCHARGING   =  -20,    0,    5,   10,   15,   45,   55
MAX_DISCHARGE_CURRENT_T_FRACTION = 0.00, 0.20, 0.30, 0.40, 1.00, 1.00, 0.00

; --------- SOC limitation (affecting CCL/DCL) ---------
CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False

SOC_WHILE_CHARGING                 =   98,   95,   90,   85
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.10, 0.20, 0.50, 1.00

SOC_WHILE_DISCHARGING              =    5,   10,   15,   20
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.10, 0.20, 0.50, 1.00

; --------- Time-To-Go ---------
TIME_TO_GO_ENABLE = True

; --------- Time-To-Soc ---------
TIME_TO_SOC_POINTS =
TIME_TO_SOC_VALUE_TYPE = 1
TIME_TO_SOC_RECALCULATE_EVERY = 60
TIME_TO_SOC_INC_FROM = False

; --------- Additional settings ---------
BMS_TYPE = Jkbms_pb, Jkbms

EXCLUDED_DEVICES =

; Default for most BMS is 1 second, some BMS may have a higher value
POLL_INTERVAL =

; Auto reset SoC
AUTO_RESET_SOC = True

; Publish the config settings to the dbus path "/Info/Config/"
PUBLISH_CONFIG_VALUES = True

; Select the format of cell data presented on dbus [Valid values 0,1,2,3]
; 0 Do not publish all the cells (only the min/max cell data as used by the default GX)
; 1 Format: /Voltages/Cell (also available for display on Remote Console)
; 2 Format: /Cell/#/Volts
; 3 Both formats 1 and 2
BATTERY_CELL_DATA_FORMAT = 1

; Simulate Midpoint graph (True/False).
MIDPOINT_ENABLE = False

; Battery temperature
; Specify how the battery temperature is assembled
; 0 Get mean of temperature sensor 1 to sensor 4
; 1 Get only temperature from temperature sensor 1
; 2 Get only temperature from temperature sensor 2
; 3 Get only temperature from temperature sensor 3
; 4 Get only temperature from temperature sensor 4
TEMP_BATTERY = 0

; Temperature sensor 1 name
TEMP_1_NAME = Temp 1
; Temperature sensor 2 name
TEMP_2_NAME = Temp 2
; Temperature sensor 2 name
TEMP_3_NAME = Temp 3
; Temperature sensor 2 name
TEMP_4_NAME = Temp 4

; Show additional info in GUI -> Serialbattery -> Parameters
GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True

; Telemetry settings
TELEMETRY = True

; --------- BMS specific settings ---------

; -- Unique ID settings
USE_PORT_AS_UNIQUE_ID = False

; -- JKBMS settings
JKBMS_CAN_CELL_COUNT = 1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT    = 15

; --------- Voltage drop ---------
VOLTAGE_DROP = 0.00
```

### Relevant log output

```shell
No log output
```

rbeudel, 2024-11-03:
No cell voltages with daly bms, too.

DIVERSDIVERS, 2024-11-04:
Hi, 

they are still there but not easy to find.

GUI_V2

on Setting Page -> scroll down to end there you find "DEBUG" -> Values -> battery

then select battery -> scroll down to end to Voltages then you can see it.

May it could be fixed in the future because under DEBUG you find a lot of more information as you can see

[image]
[image]
[image]

mr-manuel, 2024-11-04:
This repository is not maintained anymore.

GUIv2 can be installed seperately since a long time, if you read the README in my repository. See also https://github.com/mr-manuel/venus-os_dbus-serialbattery/discussions/101

The installation of GUIv2 is included in the latest nightly.

@DIVERSDIVERS this pages show all values on `dbus` for debugging.

closed 2024-11-04 by mr-manuel