Louisvdw/dbus-serialbattery #1145: JK-DB6A20S10P - raspi

abijvoet, 2024-11-25:
### Describe the problem

My JK-DB6A20S10P won't show data via JK RS485 module and (a variety of) RS485 to USB adapter(s) on my Raspi 4 B rev. 1.1. with Venus OS 3.51. 
- I used different versions of the VEnus OS, from 3.41 through latest, but no solution.
- I followed your and Andy's installation information (it should be easy...)
- I worked my way though the troubleshooting section. Basically the USB is seen bij Venus OS but there is no data visible
- USB rs485 adapter does not show rx/tx blinking
- In JK-BMS app I choose UART setting 000 - 4G-GPS Remote module communication
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Driver version

venus-image-large-raspberrypi4-20241104170247-v3.51.

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.51

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

### Relevant log output

```shell
Using username "root".
root@192.168.1.118's password:
Last login: Mon Nov 25 12:57:05 2024

root@raspberrypi4:~# tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
^C
root@raspberrypi4:~#

root@raspberrypi4:~#
root@raspberrypi4:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-11-25 14:42:12.134662500 INFO:SerialBattery:
2024-11-25 14:42:12.135121500 INFO:SerialBattery:Starting dbus-serialbattery
2024-11-25 14:42:12.136172500 INFO:SerialBattery:Venus OS v3.52
2024-11-25 14:42:12.136577500 INFO:SerialBattery:dbus-serialbattery v1.4.20240928
2024-11-25 14:42:28.148774500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-11-25 14:42:28.148779500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-11-25 14:42:28.281209500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:28.281724500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-11-25 14:42:28.410825500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:28.411342500 INFO:SerialBattery:Testing Daren485 at address "\x01"
2024-11-25 14:42:28.823401500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:28.823407500 INFO:SerialBattery:Testing Ecs
2024-11-25 14:42:28.885693500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:28.886182500 INFO:SerialBattery:Testing EG4_Lifepower at address "\x01"
2024-11-25 14:42:29.159086500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:29.164252500 INFO:SerialBattery:Testing EG4_LL at address "\x01"
2024-11-25 14:42:29.428235500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:29.429003500 INFO:SerialBattery:Testing HeltecModbus at address "\x01"
2024-11-25 14:42:33.122348500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:33.122789500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-11-25 14:42:42.142727500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:42.142732500 INFO:SerialBattery:Testing Jkbms
2024-11-25 14:42:42.412639500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-11-25 14:42:42.414202500 INFO:SerialBattery:Testing Jkbms_pb at address "\x01"
^C
root@raspberrypi4:~#

root@raspberrypi4:~#
root@raspberrypi4:~# tail -F /data/log/serial-starter/current | tai64nlocal
2024-11-25 14:42:05.014347500 INFO: Start service vedirect-interface.ttyUSB0 once
2024-11-25 14:42:10.301175500 INFO: Start service dbus-serialbattery.ttyUSB0 once
2024-11-25 14:44:01.051108500 INFO: Start service dbus-cgwacs.ttyUSB0 once
2024-11-25 14:44:06.328255500 INFO: Start service gps-dbus.ttyUSB0 once
2024-11-25 14:44:15.663414500 INFO: Start service vedirect-interface.ttyUSB0 once
2024-11-25 14:44:20.934913500 INFO: Start service dbus-serialbattery.ttyUSB0 once
2024-11-25 14:46:11.699122500 INFO: Start service dbus-cgwacs.ttyUSB0 once
2024-11-25 14:46:14.940718500 INFO: Start service gps-dbus.ttyUSB0 once
2024-11-25 14:46:24.268081500 INFO: Start service vedirect-interface.ttyUSB0 once
2024-11-25 14:46:29.543947500 INFO: Start service dbus-serialbattery.ttyUSB0 once
^C
root@raspberrypi4:~#

root@raspberrypi4:~#
root@raspberrypi4:~# ls -l /dev/ttyUSB*
crw-rw----    1 root     dialout   188,   0 Nov 25 14:49 /dev/ttyUSB0
root@raspberrypi4:~#

root@raspberrypi4:~#
root@raspberrypi4:~# head /data/var/lib/serial-starter/*
sbattery
root@raspberrypi4:~#

root@raspberrypi4:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: down 8 seconds, normally up
root@raspberrypi4:~#
root@raspberrypi4:~#
root@raspberrypi4:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: up (pid 796) 8 seconds
root@raspberrypi4:~#

root@raspberrypi4:~# ls -l /dev/ttyUSB*
crw-rw----    1 root     dialout   188,   0 Nov 25 14:01 /dev/ttyUSB0
root@raspberrypi4:~#

root@raspberrypi4:~# uptime
 14:53:52 up  1:56,  load average: 0.32, 0.29, 0.20
root@raspberrypi4:~#

root@raspberrypi4:~#
root@raspberrypi4:~# ps | grep serial
  883 root      1608 S    supervise serial-starter
  902 root      1620 S    multilog t s25000 n4 /var/log/serial-starter
  930 root      3044 S    {serial-starter.} /bin/bash /opt/victronenergy/serial-starter/serial-starter.sh
 1845 root      1608 S    supervise dbus-serialbattery.ttyUSB0
 1848 root      1752 S    multilog t s25000 n4 /var/log/dbus-serialbattery.ttyUSB0
21984 root      3044 S    {start-serialbat} /bin/bash /opt/victronenergy/dbus-serialbattery/start-serialbattery.sh ttyUSB0
21986 root     35888 S    python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB0
22097 root      2692 S    grep serial
root@raspberrypi4:~#

root@raspberrypi4:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@raspberrypi4:~#

root@raspberrypi4:~# dmesg | grep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:29:95:BA vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootwait
[    1.446359] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2
[    1.455492] printk: console [ttyS0] disabled
[    1.455629] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 37, base_baud = 62500000) is a 16550
[    2.783441] printk: console [ttyS0] enabled
[    4.498499] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
root@raspberrypi4:~#

root@raspberrypi4:~# lsmod | grep usb
usbserial              36864  3 ch341
root@raspberrypi4:~#
```

mr-manuel, 2024-11-25:
Did they send you the RS485 adapter with the BMS or did you choose it yourself? There are different versions for 8S BMS and the other.

You could try to connect via TTL. You find everything in the documentation.

abijvoet, 2024-11-25:
Hi Manuel,I selected it myself, even twice, so maybe I did choose the 8s version.Now I’ll try this one, it’s on Its way: AliExpress: €2,70 | 1Pcs PL2303 Ta Usb Ttl RS232 convert serial PL2303TA https://a.aliexpress.com/_EJndyhBThat might do the trick, I hope… Is this Ok?Kind regards,ArnoudOn 25/11/2024 at 17:44, Manuel wrote:
                
            
            
                
                    From: "Manuel" ***@***.***>Date: 25 November 2024To: "Louisvdw/dbus-serialbattery" ***@***.***>Cc: "abijvoet" ***@***.***>,"Author" ***@***.***>Subject: Re: [Louisvdw/dbus-serialbattery] JK-DB6A20S10P - raspi  (Issue #1145)
                
                
Did they send you the RS485 adapter with the BMS or did you choose it yourself? There are different versions for 8S BMS and the other.
You could try to connect via TTL. You find everything in the documentation.

mr-manuel, 2024-11-26:
If you still need help, open an issue in my repository.

closed 2024-11-26 by mr-manuel

abijvoet, 2024-12-03:
I'm not sure this is the right place to continue this issue, if not, point out the right place please. Now I got this message after installing a new RS485 (16S) adapter:
....
2024-12-03 18:38:06.821128500 INFO:SerialBattery:Testing Jkbms
2024-12-03 18:38:22.752242500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:196/32897]
...

What does the '[len:196/32897]' mean?